realsenseai/librealsense #10171: D450 connection issues and unable to update from recovery

v-lopez, 2022-01-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | 450  | 
| Firmware Version                   | ??? | 
| Operating System & Version |   Win 10 and Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.13.0                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.50.0                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  others                           | 

### Issue Description

I have two D450 cameras and two D4 boards, I cannot connect to any of them.
I have tried on 3 computers (2 Ubuntu 20.04 and a Windows 10), with 6 different USB cables.

Initially, the problem was that the board + camera was not detected by any of the Realsense tools (viewer, enumerate-devices, fw-update...).
On linux the device appeared as "Invalid SKU".
On windows the device appeared as Intel 4n0.

After following all the guides I could and nothing worked, I tried plugging a D4 board without the camera attached.

The device is then listed as "Depth Camera 400" and is detected by the Realsense Viewer. 
The realsense viewer asks for a firmware update (to 5.13.0.50), and I can apply it  succesfully.

After that, I disconnect the board, connect the D450 camera to the D4 board, and reconnect to the PC.
It is now detected by the realsense viewer, as a "Realsense D4XX USB2 Recovery", and asks for another FW update (to 5.13.0.50).
However, this update never finishes, and is stuck at the step "waiting for the device to reconnect". Eventually it aborts waiting.

If I reconnect, it asks for the same update, and doesn't finish either with the same problem.

I also attempted https://support.intelrealsense.com/hc/en-us/community/posts/360039449253-D435i-Firmware-Recovery-Failed and reset the calibration to the gold factory settings with no luck.

MartyG-RealSense, 2022-01-20:
Hi @v-lopez  How are you connecting the D4 board and the depth module together - with a custom-made flexible cable or with a pre-made D4 Board that you purchased?  In a discussion in the link below, Intel advised that an interposer designed for the D435 would not work correctly with D450 and you should create your own interposer for it.

https://support.intelrealsense.com/hc/en-us/community/posts/1500001048801-D450-modules-are-being-recognized-as-D430i

The discussion provides a source for ordering an interposer cable if you do not want to make your own.

v-lopez, 2022-01-21:
The D4 Board was not V3. 
Today we received a batch of the V3 ones and it works out of the box.

Thanks @MartyG-RealSense for the swift answer.

closed 2022-01-21 by v-lopez

MartyG-RealSense, 2022-01-21:
That's great news, @v-lopez - thanks very much for the update   :)