realsenseai/librealsense #10172: Rs-server: Waiting for device using a D455

Nekzen, 2022-01-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (Open RealSense Viewer --> 2.50.0) | 
| Operating System & Version |   {Win 10 / Raspbian for server | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC/Raspberry Pi/  |
| SDK Version                            |  {legacy / 2.50.50                          | 
| Language                            |  {French }                          | 
| Segment			|  {AR}         | 

### Issue Description

My end-goal is to be able to connect multiple D455 cameras, each one on a raspberry pi 4 and be able to stream all feeds on a W10 computer. I plan to use a local network using a switch.

I found this tutorial that was super helpful:

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

I am using the pre-built image that is linked in section 2.3.
However, I can't get access to the feed of the camera on W10.

The tutorial states that you must enter the IP-address of the camera when adding a network device on realsense2 (Linux server running rs-server) which here is my RPI4.

I used PuTTY to access the Pi4 remotely using the IP address I found with ifconfig. In my case it is 169.254.58.202

I successfully connected to the terminal but when I try to launch rs-server from the RPi4, I have the following output:

**R-server is running
Waiting for Device...**

and nothing else happens.

Obviously the D455 is connected to the RPi4  on one of the two 3.0 usb slots but it doesn't seem to be picked up when checking for devices. At least this is my understanding of the problem.

I then go back to W10, run realsense2 and add a network device. When entering the address of the RPI4, I get the following error : **[RsRTSPClient::getStreams] error: client time out - 3**
Which perfectly makes sense because the client is not technically "running".

I'm not sure what I'm doing wrong here?

Thanks for anyone who will have a suggestion.

MartyG-RealSense, 2022-01-20:
Hi @Nekzen  If you are using the pre-built SD card image from the ethernet guide, my recollection is that the image used RealSense SDK version **2.34.0**.  If it is still using 2.34.0, support for the D455 camera model was not added until SDK version **2.38.1**.

Nekzen, 2022-01-20:
Thanks for the quick answer @MartyG-RealSense !

> Hi @Nekzen If you are using the pre-built SD card image from the ethernet guide, my recollection is that it was used with RealSense SDK version **2.34.0**. If it is still using 2.34.0, then it would not know how to access the D455 camera model as support for D455 was not added until SDK version **2.38.1**. I see though that you have SDK 2.50.0 installed, so that would likely not be the problem.

Version 2.5.0 is installed on the computer with Windows 10 though. According to what you just wrote, I understand that rs-server starts with sdk version 2.34.0 ? If so then it makes sense that D455 isn't supported. I also have access to a D415 and a D435. I will try with both other cameras and update you on this. If it ever works with one camera, will I be able to make it work with 2 devices simultaneously ? 

> Are you ssh'ing to the Pi from Putty on the Windows host machine and then starting rs-server from Putty over the ssh connection with **sudo systemctl restart rs-server.service**. Or are you starting rs-server directly on the Pi?

I have tried both methods and it didn't work either way. Would you recommend one works best ?

Nekzen

MartyG-RealSense, 2022-01-20:
I would recommend following the guide as closely as possible during initial setup and testing, including using a static IP.  Then if it still does not work correctly then it should be easier to diagnose than if you are using customizations such as your own IP address.

Yes, please do try the other camera models and update here regarding whether it makes a difference.

RealSense ethernet projects - including the older **EtherSense** one for Python in the link below - typically have one camera per board.  So a two-camera setup should have two individual boards, each with a camera attached.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/ethernet_client_server/README.md

I have seen cases where rs-server has been started over the ethernet connection OR directly on the Pi.  So as far as I am aware it should not make much difference how it is started.

Nekzen, 2022-01-21:
> Yes, please do try the other camera models and update here regarding whether it makes a difference.

Unfortunately, it did not make a difference. I can't get farther than "Waiting for device...." using the D415 and D435
The cameras are working properly because I tested them in realsense-viewer directly.

> RealSense ethernet projects - including the older **EtherSense** one for Python in the link below - typically have one camera per board. So a two-camera setup should have two individual boards, each with a camera attached.

Yes, it was what I meant in my previous message. There will be a single camera per RPi4.

Is there any way that I can make it work ?

MartyG-RealSense, 2022-01-21:
There was a previous case at https://github.com/IntelRealSense/librealsense/issues/9946#issuecomment-966712875 of someone who was also using the pre-built Pi 4 image and receiving the **Waiting for device** message from rs-server.  I referred the matter to Intel but a solution had not been found at the time of writing this.

Intel have also said that an ethernet side branch of the SDK will be released soon, so one option may be to wait for that release to see whether it fixes the waiting for device issue.  A time schedule for release of the side branch is not currently available though.

I have not seen a previous case that refers to the **RsDestructor** message.

Is your Pi board being powered by an external power source connected via USB-C?  Intel's ethernet guide states that this will be required if you are not using a Power Over Ethernet (PoE) 'hat' on your Pi board.

[image]

In regard to multiple boards with one camera per board, the guide illustrates that this is possible.  It looks as though the cameras are connected to an ethernet switch box, and then a single cable runs from the switch box to the host computer.  I would speculate that the multiple cameras would then be listed in the Viewer by adding a Network Camera connection for each individual camera (this is not something that I have personally tested though).

Nekzen, 2022-01-24:
> There was a previous case at #9946 (comment) of someone who was also using the pre-built Pi 4 image and receiving the Waiting for device message from rs-server. I referred the matter to Intel but a solution had not been found at the time of writing this.

Well this is quite unfortunate I have to say.

> I have not seen a previous case that refers to the RsDestructor message.

I will upload screenshots/pictures of the configuration when I get back to the lab.

> Is your Pi board being powered by an external power source connected via USB-C? Intel's ethernet guide states that this will be required if you are not using a Power Over Ethernet (PoE) 'hat' on your Pi board.

Yes, I am using an external power source as I do not have any PoE shield.

> In regard to multiple boards with one camera per board, the guide illustrates that this is possible. It looks as though the cameras are connected to an ethernet switch box, and then a single cable [...]

This is what I understood by looking at the pictures !

Going forward I will probably try a fresh install on a Mate distribution on one of the Rpi4 and see if I can launch the rs-server command without trouble.

MartyG-RealSense, 2022-01-24:
If you are exploring the possibility of using Ubuntu on your Pi, another option that could be considered is **Ubuntu Server 20.04**.  This has worked well in the past when using the librealsense SDK with a Pi 4.

https://ubuntu.com/download/raspberry-pi

the link below has an example of a guide for setting up a Pi 4 with Ubuntu Server 20.04, librealsense and ROS.

https://ramith.fyi/setting-up-raspberry-pi-4-with-ubuntu-20-04-ros-intel-realsense/

I look forward to receiving your next set of test results.  Good luck!

MartyG-RealSense, 2022-01-30:
Hi @Nekzen  Do you require further assistance with this case, please?  Thanks!

Nekzen, 2022-02-01:
Hello, thanks for getting back to me. 

I setup Ubuntu server on the Rpi4 and started following ACrobotic's tutorial on installing SDK 2.0.

Unfortunately I ran into this error: 

[image]

Regarding undefined references in the apriltag library (installed via apt-get install libapriltag-dev).

So I couldn't test with Ubuntu server.

Regarding my current setup: (Pre-image with the ethernet switch)

[image]

[image]

I didn't get any progress on that, I am still blocked with the RsDestructor message. Would a screenshot of this be useful ?

MartyG-RealSense, 2022-02-01:
The Acrobotic guide is not designed for Ubuntu.  It installs **Raspberry Pi OS**, which is based on a different Linux operating system called **Debian**.

https://en.wikipedia.org/wiki/Raspberry_Pi_OS

The **Ramith** installation guide mentioned earlier uses Ubuntu Server.

https://ramith.fyi/setting-up-raspberry-pi-4-with-ubuntu-20-04-ros-intel-realsense/

Nekzen, 2022-02-01:
@MartyG-RealSense 

I restarted the tutorial over. To explain in detail what I did wrong previously: I ignored the copy-paste part of Ramith's tutorial: 

[](https://answers.ros.org/question/363889/intel-realsens-on-ubuntu-2004-ros-noetic-installation-desription/)

as it showed as **deprecated.**

and went directly on :

[](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)

which as this point is the same as the acrobotic's tutorial (more or less).

Now look at this. Wonderful.

[image]

I'm still having issues with **waiting for device** as it does not seem to pickup the camera 100% of the time. I have to unplug the camera sometimes and it does not seem to be related to the choice of the USB 3.0 port.

Thanks for your help. 

Next i'll try to connect several cameras but that is a problem for another day !

Nekzen

MartyG-RealSense, 2022-02-01:
Thanks so much for sharing the details of your solution with the RealSense community  :)

closed 2022-02-04 by Nekzen