realsenseai/librealsense #10173: Save single frameset without frame number

tantalishe, 2022-01-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Operating System & Version |   Linux Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.13.0-27-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.50.0.3812                      | 
| Language                            |  Python 3.8                        | 

### Issue Description:

I am trying to save frameset via save_singe_frameset:

```python
saver = rs.save_single_frameset('filename')
saver.process(frame)
```
and get something like this:

```
filename14.bag
```
Is it possible to save frameset without frame number/id in the end of filename?

MartyG-RealSense, 2022-01-20:
Hi @tantalishe I have not seen the following Python instruction used in a pyrealsense2 script, but perhaps you could define a **%** for the filename string so that it only uses the first 8 characters of the string and excludes all characters after the 8 characters in the word 'filename'.  The link below demonstrates this principle. 

https://www.w3resource.com/python-exercises/python-basic-exercise-120.php

tantalishe, 2022-01-23:
If I understood correctly, I was supposed to do something like
```python
filename = "original_filename"
saver = rs.save_single_frameset('%.8s' % filename)
saver.process(frame)
```
but i still get
```
original14.bag
```
Anyway, thanks for your answer. I think the number is added after the filename is processed, and maybe this can be fixed through some function parameters. But the documentation that I found is not very detailed on this

MartyG-RealSense, 2022-01-23:
My thinking was that perhaps the filename string with the added number could be stored in a variable and then that variable could be character-clipped to a maximum length of 8 to produce the final frame filename.

closed 2022-01-23 by tantalishe