realsenseai/librealsense #10174: Poor streaming performance with v2.50.0 and RSUSB backend

tpwrules, 2022-01-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.5.8.1 | 
| Operating System & Version |  Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.4.0                                      | 
| Platform                                 | PC  |
| SDK Version                            | 2.50.0                      | 
| Language                            |  Realsense Viewer                   | 

### Issue Description
When trying to stream RGB video from an L515 using the RSUSB backend and `realsense-viewer` in the new v2.50.0 SDK, the video is jerky and the dropped frame percentage hovers between 10-20%. Additionally, the kernel log is spammed with messages indicating the connection of a new `uvcvideo` device, as if the camera is constantly disconnecting and reconnecting every second or so. The log pane in `realsense-viewer` doesn't show anything.

If I switch to the v2.49.0 SDK, try to stream RGB video from the same camera, and make no other changes to the setup, the video stream is rock solid and no frames are dropped.

MartyG-RealSense, 2022-01-21:
Hi @tpwrules  The RSUSB backend had changes to it implemented in SDK 2.50.0 - as described in https://github.com/IntelRealSense/librealsense/pull/9821 - and so this might be affecting your particular computer (I have not seen similar reports about 2.50.0 from other RealSense users).

RealSenseSupport, 2022-02-07:
@tpwrules Did you get chance to test with SDK v2.50 to see if you still get the issue? Looking forward to your update. Thanks!

tpwrules, 2022-02-07:
It looks like the libusb object is destructed a couple times a second at least when using realsense-viewer, which means the device is constantly detached and reattached from v4l, causing these glitches.

RealSenseSupport, 2022-02-11:
@tpwrules I can reproduce the issue you've got. Will track this as a bug. Any update will let you know. Thanks!

PeixinC, 2022-08-30:
@MartyG-RealSense @RealSenseSupport. 
Hi, is this issue solved in the latest librealsense v2.51.1? I have the same problem using a Raspberry Pi 4 with D435i. The kernel log shows that the camera repeatedly connects and disconnects, causing severe frame drops. If I use ROS package usb_cam, I observe no frame drop at all. I can also confirm that downgrading librealsense to v2.49.0 solves the problem, but after using the RSUSB backend, applications using the v4l backend can't find the device.

MartyG-RealSense, 2022-08-30:
Hi @PeixinC  A fix for RSUSB / V4L interoperability was implemented in 2.50.0.

[image]

Using 2.50.0 may resolve your problem with the V4L backend not finding the device when using the RSUSB backend.  However, you have been experiencing repeated disconnects and reconnects with 2.50.0.  Is that correct, please?

There have not been SDK updates to RSUSB in 2.51.1 according to the SDK release notes.

PeixinC, 2022-08-30:
Thanks for the information. Yes, you are correct. But according to tpwrules who found that

> I only have this issue on 2.50.0. 2.49.0 and below work fine.
> 
> I tested reverting PR #9821 and that fixes the issue, so I think that PR might need revision.

I will experience frame drops if I added the code in green to 2.49.0 source code. Please tell me if I misunderstand anything.

MartyG-RealSense, 2022-08-31:
My theory would be that if 2.49.0 has good performance for you but has the V4L interoperability problem then adding the code in green may give good performance **and** resolve the V4L issue.

RealSenseCustomerSupport, 2024-11-04:
----------------------------------------------

@**tpwrules** ** **L515 is EOL. If SDK v2.49.0 works well for you, please go ahead with v2.49.0. This ticket will be closed. Thanks!

closed 2024-11-04 by MartyG-RealSense