realsenseai/librealsense #10175: Unable to view video on Jetson TX2

Kaijun101, 2022-01-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.13.50| 
| Operating System & Version |   Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |         4.9.140-tegra  | 
| Platform                                 | NVIDIA Jetson   |
| SDK Version                            |  v2.45.0                          | 
| Language                            |  C                    | 
| Segment			|  Robot- ROS v2.3.0             | 

### Issue Description

When I run realsense-ros there is a continuous log of control transfer error. When used with Realsense viewer, there is no rgb streaming too. I have attached the log for reference. The sdk was build from source with cuda support too.

... logging to /home/jetson/.ros/log/c5e65c10-7a88-11ec-868d-00044bc725a4/roslaunch-jetson-desktop-23476.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jetson-desktop:44513/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 60
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 90
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: False
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
[... 281 lines cut ...]
 21/01 15:25:25,220 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:25,280 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:25,340 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:25,400 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:25,461 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:25,521 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:25,580 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:25,641 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:25,701 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:25,762 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:25,821 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:25,882 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:25,942 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:26,002 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:26,063 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:26,122 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:26,183 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:26,243 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:26,303 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:26,363 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:26,423 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:26,484 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:26,543 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:26,603 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 15:25:26,664 ERROR [546024432000] (synthetic-stream.cpp:48) Exception was thrown during user processing callback:

MartyG-RealSense, 2022-01-21:
Hi @Kaijun101  I note that you have the pointcloud filter enabled in your RealSense ROS roslaunch log.   There is currently a known issue with using the pointcloud filter with Jetson boards specifically that can result in problems such as missing color or very low FPS.  These problems typically disappear if the pointcloud filter is not enabled.  https://github.com/IntelRealSense/realsense-ros/issues/1967 provides more information about this.

In regard to no RGB stream in the RealSense Viewer, are you receiving a **No Frames Received** message in the RGB stream panel?  If so, can you confirm that (1) you do not have the Motion Module enabled, and (2) you are not actively running a RealSense ROS launch at the same time as attempting to access the RGB stream in the Viewer.

Kaijun101, 2022-01-21:
Thanks @MartyG-RealSense for your reply.

Regarding the pointcloud, I have disabled point cloud using this command: roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
But the console still generating continuous stream of errors. Below are the run logs.

... logging to /home/jetson/.ros/log/37e86530-7aa0-11ec-b025-00044bc72037/roslaunch-jetson-desktop-24605.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jetson-desktop:40199/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
[... 196 lines cut ...]

    FORCE_RUSB_BACKEND

-- Build files have been written to: /home/jetson/librealsense/build

Realsense-viewer
=============
With regards to  realsense-viewer, I do not see the no RGB stream message and when only the RGB camera option is toggled this is the error message I received
 21/01 16:58:34,667 WARNING [548069695264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/01 16:58:34,919 WARNING [548069695264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/01 16:58:35,137 WARNING [548069695264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/01 16:58:35,189 WARNING [548069695264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/01 16:58:35,239 WARNING [548069695264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/01 16:58:35,290 WARNING [548069695264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/01 16:58:35,340 WARNING [548069695264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/01 16:58:36,937 ERROR [547742543648] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 16:58:36,961 ERROR [547742543648] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
[... cut ...]
 21/01 16:58:37,160 ERROR [547742543648] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 16:58:37,193 ERROR [547742543648] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 21/01 16:58:37,226 ERROR [547742543648] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_

MartyG-RealSense, 2022-01-21:
The log states that depth FPS is configured for 90 and color is configured for 60.  Can you try the roslaunch instruction below please, which sets depth and color to 640x480 and 30 FPS, to confirm whether or not a lower FPS improves stability, please.

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30**

Kaijun101, 2022-01-22:
When I run realsense-viewer, I still cannot view rgb stream. Below are the log when the depth stream and RGB stream are toggled on. I toggled the depth stream before toggle the RGB camera stream.

#When depth stream is toggled on

 22/01 13:47:15,556 WARNING [547926273824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/01 13:47:15,808 WARNING [547926273824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/01 13:47:15,859 WARNING [547926273824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/01 13:47:16,010 WARNING [547926273824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/01 13:47:16,060 WARNING [547926273824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

## When RGB Camera option is toggled on but no RGB video feed
 22/01 13:47:23,479 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:23,501 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:23,534 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:23,567 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:23,601 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:23,634 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:23,668 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:23,701 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:23,734 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:23,767 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:23,804 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:23,835 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:23,870 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:23,902 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:23,935 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:23,968 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:24,002 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:24,036 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:24,068 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:24,101 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:24,134 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:24,169 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:24,202 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 22/01 13:47:24,235 ERROR [547557941024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
[... cut ...]
 22/01 13:50:07,445 ERROR [546230198656] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
Done

Building of Realsense SDK
=====================
Could there be some issue with how the SDK is being built. Do you also have recommendations on how I can build SDK v2.45.0 with CUDA support, on Jetson TX2?

MartyG-RealSense, 2022-01-22:
You could build librealsense from source code with CMake on Jetson using the **RSUSB backend** installation method and include CUDA support by having the term **DBUILD_WITH_CUDA:bool=True** in the CMake build instruction.  https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 has Jetson installation instructions for an RSUSB build.

You can obtain the source code for librealsense 2.45.0 from the 'Assets' file list at the bottom of the version information listing at the link below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.45.0

[image]

Kaijun101, 2022-01-24:
### Issue Description
Hi @MartyG-RealSense, I am currently using auvidea j120 board with jetson TX2 and D435i.
However, there is some error i am currently facing when i am running realsense-viewer. When i build with cuda, my RGB is not working and that error shows up. It works fine without building with cuda.

<pre>ERROR [547423751968] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_</pre>

**Steps taken**
 $ sudo apt-get install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev
 $ sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
 $ git clone https://github.com/IntelRealSense/librealsense.git
 $ cd librealsense
 $ git checkout v2.45.0
 $ mkdir build && cd build
 $ cmake .. -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=ON -DBUILD_WITH_CUDA=true
 $ sudo make uninstall && make clean
 $ time make -j8 && sudo make install
 
 **Below is the image of the realsense-viewer for your references**
[image]

### Issue Description
I am currently using auvidea j120 board with jetson TX2 and D435i.
However, there is some error i am currently facing when i am running realsense-viewer. The D435i camera does not have RGB image and is showing this error.

<pre>ERROR [547423751968] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_</pre>

**Steps taken**
 $ sudo apt-get install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev
 $ sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
 $ git clone https://github.com/IntelRealSense/librealsense.git
 $ cd librealsense
 $ git checkout v2.45.0
 $ mkdir build && cd build
 $ cmake .. -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=ON -DBUILD_WITH_CUDA=true
 $ sudo make uninstall && make clean
 $ time make -j8 && sudo make install
 
 **Below is the image of the realsense-viewer for your references**
[image]

### Issue Description
I am currently using auvidea j120 board with jetson TX2 and D435i.
However, there is some error i am currently facing when i am running realsense-viewer. The D435i camera does not have RGB image and is showing this error.

<pre>ERROR [547423751968] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_</pre>

**Steps taken**
 $ sudo apt-get install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev
 $ sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
 $ git clone https://github.com/IntelRealSense/librealsense.git
 $ cd librealsense
 $ git checkout v2.45.0
 $ mkdir build && cd build
 $ cmake .. -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=ON -DBUILD_WITH_CUDA=true
 $ sudo make uninstall && make clean
 $ time make -j8 && sudo make install
 
 **Below is the image of the realsense-viewer for your references**
[image]

When i open the realsense-viewer, they prompt me to update the firmware to 5.13.0.50. My current version is 05.12.13.50. Will there be any issue if i update and use a older version of realsense-viwer?

MartyG-RealSense, 2022-01-24:
https://github.com/jetsonhacks/buildLibrealsense2TX/issues/13#issuecomment-573976359 suggests making an edit to the /bashrc file and then reloading the terminal if problems are occurring on Jetson when CUDA support is enabled.

Using an older version of the RealSense SDK / RealSense Viewer should be fine, up to a point.  If you have firmware **5.12.8.200** or newer installed then the minimum SDK version used should be 2.39.0, due to internal changes in the firmware from firmware 5.12.8.200 onwards.  

If you anticipate needing to use SDK / Viewer versions older than 2.39.0 then firmware **5.12.7.100** should be used, as this is the version before the internal firmware changes were introduced that made 2.39.0 the minimum SDK.

Kaijun101, 2022-01-24:
Hi @MartyG-RealSense, thanks for all your help so far but i am currently still unable to successfully build using cuda 10.2 using the above suggested information.

Below is the list of the full build log that i did which i hope would be beneficial in debugging this issue that I am facing.

**Steps taken (After updating bashrc)**
 $ sudo apt-get install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev
 $ sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
 $ git clone https://github.com/IntelRealSense/librealsense.git
 $ cd librealsense
 $ git checkout v2.45.0
 $ mkdir build && cd build
 $ cmake .. -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=ON -DBUILD_WITH_CUDA=true
 $ sudo make uninstall && make clean
 $ time make -j8 && sudo make install

**Full build log file of the above mention build**
[Build_log.docx](https://github.com/IntelRealSense/librealsense/files/7927712/Build_log.docx)

**Same error faced**
<pre>ERROR [547423751968] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_</pre>

MartyG-RealSense, 2022-01-24:
I researched the issue further and found another RealSense user who experienced the same error message at https://github.com/IntelRealSense/realsense-ros/issues/1604#issuecomment-789562010

In that same case that user later commented, "It got addressed by adding more swap memory.  As it turned out it was due to CUDA memory allocation error.  It seems without extra swap it was possible to get only part of topics published".

A **JetsonHacks** guide at the link below provides advice about increasing swap memory for CUDA to use.

https://www.jetsonhacks.com/2019/11/28/jetson-nano-even-more-swap/

Kaijun101, 2022-01-25:
I tried increasing the swap to 8gb and rebuilding it but the same error pops out too.

MartyG-RealSense, 2022-01-25:
Given that the error message references intrinsics, let's try resetting the camera to its default factory-new calibration settings in the RealSense Viewer using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

Kaijun101, 2022-01-26:
Tried with this method, but still got the same error. This issues seems to persistent when building with cuda on a TX2 Jetson.

MartyG-RealSense, 2022-01-26:
The CUDA support in librealsense provides Nvidia graphics GPU acceleration for a limited range of functions (color conversion, pointcloud and align).  If your particular application is not dependent on depth-color alignment or pointcloud generation then you may be able to build librealsense without CUDA support and not see a significant difference in performance.

Kaijun101, 2022-01-28:
Hi, @MartyG-RealSense, do you have a list of fucntions that can benefit fro, cuda support?

MartyG-RealSense, 2022-01-28:
The list mentioned above is basically everything supported - alignment, pointcloud and color conversion.  Further technical information about the CUDA support for these functions can be found at te links below.

https://github.com/IntelRealSense/librealsense/pull/1866
https://github.com/IntelRealSense/librealsense/pull/2670

MartyG-RealSense, 2022-02-03:
Hi @Kaijun101  Do you require further assistance with this case, please?  Thanks!

Kaijun101, 2022-02-04:
Hi @MartyG-RealSense, thanks for assisting me on this issue, we manage to resolve the issue. It turn out to be a native cuda issue.

closed 2022-02-04 by Kaijun101

MartyG-RealSense, 2022-02-04:
You are very welcome.  It's great to hear that you achieved a solution.  Thanks for the update!

afonsofonbraga, 2022-02-21:
@Kaijun101 Hey, I believe I have the same issue as you had. I'm using an Advantech PC based on Tegra TX2 with the CUDA 10.2 as well. I also tried different resolutions with realsense-viewer but still getting the:
```
(synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
```
Could you please share your solution so I can fix it too?

lamiazain, 2022-02-28:
@afonsofonbraga I have same problem Jetson nano & CUDA10.0 Jetpack 4.3. If you @Kaijun101 can provide us with the solution, this would be very helpful.

Kaijun101, 2022-03-01:
Hi @lamiazain @afonsofonbraga , when i was debugging i notice that my CUDA was not build properly on my jetson TX2. One way you can check is to build realsense without cuda first and see if this error pops up. If there is no more error most likely you did not link your CUDA in  /.basrc file or your CUDA is not build properly on your device.

MartyG-RealSense, 2022-03-01:
Thanks so much @Kaijun101 for providing advice to @lamiazain   :)

afonsofonbraga, 2022-03-01:
@Kaijun101 Thanks for the reply! I'll test it later :smile: