realsenseai/librealsense #10177: Poor Framerate Performance on Raspberry Pi 4

yusufcc, 2022-01-21:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D430 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Raspbian GNU/Linux 10 (buster) | 
| Kernel Version (Linux Only)    | 5.10.63-v7l+                                        | 
| Platform                                 | Raspberry Pi 4B 2GB |
| SDK Version                            |   2.50.0                           | 
| Language                            |  Realsense Viewer / Python                 | 

### Issue Description
I can not receive more than 15 FPS at 640x480 Depth and IR stream. I tested this issue with 2 different Raspberry Pi 4B 2GB. 

I installed SDK with `-DFORCE_RSUSB_BACKEND=ON -DCMAKE_BUILD_TYPE=Release` flags. 

When I tested the same sensor on my Ubuntu Desktop PC, I can get higher performance. But I've read that some users can stably get 30 FPS at 1280x720 resolution on Raspberry Pi 4B. 

According to your observations, how is the FPS performance on Raspberry Pi 4B?

MartyG-RealSense, 2022-01-21:
Hi @yusufcc  The original Raspberry Pi 4 had 4 GB of memory.  So it may be worth first trying to increase the size of your 2 GB Pi's **swapfile**.  Once a Pi application has used up the real memory capacity then it switches to using slower virtual memory from the drive storage space. 

The **Acrobotic** RealSense installation guide for Pi 4 provides information about configuring the swapfile size.  

https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

Instead of setting it to '2048' as the guide suggests, you could try '4096' to account for the memory capacity difference between your Pi and the original 4 GB model.

yusufcc, 2022-01-21:
Unfortunately, increasing swap size did not fix the problem. I had tested with Raspberry Pi 4B 8GB before, but I did not mention it above because I did the SDK 2.41.0 installation with `-DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_LIBUVC=true`  at that time. And there was no significant difference.

MartyG-RealSense, 2022-01-21:
If you are testing with the RealSense Viewer, are you setting the depth and infrared streams to 640x480 and 30 FPS but are only getting an FPS readout of 15 for both streams if you overlay the FPS information with the 'i' button?

Or does the Viewer not allow you to set 30 FPS in the drop-down selector?

[image]

MartyG-RealSense, 2022-01-21:
The **Frame Drops Per Second** graph is useful as a rough indicator but is not as useful as a diagnosis of FPS performance, since frame drops can be reported even if FPS is not being impacted, as described by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/7488#issuecomment-704124850

I note in your image that auto-exposure is disabled.  If you are using manual exposure then using certain ranges of exposure value can cause a drag on FPS.  Another RealSense team member describes this at https://github.com/IntelRealSense/librealsense/issues/1957#issuecomment-400715579 and states that use of manual exposure can affect depth performance, especially if the camera is used in low-light conditions. In such a situation, they suggest increasing gain to improve performance.

closed 2022-02-03 by MartyG-RealSense