realsenseai/librealsense #10179: External Trigger using D435/D455

asb2111991, 2022-01-23:
I am trying to find any information/tutorial on triggering an external light source using the D435 or D455. I want to trigger an external light source and turn it On or OFF before taking images. Any help is much appreciated. Thank you.

MartyG-RealSense, 2022-01-23:
Hi @asb2111991  A useful reference may be the information in https://github.com/IntelRealSense/librealsense/issues/4574 for controlling an external RGB flash with the camera sync pin's pulses.

asb2111991, 2022-01-23:
Thank you @MartyG-RealSense . That certainly helps. I will try to use this information and will comeback if I have any further questions.

MartyG-RealSense, 2022-02-05:
Hi @asb2111991  Do you require further assistance with this case, please?  Thanks!

asb2111991, 2022-02-10:
Hello @MartyG-RealSense , Can you please point me to any example/source that has information on which pins to connect for PNP or NPN triggering and the code that lets me assign the trigger timing. Thank you.

MartyG-RealSense, 2022-02-11:
Typically, Pin 5 (the **Sync** pin) of the camera's hardware sync pin set is used for triggering.

Page 53 of the current edition of the data sheet document for the 400 Series cameras provides a list of hardware sync pin functions, shown in the image extract below.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

[image]

In your particular application, the RealSense **External Synchronization (Genlock)** system may be the appropriate tool to use.  It provides fine control over trigger timing and it will wait **indefinitely** for a trigger signal to be received before activating sync.  

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-2-principles-of-operation

It can also be used with an external **signal generator** device instead of having a camera generate the trigger pulse.

[image]

asb2111991, 2022-02-11:
Thank you @MartyG-RealSense. Examples in Python will be really helpful

asb2111991, 2022-02-11:
And should I have an external signal generator for this to work?

asb2111991, 2022-02-11:
I have made all the connections and with the Output Trigger Enabled, I cannot see any difference in the voltage levels of pin 5 relative to the ground. Also, checking the voltage level on pin 8, I did not see 3.3V at any point, it was almost next to zero. Please help I seem to miss something. All I am looking for is to turn the flash on when I am trying to grab a frame. I am only working with images and not videos.

MartyG-RealSense, 2022-02-12:
https://github.com/IntelRealSense/librealsense/issues/8313 has an example of Python code for setting the **Inter Cam Sync Mode** value to define the camera's sync behavior.

[image]

If you are aiming to have an external flash activate when there is a frame capture then I am thinking that the appropriate setup may be to set the RealSense camera's 'Inter Cam Sync Mode' to a vlaue of '1' so that the camera acts as the **Master** in control of the triggering process (i.e you are not using an external signal generator) and emits a trigger pulse from pin 5 when a frame capture takes place.

A wire would be attached between pin 5 of the camera and the external flash so that the trigger pulse travels outward from the camera along the cable.  A 400 Series camera will only transmit a trigger pulse from pin 5 if its Inter Cam Sync Mode is set to '1'.

The voltage of the pulse needs to be sufficient for the flash to be able to react to it.  Pin 5 will be 1.8 V, so - as described in https://github.com/IntelRealSense/librealsense/issues/4574 - a **voltage shifter** can be used to increase the voltage to 5V (or whatever voltage your particular flash controller requires).

asb2111991, 2022-02-13:
Hi @MartyG-RealSense, Thank you for the continued support. Unfortunately, I was not able to trigger the external device using the code or RealSense Viewer. Please correct me if I am wrong. My understanding from other issues and your comments is that when I set the 'Inter Cam Sync Mode' to '1' and then turn on the 'Output Trigger Enabled' checkbox, the voltage on Pin 5 raises to 1.8V with the depth module triggering, stays there for a little over 100us and falls back to 0 relative to Pin 9/GND.

Now, for Pin 8, should I supply it with 3.3V or should I expect 3.3V from it? In case I have to expect it to deliver 3.3V, I am not able to read the 3.3V on the meter. Please let me know if I am going in the right direction. Thank you.

MartyG-RealSense, 2022-02-13:
Pin 5, which is 1.8 V, will rise and remain "high" for > 100us and fall back to 0, yes.

Typically, Pin 8 is not involved in sync cabling.  The wires go to Pin 5 and Pin 9.  This can be seen in the sync cable blueprint in the Genlock paper, where each pin except 5 and 9 has an 'x' beside it.  In electrical circuits in general, a connection to 'ground' is required to be made in order for electricity to be able to flow.

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-3-camera-connections

[image]

You can see such a 2-wire Pin 5 and Pin 9 connection in the image below from the Genlock paper of a sync cable.

[image]

So my expectation would be that if Pin 9 is grounded then you should be able to measure 1.8 V from Pin 5 when a trigger pulse is released.

Here is a real-world example of sync cabling to 5 and 9 that a RealSense user created.

[image]

In their particular case, the cable from the Master camera lead to the hardware sync pin set of another camera that was a Slave, whilst in your application you are aiming to send the pulse to an RGB flash.

[image]

At this point, I believe that the next goal should be to see whether you can read 1.8 V from Pin 5, and then afterwards work out how to interface that trigger pulse with with the external flash that will be receiving it.

asb2111991, 2022-02-18:
I followed the comments and found tested for the 1.8V pulse by varying the FPS. As the RealSense Viewer lets me increase the FPS upto 300 frames, I varied the FPS from the min value (5 FPS) to the max (300 FPS) and could see the increase in the voltage read on Pin 5 from the multimeter as the number of ulses increase with FPS. So, now, I have to find the right level shifter to make the 1.8 V read as 5 V. I will try to get that board and will post my updates here.

Thank you @MartyG-RealSense for all your help

roelofvandijk, 2022-02-19:
For future users, also take a look at https://github.com/IntelRealSense/librealsense/issues/3084#issuecomment-465825421 for video examples of the correct settings for the external triggering in the Realsense Viewer.

Here you can find good documentation on making trigger cables and purchasing the required parts:
- https://vcl3d.github.io/VolumetricCapture/docs/hardware/rs2_hardware/ 
- https://www.youtube.com/watch?v=E3PzERRZgHU <- example video of making cables

@MartyG-RealSense, maybe some of that information can make it into the Intel documentation?

The settings you want are RS2_OPTION_OUTPUT_TRIGGER_ENABLED to 1 (enabled) and RS2_OPTION_INTER_CAM_SYNC_MODE to 1 (master).

https://intelrealsense.github.io/librealsense/doxygen/rs__option_8h.html#a8b9c011f705cfab20c7eaaa7a26040e2ad774d368144580c95c5b26e870e9c2db

In Python it should be something like this:
```
device.set_option(rs.option.inter_cam_sync_mode, 1)  # 1 means master mode
device.set_option(rs.option.output_trigger_enabled, 1)  # 1 means enabled
```

In C++
```
depth_sensor.set_option(RS2_OPTION_INTER_CAM_SYNC_MODE, 1.f);
depth_sensor.set_option(RS2_OPTION_OUTPUT_TRIGGER_ENABLED, 1.f);
```

You can keep the sync mode at 1 (master) and turn the trigger on/off while the sensor is streaming, just like you would with the checkbox in the Realsense Viewer.

Furthermore, the trigger time might not be what you think it is. There are some important distinctions between models, as D415 does not support Genlock (because of the rolling shutter):
- Genlock (mode 4): Optical capture starts with Trigger.
- Slave mode (modes 2 and 3): Readout (not capture) starts with Trigger.

See this image for an illustration of the Genlock timing, for the slave mode, the trigger is shifter to the right to the readout step!

[image]

This means to get correct lamp synchronization working, you might have to put a delay on your signal to correct triggering, dependent on the Realsense model you are using.

For details, see the table at the bottom of this page:
https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-3-camera-connections

MartyG-RealSense, 2022-02-20:
Thanks so much @roelofvandijk for your very detailed feedback!  I have added a **Documentation** label to this case so that it can be reviewed at a future date by a member of the RealSense team responsible for documentation.  This case should therefore be kept open.

closed 2026-10-23 by MartyG-RealSense