realsenseai/librealsense #10181: How to get a .ply file of live stream of the INtelrealsense D435i camera?

Abirami-Segar, 2022-01-23:
I am using ubuntu 18.04. I want to get .ply for a live feed of streams. I want to use them in MATLAB. I am only able to get .ply using the GUI(by tying "realsense-viewer" in terminal) Other methods involving c++(even after including few commands mentioned by few users) and python dont work?

MartyG-RealSense, 2022-01-23:
Hi @Abirami-Segar  If you are using the RealSense MATLAB wrapper then its programming language is almost 1:1 the same as C++ so any MATLAB wrapper script will be based around the structure of C++.

An example of code for saving pointcloud data as a ply file in the MATLAB wrapper is at https://github.com/IntelRealSense/librealsense/issues/4232#issuecomment-503775364 (that code will of course be similar to C++).

Abirami-Segar, 2022-01-24:
As far as I know, interfacing the realsense camera with MATLAB/Simulink on the ubuntu platform is not possible. IT could only be achieved in windows. Kindly correct me if i am wrong

MartyG-RealSense, 2022-01-24:
If you are referring to using the RealSense MATLAB wrapper that is normally for Windows 10 then although the wrapper is designed for Windows 10, there was someone in the link below who got it working with Ubuntu.  The section of the page where an installation on Ubuntu was successfully achieved is under the **December 4, 2018** heading.

https://staff.fnwi.uva.nl/a.visser/activities/IMAV/Labbook2018.html

Abirami-Segar, 2022-01-24:
I stream images from the camera to simulink via ROS. I want to stitch few point clouds as recorded by the camera for my project. In order to do that i need the point clouds as .ply files. So, I was wondering if i could publish these messages(ply) to simulink

closed 2022-02-05 by MartyG-RealSense