realsenseai/librealsense #10182: D435i Depth System Broken after Dynamic Calibration

ridanlukita, 2022-01-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.13.00.50 | 
| Operating System & Version |   Linux (Ubuntu 20.04 LTS)  | 
| Kernel Version (Linux Only)    |  5.11.0-43-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.50                          | 
| Language                            |  -                          | 
| Segment			|  others                           | 

### Issue Description
After I do Dynamic Calibration (v2.11.1.0), depth reading broken to become like this
[image]

This is the first time I encountered like this, usually calibration just works normally.

I have done:
- Hardware reset
- Updating firmware to latest version
- On-chip, focal length, tare calibration (all failed)
[image]

When I try to use Dynamic Calibration again, the blue area always reappear and therefore I cannot pass the rectification phase.
[image]
[image]

Is there any suggestion to solve this problem?

MartyG-RealSense, 2022-01-24:
Hi @ridanlukita  Do you see an improvement if you follow the steps below to reset the calibration to factory-new defaults, please?

1.  Enable the depth stream.

2.  Go to the **More** option in the Viewer and select the **Calibration Data** menu option.  

[image]

3.  When the 'Camera Calibration' window pops up, left-click on the box labeled 'I know what I'm doing' in the bottom corner of the window.  This unlocks the **Restore Factory** button, making it selectable.  When 'Restoe Factory is clicked, it will reset the calibration values of the camera to their factory-new defaults.

[image]

4.  Finally, left-click on the **Write Table** button to write the reset calibration values to the camera hardware's storage and automatically close the Camera Calibration window.

ridanlukita, 2022-01-25:
Now it has been back to normal! The more I learn about this device, the more I realized there's enormous amazing physics applied behind it! Thanks a lot! @MartyG-RealSense

closed 2022-01-25 by ridanlukita

MartyG-RealSense, 2022-01-25:
You are very welcome, @ridanlukita - it's excellent news that your camera has been corrected.  Thanks very much for the update. :)