realsenseai/librealsense #10184: pyRealSense: Load JSON preset and record to file

danfleck, 2022-01-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   MacOS 10.15.7  | 
| Platform                                 | Mac (PC)  |
| SDK Version                            |  2.49.0                       | 
| Language                            |  python                           | 
| Segment			|  others                          | 

### Issue Description
When using pyRealSense to try and load JSON presets and record, I receive an error:

**RuntimeError: null pointer passed for argument "dev"**

This occurs when trying to load the JSON using code similar to the DeviceManager code. This code:
```
        device = pipeline_profile.get_device()
        advanced_mode = rs.rs400_advanced_mode(device)
        advanced_mode.load_json(json_text)
```
This issue seems very similar to: https://github.com/IntelRealSense/librealsense/issues/6739 except I'm using python instead of C++.

If I comment out the line:        `rs_config.enable_record_to_file(filename)`
everything works fine.

If I comment out the `advanced_mode.load_json(json_text)` call above, it also works fine. With both the load_json and the enable_record_to_file the error appears. I'm trying to load ShortRangePreset.json if that helps.

MartyG-RealSense, 2022-01-24:
Hi @danfleck   https://github.com/IntelRealSense/librealsense/issues/8868#issuecomment-823314199 may be a helpful Python json loading reference to compare to your own code.

danfleck, 2022-01-25:
Thanks Marty. I'm not having problems reading the JSON file but I tried the different code you linked in to make sure and it didn't make a difference. 

If I remove the line:
`rs_config.enable_record_to_file(filename)`
everything works fine. 

Do you have an example that reads a JSON config file and enables recording to a file? That is where the problem is I think.

Thanks for your help!

MartyG-RealSense, 2022-01-25:
May I confirm that you are placing the **rs_config.enable_record_to_file** instruction on the line before the pipe start instruction and not after the pipe start, please?

danfleck, 2022-01-25:
Yes. My code looks like this:
 ```
        # Where should we record to file?
        rs_config.enable_record_to_file(filename)

        # Enable the streams from all the intel realsense devices
        rs_config.enable_stream(rs.stream.depth, self.depth_resolution_width,
                                self.depth_resolution_height, rs.format.z16, self.frame_rate)
        rs_config.enable_stream(rs.stream.infrared, 1, self.depth_resolution_width,
                                self.depth_resolution_height, rs.format.y8, self.frame_rate)
       
        # Enable the device
        self.pipeline = rs.pipeline()
        self.pipeline_profile = self.pipeline.start(rs_config)
```

MartyG-RealSense, 2022-01-25:
Does it work if you specify a filename to record to instead of using the **filename** string.  For example:

`rs_config.enable_record_to_file('test.bag')`

danfleck, 2022-01-25:
No, I tried it and get the same error. 

I am using it successfully without loading the JSON presets now, so I think everything is working correctly (no null filename). However, the camera settings aren't great because the JSON file isn't loaded.

MartyG-RealSense, 2022-01-25:
It may be worth exporting a json from the RealSense Viewer and loading the filename into your script.  I suggest this because if you are testing with the ShortRangePreset.json file from the 400 Series visual presets page, I have had errors with loading downloaded json links from that page into the RealSense Viewer recently.  

You could see whether your test json file is okay by attempting to load it into the Viewer with the json load option near the top of the Viewer side-panel (adjacent to the json drop-down selector menu) and see whether it can be loaded into the Viewer successfully.

danfleck, 2022-01-25:
Here is a simplified (mostly) example of the problem. If you comment out either 

`rs_config.enable_record_to_file(filename)` 
or 
`self.load_settings_json(self.pipeline_profile, "../ShortRangePreset.json")`

it works. Otherwise, it fails with the error. (Change the extension back to .py)

[ErrorRecordingExample.py.txt](https://github.com/IntelRealSense/librealsense/files/7934800/ErrorRecordingExample.py.txt)

danfleck, 2022-01-25:
Sorry, you don't need my rs_utils import either. Here is a version with that commented out
[ErrorRecordingExample.py.txt](https://github.com/IntelRealSense/librealsense/files/7934830/ErrorRecordingExample.py.txt)
.

danfleck, 2022-01-25:
I actually can't load any JSON presets into Realsense-viewer actually because of a different problem. I'm on a Mac and all the files are grayed out when using the file chooser from realsense-viewer. I can navigate directories, but I can't load any files because I can't select them. (This is a problem for a different problem report though. The JSON seems to load fine in my code as long as the enable_write_to_file is commented out.)

MartyG-RealSense, 2022-01-25:
Another approach to setting a json would be to load one of the pre-defined RealSense presets that are built into the SDK.  These can be loaded using a different method to the procedure that loads a custom-defined json, so this might allow a json and record_to_file to co-exist in your script at the same time.  A Python script for this is at https://github.com/IntelRealSense/librealsense/issues/2577#issuecomment-432137634

danfleck, 2022-01-25:
Thanks Marty. That does work I can set it using the other approach with built-in presets.

It still seems like a bug though that I can't load the JSON presets file. Were you able to confirm that with the code I posted? 

I'll try and use the other presets but the short range JSON file seemed to be the best for my application.

MartyG-RealSense, 2022-01-25:
I am not able to test code myself but it is not the first case over the years in which I have seen such conflicts.  

I would recommend the **Medium Density** preset for a wide range of applications, as it provides a good balance between accuracy and the amount of detail on the image, whereas High Accuracy tends to over-strip the imag eof detail.

danfleck, 2022-01-25:
Thanks Marty! The only preset names I see are:
        00: "Custom"
        01: "Default"
        02: "Hand"
        03: "High Accuracy"
        04: "High Density"

MartyG-RealSense, 2022-01-25:
Medium Density is one of the built-in presets.  My understanding is that the presets are enumerated in the SDK in the order that they are displayed in the RealSense Viewer's preset list, so as Medium Density is the 6th in the list, it may be '05'

[image]

MartyG-RealSense, 2022-01-31:
Hi @danfleck  Do you require further assistance with this case, please?  Thanks!

danfleck, 2022-01-31:
Hi @MartyG-RealSense,

The workaround you suggested is working for now, but I was hoping someone at Intel could confirm the bug and make a determination if it was something that was actually going to be fixed. I still can't use the JSON file while recording to a file.

If it's not going to be fixed though let me know and I can just close it.

Thanks for your help!

MartyG-RealSense, 2022-01-31:
In my opinion it is unlikely that this particular issue will be investigated by Intel for the possibility of a bug fix.

closed 2022-01-31 by danfleck