realsenseai/librealsense #10185: Reading xyz of point cloud. Intelrealsense, Ubuntu 18.04, D435i

sbrgrsh, 2022-01-25:
Hi 
 I want to generate a point cloud reference model of an object using (https://de.mathworks.com/help/vision/ref/pointcloud.html) .This basically boils down to generating xyz points of the object. I found a similar issue (#9823) I wish to know how to get similar results using a D435i running on a ubuntu 18.04 platform. Please guide through the steps.

MartyG-RealSense, 2022-01-25:
Hi @sbrgrsh  If your preference would be to generate the pointcloud directly in MATLAB without using RealSense SDK scripting, there is a tool available for MATLAB that converts a **ply** pointcloud file to a MAT file with XYZ coordinates and RGB color channels.

https://uk.mathworks.com/matlabcentral/fileexchange/54041-plytomat

A ply file can be generated in the pre-made **RealSense Viewer** software by exporting from its 3D pointcloud mode.

*********

Whilst the RealSense SDK has a MATLAB compatibility' wrapper' that can be used to write programs that interface with MATLAB, it is designed for use with Windows 10.  There was one article where someone got it working on Ubuntu and shared their method for creating librealsense_mex.mexa64. At the link below, you can find the details under the December 4, 2018 heading.

https://staff.fnwi.uva.nl/a.visser/activities/IMAV/Labbook2018.html

If you decide to try to get the MATLAB wrapper working in Ubuntu and succeed then I can assist in providing wrapper script references for obtaining a pointcloud with xyz through the wrapper.

sbrgrsh, 2022-01-25:
I want to generate pointcloud for one particular object, neglecting the background or any other object if at all it is captured by the camera. I have tried to directly export the .ply file from realsense-viewer to MATLAB but I never was able to get rid of the background. So I thought if it was possible to generate point cloud by getting all of the XYZ points associated with the object and get a point cloud like this (https://github.com/IntelRealSense/librealsense/issues/9823#issuecomment-1003527761)

MartyG-RealSense, 2022-01-25:
If the post-processing **Threshold Filter** of the Viewer was used then you could reduce the maximum observable depth distance of the camera (defined wih the filter's **Max Distance** setting) and so exclude the background from the live image before exporting the ply.

[image]

MartyG-RealSense, 2022-01-25:
https://github.com/IntelRealSense/librealsense/issues/8368#issuecomment-828631563 suggests some approaches about ways to stitch pointclouds together into a single combined cloud from within MATLAB.

Outside of MATLAB, if you are able to use OpenCV and C++ programming language then a RealSense SDK tool called **rs-kinfu** may be helpful.  You can move a single camera around an object and progressively build up a pointcloud in realtime instead of having to stitch multiple views and then export the image as a ply.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/kinfu

sbrgrsh, 2022-01-26:
Hi 

In the link you provided,  https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/kinfu , the official documentation links are not active. Kindly do update the links. I am a beginner and I need those links.

MartyG-RealSense, 2022-01-26:
It looks as though the OpenCV official documentation site no longer has kinfu instructions for OpenCV 3.4 but does have them for OpenCV 4.  

https://docs.opencv.org/4.0.0/d8/d1f/classcv_1_1kinfu_1_1KinFu.html

I have added a **Documentation** label to this case to highlight it to RealSense team members who handle documentation to review it at a future date.  This case should therefore be kept open.  Thanks for the report!

closed 2026-10-23 by MartyG-RealSense