realsenseai/librealsense #10186: Calibrate D435i?

w407022008, 2022-01-25:
Hi,
As my d435i was not always producing the right depth measurement, I referred to [this video](https://www.youtube.com/watch?v=ZSws9GDdflQ) and used [the software](https://www.intel.com/content/www/us/en/download/645988/intel-realsense-d400-series-dynamic-calibration-tool.html?) to try to calibrate it. However, I was never able to get the same results as in the video, as the video showing below.  What should a correct calibration look like? Do you have more detailed instructions on how to do this? I am unable to complete this step according to the pdf given.

https://user-images.githubusercontent.com/26375711/151032602-8ba8dd31-0fd4-4f21-ad73-3a09c2e03ef5.mp4

MartyG-RealSense, 2022-01-25:
Hi @w407022008  Did you print the calibration target image out on **US Letter** size paper?  The target PDF document is formatted in US Letter dimensions.  A RealSense user at https://github.com/IntelRealSense/librealsense/issues/10074 found that they had sub-par calibration results if they printed the target onto A4 size paper used by countries such as the United Kingdom and Australia.

The Dynamic Calibration tool offers the feature of displaying the target as a digital image on the screen of an Android device instead of printing and mounting it.

***************

The RealSense SDK has an alternative calibration system that you can try called **On-Chip**.  It is accessible from the **More** option in the RealSense Viewer program, near the top of the Viewer's options side-panel.  Instead of pointing the camera at a paper or Android target, the camera is pointed at a plain wall.  Intel have published a guide for this tool at the link below.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

The On-Chip tool provides a 'health check" score that describes the quality of the calibration, such as good or could be improved.  You can then re-perform the calibration until you get a result that you are satisifed with and finally write that calibration to the camera hardware.

***************

A third option would be to reset the camera to its default factory-new calibration settings in the RealSense Viewer using instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

w407022008, 2022-01-25:
Hi, @MartyG-RealSense . Thank you, this is a very comprehensive guide. For the first one. No, I printed it on A4 paper,, and I chose the original size to print it as requested. I then also measured the black and white grids to match the size exactly. I will try the calibration on- chip according to this link and maybe it will solve the problem I am facing. I will close this issue when the calibration is successful. thank you again.

MartyG-RealSense, 2022-01-25:
You are very welcome.  Please do report back with your results after tests.  Good luck!

w407022008, 2022-01-25:
Thanks @MartyG-RealSense . By the way, I am using the d435i to implement ORBSLAM on my robot. I would like to ask if you know how to get all the parameters needed in the _yaml_ file of ORBSLAM in a complete way?

MartyG-RealSense, 2022-01-25:
A RealSense user who was using ORB-SLAM2 with a D435 shared the details of their yaml file at https://github.com/IntelRealSense/librealsense/issues/2580

You can retrieve the full calibration details of your own D435i camera's intrinsics and extrinsics with the RealSense SDK's **rs-enumerate-devices** tool by launching it in calibration information mode with the command below:

**rs-enumerate-devices -c**

MartyG-RealSense, 2022-01-25:
Some RealSense users use a calibration tool called **Kalibr** for detailed D435i IMU information.

https://github.com/ethz-asl/kalibr

The official data sheet document for the Bosch IMU component used in the D435i is here:

https://www.bosch-sensortec.com/products/motion-sensors/imus/bmi055/

closed 2022-02-07 by MartyG-RealSense