realsenseai/librealsense #10187: GL_INVALID_VALUE

EzEsh44, 2022-01-26:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:572

```
I stepped away from issue #9993 and it lapsed. I updated to the 25.0 release with no luck.

Thanks in advance

EzEsh44, 2022-01-26:
I forgot to add that I am using a d435 camera

MartyG-RealSense, 2022-01-26:
Hi @EzEsh44  Did you install the RealSense SDK using the automated installer file in the **Assets** list of the SDK 'Releases'page, please?

https://github.com/IntelRealSense/librealsense/releases

[image]

EzEsh44, 2022-01-26:
Yes, that was installed when getting the above error

MartyG-RealSense, 2022-01-27:
As you installed the full SDK with the installer from the Releases page, you should be able to access its pre-made example programs on Windows.  Could you use the instructions at https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656 to run an example called **rs-capture.exe** and let me know whether it works, please?

If rs-capture is able to run successfully then it would suggest that the problem is not related to **OpenGL** graphics though (both the Viewer and rs-capture require OpenGL support to run).  If rs-capture can run and access the camea, this would also indicate that the Windows privacy settings are not blocking access to the camera hardware or camera applications.

EzEsh44, 2022-01-28:
rs-capture is able to run and show an RGBD image

helcl42, 2022-01-31:
Hello, I have the same issue(the same platform, version, camera), which is still present even though the standalone Intel.RealSense.Viewer.exe is used.

MartyG-RealSense, 2022-01-31:
Hi @helcl42  Have you checked that the **rs-capture.exe** program works correctly on your particular Windows computer, as described in https://github.com/IntelRealSense/librealsense/issues/10187#issuecomment-1023058467

EzEsh44, 2022-01-31:
When running the Intel.RealSense.Viewer.exe file from the 'Assets' file list I get an access violation error
[image]

MartyG-RealSense, 2022-01-31:
@EzEsh44   As you are using Windows, please check in the Windows camera privacy settings that access to the camera hardware and permission for applications to access the camera are both set to On.  Instructions for doing so are at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

helcl42, 2022-01-31:
> Hi @helcl42 Have you checked that the **rs-capture.exe** program works correctly on your particular Windows computer, as described in [#10187 (comment)](https://github.com/IntelRealSense/librealsense/issues/10187#issuecomment-1023058467)

Yes, the rs-capture thing works great.

MartyG-RealSense, 2022-01-31:
@helcl42  If rs-capture works correctly then that indicates that the problem is not related to the OpenCV graphics drivers on your computer.

Which camera firmware driver version is your D435 using, please?

helcl42, 2022-01-31:
> @helcl42 If rs-capture works correctly then that indicates that the problem is not related to the OpenCV graphics drivers on your computer.
> 
> Which camera firmware driver version is your D435 using, please?

Here it is: 1) Name: Intel RealSense D435, serial number: 817612071274, update serial number: 814113021582, firmware version: 05.12.15.50, USB type: 3.0

MartyG-RealSense, 2022-01-31:
If you are using SDK 2.50.0 with firmware 05.12.15.50 then there should not be a problem with that combination.

Next, please try downloading the older 2.49.0 version of the standalone **Intel.RealSense.Viewer.exe** Viewer to check whether the problem occurs on that Viewer version too.  

https://github.com/IntelRealSense/librealsense/releases/tag/v2.49.0

Remember that once the Viewer .exe file is downloaded, it should be launched with a double-leftclick from the folder on your computer that you downloaded it to and not with the launch shortcut icon on the Windows desktop.

helcl42, 2022-01-31:
> If you are using SDK 2.50.0 with firmware 05.12.15.50 then there should not be a problem with that combination.
>
> Next, please try downloading the older 2.49.0 version of the standalone [...]

Sure I have tested both, the 2.49 does not seem to run either.

Version 2.49 says:
[image]

Version 2.50 says:
[image]

MartyG-RealSense, 2022-01-31:
@helcl42   And the similarly designed **Depth Quality Tool** produces the same errors when launched?

helcl42, 2022-01-31:
> @helcl42 And the similarly designed **Depth Quality Tool** produces the same errors when launched?

Surprisingly it does not, it just quietly terminates without any message box.

MartyG-RealSense, 2022-01-31:
Thanks very much for your patience.  The **SEH Exception** error can be difficult to resolve unfortunately, and some solutions work for some people but not for others.

For example, in https://github.com/IntelRealSense/librealsense/issues/9885 a RealSense user solved it by installing the even older **2.48.0** version.  But that same advice did not work for another person with an SEH Exception.

EzEsh44, 2022-01-31:
> @EzEsh44 As you are using Windows, please check in the Windows camera privacy settings that access to the camera hardware and permission for applications to access the camera are both set to On. Instructions for doing so are at [#3381 (comment)](https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789)

Privacy settings are on for the camera and for the realsense viewer

MartyG-RealSense, 2022-02-01:
@EzEsh44   You mentioned earlier that you were able to display an **RGBD** image with the rs-capture example.  May I confirm please that both the RGB and depth streams were displayed by your D435 (like the image below), or just the RGB stream.

[image]

If both the RGB and depth streams are able to be displayed by your rs-capture then this would indicate that the camera hardware and drivers are okay.

**********

As a next step, I would suggest uninstalling the RealSense SDK fully instead of overwriting the previous installation by running the automated SDK installer file again.

1.  Type **programs** into the text box at the bottom of your Windows screen to find the **Add or remove programs** tool.

[image]

2.  In the **Apps and Features** alphabetical list of programs that appears, scroll down to the **Intel RealSense SDK** entry and left-click on the 'Uninstall' button to cleanly remove the SDK from your computer.

3.  Go back to the RealSense SDK 'Releases' page and download and run the **Intel. RealSense.SDK-WIN10** installer file again.

https://github.com/IntelRealSense/librealsense/releases

Alternatively, if you have downloaded it recently then you could check in the **C: > Users > [your user account name] > Downloads** folder to find the previously downloaded installer file and run it again from the Downloads folder instead of re-downloading it.

helcl42, 2022-02-01:
> @EzEsh44 You mentioned earlier that you were able to display an **RGBD** image with the rs-capture example. May I confirm please that both the RGB and depth streams were displayed by your D435 (like the image below), or just the RGB stream.
> 
> [image]
> 
> If both the RGB and depth streams are able to be displayed by your rs-capture then this would indicate that the camera hardware and drivers are okay.
> 
> If rs-capture is able to display both depth and RGB correctly then next I would suggest uninstalling the RealSense SDK fully instead of overwriting the previous installation by running the automated SDK installer file again.
> 
> 1. Type **programs** into the text box at the bottom of your Windows screen to find the **Add or remove programs** tool.
> 
> [image]
> 
> 2. In the **Apps and Features** alphabetical list of programs that appears, scroll down to the **Intel RealSense SDK** entry and left-click on the 'Uninstall' button to cleanly remove the SDK from your computer.
> 
> [image]
> 
> 3. Go back to the RealSense SDK 'Releases' page and download and run the **Intel. RealSense.SDK-WIN10** installer file again.
> 
> https://github.com/IntelRealSense/librealsense/releases
> 
> Alternatively, if you have downloaded it recently then you could check in the **C: > Users > [your user account name] > Downloads** folder to find the previously downloaded installer file and run it again from the Downloads folder instead of re-downloading it.

Unfortunately, I run into the same issue after uninstall/install procedure. I have even tested version 2.48 with the same result.

MartyG-RealSense, 2022-02-01:
@helcl42   Can you check what **build version** your Windows is, please.  You can do this by going to the System Information window by typing **system** into the text box at the bottom of the Windows screen and looking at the first few lines of information.

[image]

helcl42, 2022-02-01:
> @helcl42 Can you check what **build version** your Windows is, please. You can do this by going to the System Information window by typing **system** into the text box at the bottom of the Windows screen and looking at the first few lines of information.
> 
> [image]
> 
> I ask because there have been a couple of other cases in the past week involving the SEH exception error on Windows and so I am wondering if a recent Windows update might have been causing a conflict.

Alright, here it is:
[image]

MartyG-RealSense, 2022-02-01:
@helcl42  Thanks very much.  19043 is also known as **21H1**, whilst the later 19044 is **21H2**.  I have 21H2 on my own Windows PCs.

@EzEsh44  What is the build version on your own Windows PC, please?

MartyG-RealSense, 2022-02-07:
Hi @EzEsh44  Do you require further assistance with this case, please?  Thanks!

EzEsh44, 2022-02-07:
Hi @MartyG-RealSense, yes I still need assistance. 
this is my rs-capture:
[image]

reinstalling now.

EzEsh44, 2022-02-07:
[image]

I downloaded the older version and it opens however, the GUI allows me to move the 3d screen but I cannot turn on the camera/depth or any other button on the screen

EzEsh44, 2022-02-07:
well... not sure what the magic bullet was but installing 25.0 seems to have worked. Up and running again
[image]

MartyG-RealSense, 2022-02-07:
I have heard a few times in the past of a non-working modern Viewer version suddenly working after an old Viewer version has been launched.  It's excellent to hear that it is working for you now. :)

helcl42, 2022-02-13:
> I have heard a few times in the past of a non-working modern Viewer version suddenly working after an old Viewer version has been launched. It's excellent to hear that it is working for you now. :)

I have tried the same, just downloaded v2.21 and when I ran it for the first time and the app window just looked like it has zero dimensions(I could see only its decorations) and after maximizing it it told me the following:

[image]

After uninstalling 2.21 and installing 2.50 it seems to be working again.
Thanks for help.

MartyG-RealSense, 2022-02-13:
Thanks very much @helcl42 for the confirmation that the method worked for you too!