realsenseai/librealsense #10189: L515 global timestamps

matthiasverstraete, 2022-01-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  5.4.0-96-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.49.0 }                          | 
| Language                            |  python                          | 
| Segment			|  others                          | 

### Issue Description
I want sync multiple L515 cameras. Therefore I need a global timestamp for each frame. I already enabled the global_time with 

`sensor.set_option(rs.option.global_time_enabled, 1.0)`

However, the timestamp I get is still in hardware clock (`frame_set.get_frame_timestamp_domain()`).

Is there something else I should do in order to get global timestamps?

_A bonus question, is it correct that the L515 doesn't support sensor timestamp? This would be better for my use case, but if it's not possible, that's fine._

Thanks!

MartyG-RealSense, 2022-01-26:
Hi @matthiasverstraete  Do you have a patched kernel so that hardware metadata can be received?  If you built librealsense from Debian packages or from source code using the **RSUSB backend** method then hardware metadata support should be included.  If librealsense was built from source code using the instructions at the **installation.md**
page (a non-RSUSB installation method) then you would need to run a patch script - `./scripts/patch-realsense-ubuntu-lts.sh ` - in order for hardware metadata to be provided.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

[image]

A RealSense team member describes in https://github.com/IntelRealSense/librealsense/issues/2188#issuecomment-409985803 how sensor_timestamp requires the kernel patch.

Having said that, I do not have a reference that definitively states whether sensor_timestamp is available on L515.  However, https://github.com/IntelRealSense/librealsense/issues/8307 details how an L515 user had patched the kernel and still could not access sensor_timestamp.  And a comparison of 400 Series vs L515 frame metadata in the RealSense Viewer shows that the 400 Series has the sensor timestamp listed but L515 does not.

**400 Series**

[image]

**L515**

[image]

If you wish to sync multiple L515 cameras, Intel provide a white-paper guide document for doing so.

https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup

matthiasverstraete, 2022-01-26:
I installed librealsense as a pip package (pyrealsense2) so I don't know if this patch should be applied in that case?

Also, I can get the frame timestamp with `frames.get_frame_metadata(rs.frame_metadata_value.frame_timestamp)`. However, it's always expressed in the hardware clock, while the documentation seems to indicate that if global time is enabled, that it should be expressed in the system clock. Or am I making wrong assumptions?

MartyG-RealSense, 2022-02-03:
Hi @matthiasverstraete  Do you require further assistance with this case, please?  Thanks!

matthiasverstraete, 2022-02-07:
Sorry for the late reply. I will be using the backend timestamp for now.
Thanks for the assistance.

closed 2022-02-07 by matthiasverstraete

MartyG-RealSense, 2022-02-07:
No problem at all, @matthiasverstraete - thanks very much for the update!