realsenseai/librealsense #10191: Getting camera feed on Android App

AlaricLcrs, 2022-01-26:
Hello ! 

| Required Info                      

| Camera Model                       D435i
| Operating System & Version          Ubuntu 16 
| Platform                                 NVidia Jetson, Android phone
| Language                            Python / Android (Java) 
| Segment			Robot A1

I am currently trying to create an Android app that would display a Realsense d435i feed, but the camera is connected to an Nvidia board (because it's on a robot) and not directly to the phone. My first thought was to try to get the feed from the ROS topic generated when launching the camera. I can connect to the NVidia Jeston using its IP adress with a JSch object. So if anyone knows of such a project (open source) that has been done or has any ideas that would help me, I would love to know ! 

Thanks !

MartyG-RealSense, 2022-01-26:
Hi @AlaricLcrs  I have not seen a RealSense project similar to your description.  A discussion on the Nvidia forums suggest using an Android emulator such as **Qemu** on the Jetson.

https://forums.developer.nvidia.com/t/jetson-nano-android-app/75570/11

 You then have the problem of accessing the camera in the Qemu emulation.  Googling for the term **qemu android emulator camera** offers some suggestions in the search results.  I cannot offer any guarantees about what level of success you would have in accessing a RealSense camera with such a method though.

*********

If you are using ROS, an alternative that does not involve Android would be to install the RealSense ROS compatibility wrapper on the Jetson board on the robot and then create an ssh connection from another computer to the RealSense ROS installation running on the Jetson to access the camera streams.

https://github.com/IntelRealSense/realsense-ros

MartyG-RealSense, 2022-02-03:
Hi @AlaricLcrs  Do you require further assistance with this case, please?  Thanks!

AlaricLcrs, 2022-02-03:
Hello thank you for your help ! 

I have not had the time to try it yet but will soon, I'll update on how it's going !

closed 2022-02-20 by MartyG-RealSense