realsenseai/librealsense #10192: Hand Tracking for Intel RealSense

chinmaychinara91, 2022-01-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |  NA                                    | 
| Platform                                 | PC |
| SDK Version                            | legacy / 2.50.0                         | 
| Language                            |  C#/unity                           | 
| Segment			|  others                         | 

### Issue Description
I am looking for features for Hand Tracking and Gesture recognition. Am I missing something or they are not there?

Chinmay

MartyG-RealSense, 2022-01-27:
Hi @chinmaychinara91   The RealSense SDK 2.0 (librealsense2) does not have built-in hand tracking and gesture recognition features.  There are RealSense-compatible commercial software solutions that can provide this functionality though.

**HAND TRACKING**

Hand tracking follows the general hand position but does not provide finger joint tracking

**Cubemos Skeleton Tracking SDKk**
https://www.intelrealsense.com/skeleton-tracking/

**Nuitrack SDK**
https://nuitrack.com/

**Teleidoscope Skeleton Tracking
https://teleidoscope.com/skeletal-detector

**TouchDesigner - RealSense CHOP (requires Cubemos Skeleton Tracking SDK)
https://docs.derivative.ca/RealSense#RealSense_CHOP

**LIPSense 3D Body Pose**
https://www.lips-hci.com/3d-body-pose-sdk

**GESTURE RECOGNITION**

**Gestoos**
https://www.gestoos.com/developer-center

MartyG-RealSense, 2022-02-03:
Hi @chinmaychinara91  Do you require further assistance with this case, please?  Thanks!

chinmaychinara91, 2022-02-03:
Thank you so much for all the resources. You can close the issue.

MartyG-RealSense, 2022-02-03:
You are very welcome - thanks very much for the update.  As you suggest, I will close the issue.  Thanks again!

closed 2022-02-03 by MartyG-RealSense