realsenseai/librealsense #10193: l500-depth.cpp:363) UVC metadata payloads are not available for stream . Please refer to installation chapter for details.

hichamhendy, 2022-01-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Ubuntu 18.04.6 LTS  | 
| Kernel Version (Linux Only)    | 4.9.201-tegra                                    | 
| Platform                                 | NVIDIA Jetson AGX Xavier with connecttect board Rogue Carrier and ARMv8 Processor  |
| SDK Version                            |  2                          | 
| Language                            |  c++/ROS                         | 
| Segment			|  ROS                         | 

### Issue Description

I am facing an issue with launching L515 either on realsense-viewer or ROS. I looked on the former issues there were no specific solution. The problem is primarily with streaming Depth-image.

As I ran 
* **Ubuntu 14/16/18/20 with LTS kernel**
  `./scripts/patch-realsense-ubuntu-lts.sh`<br />
in the source

I got 

Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Unsupported kernel version 4.9.201-tegra . The Bionic patches are maintained for Bionic Beaver LTS kernels 4.1[5/8], 5.[0/3/4]

Is this the reason why I get the error? Anyway, what is the fix?

Thanks for support

MartyG-RealSense, 2022-01-27:
Hi @hichamhendy  Jetson boards have their own dedicated patch script called **./scripts/patch-realsense-ubuntu-L4T.sh**

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend

You can also build librealsense from source code with the **RSUSB backend** installation method, which is not dependent on Linux versions or kernel versions and does not require patching.  Instructions for doing so on Jetson can be foubd at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

hichamhendy, 2022-01-27:
regarding ./scripts/patch-realsense-ubuntu-L4T.sh

I get 

Jetson Board (proc/device-tree/model): jetson-xavier
Jetson L4T version: 32.5
L4T 32.5 to use patches revision 4.4.1
Retrieve the corresponding L4T git tag the kernel source tree
Directory ../linux-4.9-source-tree is present, skipping initialization...
/media/drakula/ssd/Libraries/linux-4.9-source-tree /media/drakula/ssd/Libraries/librealsense
fatal: unable to connect to nv-tegra.nvidia.com:
nv-tegra.nvidia.com[0: 34.216.216.36]: errno=Connection timed out
nv-tegra.nvidia.com[1: 35.82.181.43]: errno=Connection timed out

The matching L4T source tree tag is 
/media/drakula/ssd/Libraries/librealsense
Create the sandbox - NVidia L4T source tree(s)
Downloading default kernel/kernel-4.9 source...
fatal: repository '/media/drakula/ssd/Libraries/librealsense/Tegra/sources/kernel/kernel-4.9' does not exist
/media/drakula/ssd/Libraries/librealsense/Tegra/sources/kernel/kernel-4.9 source sync failed!

Downloading default kernel/nvgpu source...
fatal: repository '/media/drakula/ssd/Libraries/librealsense/Tegra/sources/kernel/nvgpu' does not exist
/media/drakula/ssd/Libraries/librealsense/Tegra/sources/kernel/nvgpu source sync failed!

Downloading default kernel/nvidia source...
fatal: repository '/media/drakula/ssd/Libraries/librealsense/Tegra/sources/kernel/nvidia' does not exist
/media/drakula/ssd/Libraries/librealsense/Tegra/sources/kernel/nvidia source sync failed!

Downloading default hardware/nvidia/soc/t18x source...
fatal: repository '/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/soc/t18x' does not exist
/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/soc/t18x source sync failed!

Downloading default hardware/nvidia/platform/tegra/common source...
fatal: repository '/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/tegra/common' does not exist
/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/tegra/common source sync failed!

Downloading default hardware/nvidia/platform/t18x/common source...
fatal: repository '/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/t18x/common' does not exist
/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/t18x/common source sync failed!

Downloading default hardware/nvidia/platform/t18x/quill source...
fatal: repository '/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/t18x/quill' does not exist
/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/t18x/quill source sync failed!

Downloading default hardware/nvidia/soc/t210 source...
fatal: repository '/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/soc/t210' does not exist
/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/soc/t210 source sync failed!

Downloading default hardware/nvidia/platform/t210/common source...
fatal: repository '/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/t210/common' does not exist
/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/t210/common source sync failed!

Downloading default hardware/nvidia/platform/t210/jetson source...
fatal: repository '/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/t210/jetson' does not exist
/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/t210/jetson source sync failed!

Downloading default hardware/nvidia/platform/t210/porg source...
fatal: repository '/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/t210/porg' does not exist
/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/t210/porg source sync failed!

Downloading default hardware/nvidia/soc/t19x source...
fatal: repository '/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/soc/t19x' does not exist
/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/soc/t19x source sync failed!

Downloading default hardware/nvidia/platform/t19x/common source...
fatal: repository '/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/t19x/common' does not exist
/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/t19x/common source sync failed!

Downloading default hardware/nvidia/platform/t19x/galen/kernel-dts source...
fatal: repository '/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/t19x/galen/kernel-dts' does not exist
/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/t19x/galen/kernel-dts source sync failed!

Downloading default hardware/nvidia/platform/t19x/jakku/kernel-dts source...
fatal: repository '/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/t19x/jakku/kernel-dts' does not exist
/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/platform/t19x/jakku/kernel-dts source sync failed!

Downloading default hardware/nvidia/soc/tegra source...
fatal: repository '/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/soc/tegra' does not exist
/media/drakula/ssd/Libraries/librealsense/Tegra/sources/hardware/nvidia/soc/tegra source sync failed!

./Tegra/sources/kernel/kernel-4.9
./scripts/patch-realsense-ubuntu-L4T.sh: line 104: pushd: ./Tegra/sources/kernel/kernel-4.9: No such file or directory

hichamhendy, 2022-01-27:
Understood
I think the problem now is that cmake can't finde CUDA Compiler 

cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=false -DBUILD_WITH_CUDA=true
-- Checking internet connection...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.50.0
-- Setting Unix configurations
-- Info: Building with CUDA requires CMake v3.8+
-- The CUDA compiler identification is unknown
CMake Error at CMake/cuda_config.cmake:3 (project):
  No CMAKE_CUDA_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CUDACXX" or the CMake cache entry CMAKE_CUDA_COMPILER to the full
  path to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMake/global_config.cmake:73 (include)
  CMakeLists.txt:48 (global_set_flags)

-- Configuring incomplete, errors occurred!
See also "/media/drakula/ssd/Libraries/librealsense-2.50.0/build/CMakeFiles/CMakeOutput.log".
See also "/media/drakula/ssd/Libraries/librealsense-2.50.0/build/CMakeFiles/CMakeError.log".

hichamhendy, 2022-01-27:
Problem solved
just followed instructions here

closed 2022-01-27 by hichamhendy

MartyG-RealSense, 2022-01-27:
That's great to hear that https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-667569588 worked for you, @hichamhendy - thanks very much for the update   :)