realsenseai/librealsense #10196: realsense d415 with jetson nano, no frame received error in realsense viewer

dsmahalle, 2022-01-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D410 } | 
| Firmware Version                   |5.13.0.50 | 
| Operating System & Version |   {Linux (Ubuntu 18)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson nano |
| SDK Version                            |  {  2.}                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

10:50:38 [Error] /sandbox/src/proc/synthetic-stream.cpp:48 - Exception was thrown during callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_

MartyG-RealSense, 2022-01-28:
Hi @dsmahalle   There is a currently active case at https://github.com/IntelRealSense/librealsense/issues/10175#issuecomment-1019744646 of a Jetson TX2 with the same error.  The RealSense user in that case found that the error was occurring if they built the librealsense SDK with CUDA support enabled, whereas the build worked fine if CUDA support was not included.

CUDA support uses an Nvidia graphics GPU such as the one on Jetson boards to accelerate color conversion, pointcloud generation and depth-color alignment operations.

MartyG-RealSense, 2022-02-05:
Hi @dsmahalle  Do you require further assistance with this case, please?  Thanks!

dsmahalle, 2022-02-05:
I have compiled it without cuda but still I am getting that error.

MartyG-RealSense, 2022-02-05:
If your Jetson Nano model has a **barrel jack** connector for providing extra power then Intel recommends enabling it using the instructions in the link below.

https://www.jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/

Are you able to successfully run RealSense programs other than the Viewer on your Nano, such as **rs-capture**?

dsmahalle, 2022-02-05:
Thanks, I'll check it out.

dsmahalle, 2022-02-10:
rs-capture is also throwing same error,

And I have written a python script for depth measurements, that throwing
runtime error frame didn't arrived within 5000

dsmahalle, 2022-02-10:
#!/bin/bash
# Builds the Intel Realsense library librealsense on a Jetson Nano Development Kit
# Copyright (c) 2016-21 Jetsonhacks 
# MIT License

LIBREALSENSE_DIRECTORY=${HOME}/librealsense
INSTALL_DIR=$PWD
NVCC_PATH=/usr/local/cuda/bin/nvcc

USE_CUDA=true

function usage ()
{
    echo "Usage: ./buildLibrealsense.sh [-n | -no_cuda] [-v | -version <version>] [-j | --jobs <number of jobs>] [-h | --help] "
    echo "-n  | --no_cuda   Build with no CUDA (Defaults to with CUDA)"
    echo "-v  | --version   Version of librealsense to build 
                      (defaults to latest release)"
    echo "-j  | --jobs      Number of concurrent jobs (Default 1 on <= 4GB RAM
                      #of cores-1 otherwise)"
    echo "-h  | --help      This message"
    exit 2
}

PARSED_ARGUMENTS=$(getopt -a -n buildLibrealsense.sh -o nv:j:h --longoptions version:,no_cuda,jobs:,help -- "$@" )
VALID_ARGUMENTS=$?

if [ "$VALID_ARGUMENTS" != "0" ]; then
   echo ""
   usage
fi

eval set -- "$PARSED_ARGUMENTS"

LIBREALSENSE_VERSION=""
USE_CUDA=true
NUM_PROCS=""

while :
do
   case "$1" in
      -n | --build_no_cuda) USE_CUDA=false   ; shift ;;
      -v | --version )      LIBREALSENSE_VERSION="$2" ; shift 2 ;;
      -j | --jobs)          NUM_PROCS="$2" ; 
                            shift 2 ;
                            re_isanum='^[0-9]+$'
                            if ! [[ $NUM_PROCS =~ $re_isanum ]] ; then
                              echo "Number of jobs must be a positive, whole number"
                              usage
                            else
                              if [ $NUM_PROCS -eq "0" ]; then
                                echo "Number of jobs must be a positive, whole number" 
                              fi
                            fi ;
       ;;
      -h | --help )         usage ; shift ;;
      # -- means the end of arguments
      --)  shift; break ;;
   esac
done

# From lukechilds gist discussion: https://gist.github.com/lukechilds/a83e1d7127b78fef38c2914c4ececc3c 
# We use wget instead of curl here
# Sample usage:
#   VERSION_STRINGS=$(get_latest_release IntelRealSense/librealsense)

function get_latest_release () {
  # redirect wget to standard out and grep out the tag_name
  wget -qO- https://api.github.com/repos/$1/releases/latest |
    grep -Po '"tag_name": "\K.*?(?=")' 
}
[... 100 lines cut ...]
echo "${green}Installing librealsense, headers, tools and demos${reset}"
sudo make install
  
if  grep -Fxq 'export PYTHONPATH=$PYTHONPATH:/usr/local/lib' ~/.bashrc ; then
    echo "PYTHONPATH already exists in .bashrc file"
else
   echo 'export PYTHONPATH=$PYTHONPATH:/usr/local/lib' >> ~/.bashrc 
   echo "PYTHONPATH added to ~/.bashrc. Pyhon wrapper is now available for importing pyrealsense2"
fi

cd $LIBREALSENSE_DIRECTORY
echo "${green}Applying udev rules${reset}"
# Copy over the udev rules so that camera can be run from user space
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && udevadm trigger

echo "${green}Library Installed${reset}"
echo " "
echo " -----------------------------------------"
echo "The library is installed in /usr/local/lib"
echo "The header files are in /usr/local/include"
echo "The demos and tools are located in /usr/local/bin"
echo " "
echo " -----------------------------------------"
echo " "

dsmahalle, 2022-02-10:
./buildLibrealsense.sh -n

-n for compiling without cuda

dsmahalle, 2022-02-13:
I have tried installing realsense from package but still am getting that
error.

>
>
> Message ID: ***@***.***
> com>
>

MartyG-RealSense, 2022-02-13:
Attempting to build the RealSense SDK from source code with the **RSUSB backend** using the guide at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 may be one of your last remaining options.

dsmahalle, 2022-02-19:
Thanks for your response.

that problem was resolved by building using previous version of
librealsense.

MartyG-RealSense, 2022-02-19:
That's great news, thanks very much for the update!  As you achieved a solution, I will close this case.  Thanks again!

closed 2022-02-19 by MartyG-RealSense

lamiazain, 2022-03-01:
@dsmahalle 

May I know What version you used?

dsmahalle, 2022-03-01:
V2. 40