realsenseai/librealsense #10197: Can't add the server to the list of repositories:

Konown, 2022-01-28:
When add the server to the list of repositories:
sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u
error:5 https://librealsense.intel.com/Debian/apt-repo hirsute InRelease
  403  Forbidden [IP: 13.226.124.17 443]
Reading package list... finish
E: Unable to download https://librealsense.intel.com/Debian/apt-repo/dists/hirsute/InRelease  403  Forbidden [IP: 13.226.124.17 443]
E: Repositories “https://librealsense.intel.com/Debian/apt-repo hirsute InRelease” no digital signature
N: The source cannot be safely updated, so the source is disabled by default.
N: See the apt-secure(8) manual for details on repository creation and user configuration.

MartyG-RealSense, 2022-01-28:
Hi @Konown   May I begin by asking which Linux version and kernel versions you are using, please.  

Also, which type of computer or computing device are you using (Nvidia Jetson, Raspberry Pi, laptop / desktop PC, etc).  Thanks!

MartyG-RealSense, 2022-02-05:
Hi @Konown  Do you require further assistance with this case, please?  Thanks!

ShengMingTang, 2022-02-10:
Hi @MartyG-RealSense , I hit similar issues.  I'm running a desktop PC when following the steps in https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

$ lsb_release -a
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
Codename:	focal

$ sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u
# some Gets and Hits
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo focal InRelease' is not signed.
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/focal/InRelease  403  Forbidden [IP: 52.218.46.153 80]
E: Failed to fetch https://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/bionic/InRelease  403  Forbidden [IP: 52.218.46.153 443]
E: The repository 'https://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease' is not signed.

I'm wondering if there are some issues with the server ?
Thank you.

MartyG-RealSense, 2022-02-10:
Hi @ShengMingTang  Does the problem still occur if you use the amended keyserver address below?

`sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE`

Konown, 2022-02-17:
> Hi @Konown May I begin by asking which Linux version and kernel version you are using, please.
> 
> Also, which type of computer or computing device are you using (Nvidia Jetson, Raspberry Pi, laptop / desktop PC, etc). Thanks!

Hi, @MartyG-RealSense I am using Raspberry Pi 4B, and the Ubuntu version is 18.04. Thanks!

Konown, 2022-02-17:
> Hi, @MartyG-RealSense I am using Raspberry Pi 4B, and the Ubuntu version is 18.04. Thanks!

Recently I tried another source code installation method and it has been successful. Apparently, there are some problems with the above installation method. Thanks!

MartyG-RealSense, 2022-02-17:
The packages on the **distribution_linux.md** installation instructions page are designed for computers with x86 / x64 architecture processors, such as laptop and desktop PCs.  They are not suitable for Arm architecture processors like those in Raspberry Pi.  Typically on Raspberry Pi, librealsense will be built for Ubuntu from source code instead.

MartyG-RealSense, 2022-02-25:
Hi @ShengMingTang and @Konown  Do you require further assistance with this case, please?  Thanks!

closed 2022-03-03 by MartyG-RealSense

MartyG-RealSense, 2025-06-27:
Thanks very much @glianghan  for sharing the solution that worked for you in your situation!