realsenseai/librealsense #10198: Strange program output

Billccx, 2022-01-29:
Dear researchers (Or anyone looking this issue):
         I got the strange output when running an example code which is slightly modified by myself, and I don't know why it was so. 
This is my source code:

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <iostream>             // for cout
#include <opencv/cv.hpp>

// Hello RealSense example demonstrates the basics of connecting to a RealSense device
// and taking advantage of depth data
int main(int argc, char * argv[]) try{
    // Create a Pipeline - this serves as a top-level API for streaming and processing frames
    rs2::pipeline p;

    // Configure and start the pipeline
    p.start();

    while (true){
        // Block program until frames arrive
        rs2::frameset frames = p.wait_for_frames();

        // Try to get a frame of a depth image
        rs2::depth_frame depth = frames.get_depth_frame();
        rs2::video_frame color = frames.get_color_frame();

        auto w = color.get_width();
        auto h = color.get_height();
        cv::Mat image(cv::Size(w, h), CV_8UC3, (void*)color.get_data(), cv::Mat::AUTO_STEP);

        std::cout<<"color frame, width="<<w<<" ,height="<<h<<std::endl;
        std::cout<<std::endl<<std::endl;

        // Get the depth frame's dimensions
        auto width = depth.get_width();
        auto height = depth.get_height();

        // Query the distance from the camera to the object in the center of the image
        float dist_to_center = depth.get_distance(width / 2, height / 2);

        // Print the distance
        std::cout << "The camera is facing an object " << dist_to_center << " meters away \r";
    }
    return EXIT_SUCCESS;
}
catch (const rs2::error & e){
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e){
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}

The output of the program is:
`

    color frame, width=1280 ,height=720
    
    
    color frame, width=1280 ,height=720ters away 
    
    
    color frame, width=1280 ,height=720ters away 
    
    
    color frame, width=1280 ,height=720ters away 
    
    
    color frame, width=1280 ,height=720ters away 
    
    
    color frame, width=1280 ,height=720ters away 
    
    
    color frame, width=1280 ,height=720ters away 
    
    ...

`

It is obviously that the output 'height=720ters away ' is wrong, but why it was so?
I am looking forward to your reply and thanks in advance.

        

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Operating System & Version |   Linux (Ubuntu 20.04) |                               
| Platform                                 | PC |
| SDK Version                            |  2               | 
| Language                            | C++                     |

Billccx, 2022-01-29:
the second line in source code is:
`#include <iostream>`

MartyG-RealSense, 2022-01-29:
Hi @Billccx  I see that your code is based on the **rs-hello-realsense** C++ example program that prints out the distance in real-world meters from the center depth pixel to an observed object / surface.

https://github.com/IntelRealSense/librealsense/blob/master/examples/hello-realsense/rs-hello-realsense.cpp

When you print the width and height of the color frame though, you are simply printing out the default color **resolution** of the L515 - 1280x720

[image]

Billccx, 2022-01-29:
> Hi @Billccx I see that your code is based on the **rs-hello-realsense** C++ example program that prints out the distance in real-world meters from the center depth pixel to an observed object / [...]

Thanks a lot for such a quick reply !
But actually, what confused me is that the output of the program is not "intact".
According to the source code, the right output should be:

  color frame, width=1280 ,height=720
  
  
  The camera is facing an object "xxx" meters away
  
  
  color frame, width=1280 ,height=720
  
  
  The camera is facing an object "xxx" meters away
  ...

however, it turned out to be:

  color frame, width=1280 ,height=720ters away 
  
  
  color frame, width=1280 ,height=720ters away 
  
  
  color frame, width=1280 ,height=720ters away
  ...

which is very strange.
I just wonder why it was so.

MartyG-RealSense, 2022-01-29:
I wonder if you are missing a quotation mark when printing the color.  The width 'w' has opening and closing quotation marks - " " - but height 'h' only has opening quotation marks but not a closing one.

[image]

Billccx, 2022-01-30:
> I wonder if you are missing a quotation mark when printing the color. The width 'w' has opening and closing quotation marks - " " - but height 'h' only has opening quotation marks but not a closing one.
> 
> [image]

Got it. Thank you so much. It seems that I made such a stupid mistake and I am really sorry for wasting your time.

MartyG-RealSense, 2022-01-30:
It's no trouble at all - I'm pleased that I could help. :)

closed 2022-02-04 by MartyG-RealSense