realsenseai/librealsense #10201: Using spatial filter and temporal filter breaks the point cloud for the second realsense d415

UmutUz, 2022-01-31:
I use 2 D415 realsenses to capture pointclouds. Each realsense 1 pointcloud. I try to apply some post processing filters on my point clouds because they are noisy as heck and practically useless otherwise. As seen in the code below, I apply the filters on each depth frame. This works for both cameras except when I apply the spatial and temporal filter then the pointclouds just break completely or arent shown in PCL viewer anymore no matter which one I apply or if both are applied. I guess I am applying them wrong? I just want to use the filters on a single depth frame.

`//realsense pointclouds
	rs2::pointcloud pcl_camera1;
	rs2::pointcloud pcl_camera2;

	rs2::context ctx;    // Create librealsense context for managing devices

	std::vector<rs2::pipeline> pipelines;

	rs2::temporal_filter temporal_filter;
	rs2::decimation_filter dec_filter;
	rs2::spatial_filter spat_filter;
	rs2::disparity_transform depth_to_disparity(true);
	rs2::disparity_transform disparity_to_depth(false);

	temporal_filter.set_option(RS2_OPTION_FILTER_SMOOTH_ALPHA, 0.3f);
	temporal_filter.set_option(RS2_OPTION_FILTER_SMOOTH_DELTA, 20.0f);

	dec_filter.set_option(RS2_OPTION_FILTER_MAGNITUDE, 2.0f);

	spat_filter.set_option(RS2_OPTION_FILTER_MAGNITUDE, 1.0f);
	spat_filter.set_option(RS2_OPTION_FILTER_SMOOTH_ALPHA, 0.6f); //vorher 0.25, danach 0.45, danach 0.40
	spat_filter.set_option(RS2_OPTION_FILTER_SMOOTH_DELTA, 8.0f);
	
	for (auto&& dev : ctx.query_devices())
	{
		rs2::pipeline pipe(ctx);

		// Start a streaming pipe per each connected device
		rs2::config cfg;
		
		cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
		pipe.start(cfg);
		pipelines.emplace_back(pipe);
	}

	rs2::frameset frames1;
	rs2::frameset frames2;

	// Drop frames so auto exposure can settle 
	for (int i = 0; i <= 200; i++) {
		 frames1 = pipelines[0].wait_for_frames();
		 frames2 = pipelines[1].wait_for_frames();
	}

	rs2::frame color1 = frames1.get_color_frame();
	rs2::frame color2 = frames2.get_color_frame();

	// Map RGB to PCL
	pcl_camera1.map_to(color1);
	pcl_camera2.map_to(color2);

	// Get frames from each camera
	cout << "Capturing color and depth frames from each camera...";

	//Smooth out pointcloud taken by Realsense
	rs2::frame depth_filtered = frames1.get_depth_frame();
	//rs2::points points_camera_unfiltered1 = pcl_camera1.calculate(depth_filtered);
	
	depth_filtered = dec_filter.process(depth_filtered);
	depth_filtered = depth_to_disparity.process(depth_filtered);
	//depth_filtered = spat_filter.process(depth_filtered);
	//depth_filtered = temporal_filter.process(depth_filtered);
	depth_filtered = disparity_to_depth.process(depth_filtered);
 
	rs2::frame depth_filtered2 = frames2.get_depth_frame();
	//rs2::points points_camera_unfiltered2 = pcl_camera2.calculate(depth_filtered2);

	depth_filtered2 = dec_filter.process(depth_filtered2);
	depth_filtered2 = depth_to_disparity.process(depth_filtered2);
	//depth_filtered2 = spat_filter.process(depth_filtered2);
	//depth_filtered2 = temporal_filter.process(depth_filtered2);
	depth_filtered2 = disparity_to_depth.process(depth_filtered2);

	// Generate the pointcloud and texture mappings
	rs2::points points_camera1 = pcl_camera1.calculate(depth_filtered);
	rs2::points points_camera2 = pcl_camera2.calculate(depth_filtered2);`

MartyG-RealSense, 2022-01-31:
Hi @UmutUz  In Intel's post-processing documentation there is a recommended order for applying the list of filters that the list in your own script is not following.  For the filters that you are using, Intel's recommended order is:

Decimation > Depth to Disparity > Spatial > Temporal > Disparity to Depth

https://dev.intelrealsense.com/docs/post-processing-filters#section-using-filters-in-application-code

I also note that you have set the Disparity to Depth filter to 'False', whilst 'Depth to Disparity' is set to 'True'.  This is a correct approach.  When using either of these two filters, both of them have to be defined, and one has to be set to True and the other to False (they cannot both be true or both be false).  But after having set Disparity to Depth to false, it is then included in your list of filters that are applied later in the script.

[image]

UmutUz, 2022-01-31:
When I remove those 2 lines the program crashes. If I remember correctly, I had an issue up here where you helped me out by telling me to include those lines. I removed them again and the programm crashes.

MartyG-RealSense, 2022-01-31:
I believe that the case was https://github.com/IntelRealSense/librealsense/issues/10110 .... reviewing the code, I did suggest adding that line.  I was quoting from another RealSense user's script.  Apparently that approach does work, even if does not make logical sense  :)  I do apologize.

On further reflection, your description of the effect on the point cloud is something that I have encountered in the RealSense Viewer, where having the temporal and spatial filters enabled in 3D point cloud mode can cause the point cloud to disappear, whilst they return if the filters are disabled.  It seems to occur on one particular computer, whilst it works fine on a different PC.

UmutUz, 2022-01-31:
Okay that is kind of interesting! I don't have the means to test it out on another PC but I guess I have to live with the results for now and turn off those filters. Thanks for your input tho :)

MartyG-RealSense, 2022-01-31:
A symptom of problems on the affected machine is if both of the Infrared streams are missing in the Stereo Module section of the RealSense Viewer's options side panel, or if only Infrared is present but not Infrared 2 even though the camera is on a USB 3 connection (Infrared 2 is not available on USB 2.1).

UmutUz, 2022-01-31:
I have USB 3 so that shouldnt be the problem I guess.

MartyG-RealSense, 2022-01-31:
Would it be possible to check which Infrared options, if any, you have listed in the Stereo Module of the Viewer on your computer, please?

UmutUz, 2022-01-31:
I have it listed in the viewer but the boxes are not ticked. I can choose between different options in the viewer for each infrared.

MartyG-RealSense, 2022-01-31:
Okay, thanks very much for checking.  It does not matter if they are not ticked.  The most important thing is that they are included in the list of stream types along with the depth stream.

If it is not possible to enable the spatial or temporal filters on that particular computer then we could try to reduce the noise instead.  Does the location that the camera is used in have fluorescent lights such as ceiling strip lights?  If so then these can introduce noise because the gas inside them flickers at a rate that is hard to see with the human eye.

UmutUz, 2022-01-31:
They are exactly those type of lights! Well looks like I need a scenery changes! Never thought of it

MartyG-RealSense, 2022-01-31:
You may not necessarily need to.  You could first try adjusting the FPS speed to closely match the operating rate of the particular lights in your location.  If you have lights that operate at 30 or 60 hz then setting the FPS to 30 or 60 should help to reduce the negative effects.

If you are in a European location that uses 50 Hz lights then it is more of a problem, as the RealSense SDK does not support 50 FPS.  In this scenario, setting an RGB option called **Power Line Frequency** value may help.

[image]

UmutUz, 2022-01-31:
Alright thanks for the tips, I will get back to this thread when I get to test it.

MartyG-RealSense, 2022-01-31:
You are very welcome.  I look forward to your test results.   Good luck!

UmutUz, 2022-02-01:
I tried your suggestions and I did notice a bit of improvement but it isn't enough sadly

MartyG-RealSense, 2022-02-01:
Would it be possible to post an RGB image of the scene that the camera is observing to see whether there might be other possibly disruptive factors in the scene, please?

UmutUz, 2022-02-01:
That's not possible _but_ I got my pointclouds back with filters on and managed to keep them not destroyed. I moved the filters into the for-loop.

 ```

for (auto&& dev : ctx.query_devices())
{
	rs2::pipeline pipe(ctx);

	// Start a streaming pipe per each connected device
	rs2::config cfg;
	temporal_filter.set_option(RS2_OPTION_FILTER_SMOOTH_ALPHA, 0.3f);
    temporal_filter.set_option(RS2_OPTION_FILTER_SMOOTH_DELTA, 20.0f);
    
    dec_filter.set_option(RS2_OPTION_FILTER_MAGNITUDE, 2.0f);
    
    spat_filter.set_option(RS2_OPTION_FILTER_MAGNITUDE, 1.0f);
    spat_filter.set_option(RS2_OPTION_FILTER_SMOOTH_ALPHA, 0.6f); //vorher 0.25, danach 0.45, danach 0.40
    spat_filter.set_option(RS2_OPTION_FILTER_SMOOTH_DELTA, 8.0f);
	cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
	pipe.start(cfg);
	pipelines.emplace_back(pipe);
}
```

And this is how I used the filters on the depth frame

```
		depth_filtered = frames1.get_depth_frame();
		depth_filtered2 = frames2.get_depth_frame();

		depth_filtered = dec_filter.process(depth_filtered);
		depth_filtered2 = dec_filter.process(depth_filtered2);

		depth_filtered = depth_to_disparity.process(depth_filtered);
		depth_filtered2 = depth_to_disparity.process(depth_filtered2);

		depth_filtered = spat_filter.process(depth_filtered);
		depth_filtered2 = spat_filter.process(depth_filtered2);

		depth_filtered = temporal_filter.process(depth_filtered);
		depth_filtered2 = temporal_filter.process(depth_filtered2);

		depth_filtered = disparity_to_depth.process(depth_filtered);
		depth_filtered2 = disparity_to_depth.process(depth_filtered2);
```

For whatever reason this fixes the issue but I don't see much of an effect of the temporal filter. Oh well, I guess that should be enough though

MartyG-RealSense, 2022-02-01:
It's excellent to hear that you achieved a solution.  Thanks very much for the update!

UmutUz, 2022-02-01:
Thanks for your help too! :)

closed 2022-02-01 by UmutUz