realsenseai/librealsense #10203: Are there every any non zero values for rs2_intrinsics::coeffs in the d415 and d435 cameras ?

GrubbyHalo, 2022-01-31:
It seems like these two models have a value of zero for all the coeffs. What cameras or distortion models populate this array of coefficients ? I have only used the d435 and d415 cameras and  they all seem to have this set to zero. This makes the ```rs2_deproject_pixel_to_point``` function quite inefficient in this particular case as this:

```
static inline void rs_deproject_pixel_to_point(float point[3], const struct rs2_intrinsics* intrin, const float pixel[2], float depth)
{
    float x = (pixel[0] - intrin->ppx) / intrin->fx;
    float y = (pixel[1] - intrin->ppy) / intrin->fy;

    float xo = x;
    float yo = y;
    .
    .
    .

    if (intrin->model == RS2_DISTORTION_BROWN_CONRADY)
    {
        // need to loop until convergence
        // 10 iterations determined empirically
        for (int i = 0; i < 10; i++)
        {
            float r2 = x * x + y * y;
            float icdist = (float)1 / (float)(1 + ((intrin->coeffs[4] * r2 + intrin->coeffs[1]) * r2 + intrin->coeffs[0]) * r2);
            float delta_x = 2 * intrin->coeffs[2] * x * y + intrin->coeffs[3] * (r2 + 2 * x * x);
            float delta_y = 2 * intrin->coeffs[3] * x * y + intrin->coeffs[2] * (r2 + 2 * y * y);
            x = (xo - delta_x) * icdist;
            y = (yo - delta_y) * icdist;
        }

    }
    .
    .
    .

    point[0] = depth * x;
    point[1] = depth * y;
    point[2] = depth;
}

```
That looping for 10 iterations on each pixel kills performance on largish images.

essentially resolves to this:

```
static inline void rs_deproject_pixel_to_point(float point[3], const struct rs2_intrinsics* intrin, const float pixel[2], float depth)
{
    float x = (pixel[0] - intrin->ppx) / intrin->fx;
    float y = (pixel[1] - intrin->ppy) / intrin->fy;

    point[0] = depth * x;
    point[1] = depth * y;
    point[2] = depth;
}

```

Unless there is something I am missing. Maybe a flag can be set within the rs2_intrinsics struct giving a hint that coefficients aren't being used and allow for a performance increase in  ```rs_deproject_pixel_to_point ```

MartyG-RealSense, 2022-01-31:
Hi @GrubbyHalo  Most of the coefficients are deliberately set to zero by intentional design, as described by a RealSense team mebber in https://github.com/IntelRealSense/librealsense/issues/1430#issuecomment-375945916

An exception to this rule is the **D455** camera model, which has non-zero coefficients (though very close to zero) on its **color** profiles.

If you are generating a point cloud then using **points.get_vertices()** may be more accurate than rs2_deproject_pixel_to_point, as discussed in detail at https://github.com/IntelRealSense/librealsense/issues/4315

GrubbyHalo, 2022-01-31:
Thanks for that @MartyG-RealSense

closed 2022-01-31 by GrubbyHalo