realsenseai/librealsense #10207: OS freezes and it seems it's related with using my realsense D435i

martinakos, 2022-02-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu  20.04  | 
| Kernel Version (Linux Only)    |  5.13.0-27-generic                                         | 
| Platform                                 | Laptop  Dell Inc. Precision 7550  BIOS 1.12.0 12/12/2021|
| SDK Version                            |  2.50.0                          | 
|librealsense2-dkms                   |1.3.18-0ubuntu1     |
| ROS Wrapper version             | 2.3.2                              |

### Issue Description
My OS freezes randomly and the only thing I can do when it happens it's to reboot.
This is a new laptop running Ubuntu 20.04 so initially I thought it was something with the laptop so I've spent almost two weeks trying things about the laptop, like upgrading to the latest BIOS firmware, latest Ubuntu updates, latest kernel, latest NVidia drivers. 

However, each time there is a freeze is related with me using the realsense D435i. I use the D435i through the official ROS wrapper ver. 2.3.2. It may take a while when I'm using the camera without problems but eventually it freezes. I've tried to reinstall the realsense SDK and ROS Wrappers multiple times but I still get the random freezes. I've installed the realsense ROS driver following both of the methods described in [here](https://github.com/IntelRealSense/realsense-ros) and I still get freezes. 

If I check the syslog at the time of the freeze I find multiple suspicious entries:

```
Jan 25 16:39:37 UKRDL64049 kernel: [ 4765.512390] general protection fault, probably for non-canonical address 0xc47f8d1f6b73239c: 0000 [#1] SMP NOPTI
Jan 25 16:39:37 UKRDL64049 kernel: [ 4765.512396] CPU: 2 PID: 906 Comm: systemd-udevd Tainted: P           OE     5.13.0-27-generic #29~20.04.1-Ubuntu
Jan 25 16:39:37 UKRDL64049 kernel: [ 4765.512399] Hardware name: Dell Inc. Precision 7550/01CFGW, BIOS 1.12.0 12/12/2021
```

```
Jan 28 15:03:23 UKRDL64049 kernel: [ 6065.459196] kernel BUG at mm/usercopy.c:99!
Jan 28 15:03:23 UKRDL64049 kernel: [ 6065.459199] invalid opcode: 0000 [#1] SMP NOPTI
Jan 28 15:03:23 UKRDL64049 kernel: [ 6065.459201] CPU: 7 PID: 115915 Comm: realsense-viewe Tainted: P           OE     5.13.0-27-generic #29~20.04.1-Ubuntu
Jan 28 15:03:23 UKRDL64049 kernel: [ 6065.459203] Hardware name: Dell Inc. Precision 7550/01CFGW, BIOS 1.12.0 12/12/2021
```
 Once at the time of the freeze the syslog just recorded \00\00\00\...  but it happened just after realsense message of unknown video format
[image]

I found a possibly related issue #10158  but there wasn't a resolution.  
I'm running out of ideas. Any suggestions to troubleshoot this problem?

MartyG-RealSense, 2022-02-01:
Hi @martinakos   There was a RealSense ROS user recently at https://github.com/IntelRealSense/librealsense/issues/10159 who was experiencing freezing and crashing with Ubuntu on a laptop.  They posted a method that resolved their particular problem at https://github.com/IntelRealSense/librealsense/issues/10159#issuecomment-1021297602

martinakos, 2022-02-01:
So this user purged all realsense related modules and built and install librealsense from source. Is that your suggestion?

MartyG-RealSense, 2022-02-01:
They are uninstalling **all** librealsense related Debian packages with the purge instruction from the librealsense package installation instructions page.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

My interpretation of the solution in the comment is that they installed librealsense and the ROS wrapper together from packages with the **Method 1** instructions of the ROS wrapper but it caused problems for their laptop and so they wanted to remove all librealsense related packages to resolve that conflict.

It sounds as though they wanted to use Method 1 package installation to ensure that they had all required dependencies such as OpenCV and then use the purge to remove all librealsense packages, but probably leaving non-librealsense dependencies installed.  

It may be an over-complicated approach.  It may be better to purge all packages, and then install librealsense first (by packages or source, however you prefer) and then use the ROS wrapper's **Method 2** to afterwards install the ROS wrapper from source.  When building the ROS wrapper from source, it always has to be done secondly after librealsense has first been successfully installed.

martinakos, 2022-02-03:
It seems I finally solved the freezes. After removing all realsense related libraries I tried to build the librealsense2 from source.
This message came up when building the dmks modules:
`Unsupported kernel version 5.13.0-27-generic . The Focal patches are maintained for Ubuntu LTS with kernel 5.4, 5.8, 5.11 only`

I booted with an older kernel 5.11.0-34-generic, two kernel updates ago, and installed the realsense-ros wrapper using Method 1 (which installs ros-noetic-librealsense2 too ). It hasn't frozen since yesterday morning. Before that change it would freeze 8 times a day if I'm working (or rather trying to work) with the camera.

Is there any list online that tells you which firmware /SDK version / linux Kernel version are supported? I know [this](https://dev.intelrealsense.com/docs/firmware-releases) list. But it doesn't say what linux kernel is needed. Unless the Release column, which is 5.11, is also the linux kernel which corresponds in this case but it may just be a coincidence in version numbers.

I think there should be some check when installing debian packages so that if I install the realsense SDK it checks that the kernel is supported and stops the install if it's not (unless forced). 

So to summary, the combination that seems to work for me is:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu  20.04  | 
| Kernel Version (Linux Only)    |  5.11.0-34-generic  | 
| SDK Version                            |  2.50.0                          | 
| ROS Wrapper version             | 2.3.2                              |

MartyG-RealSense, 2022-02-03:
Hi @martinakos  I'm pleased to hear that you found a solution.  Thanks very much for sharing the details with the RealSense community  :)

The **Supported Platforms** section of the release notes for each librealsense version has a list of officially supported kernels.

https://github.com/IntelRealSense/librealsense/releases

[image]

At the time of writing this, **5.8** is the most recent officially supported kernel.  A patch contributed by a RealSense user for support of kernel 5.11 was also added in SDK 2.50.0, as described at https://github.com/IntelRealSense/librealsense/pull/9727

The librealsense SDK can work with newer kernels that are not yet officially unsupported but there may be unpredictable consequences in regard to stability.  Building librealsense with the **RSUSB** method will provide the most stable experience on computers with unsupported kernels, as this build type is not dependent on Linux versions or kernel versions and does not require patching.

MartyG-RealSense, 2022-02-15:
Case closed due to solution achieved and no further comments received.

closed 2022-02-15 by MartyG-RealSense

Mat198, 2022-03-18:
Installing librealsense by package an then using Method 2 for Ros warper worked here as suggested by MartyG. Don't forget to install  ddynamic_reconfigure (sudo apt install ros-"rosdist"-ddynamic-reconfigure) and don't confuse it with dynamic-reconfigure.