realsenseai/librealsense #10208: Using with VS 2019/22

arothenberg, 2022-02-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D435  | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC |
| SDK Version                            | 2.32                      | 
| Language                            |  C++                      | 
| Segment			|desktop app  | 

### Issue Description
I'm getting back to a realsense project after a year and I am using vs 2019 or 22 (i have only  2015 and the 2017 costs a large fee).

I see that the directions for using the SDK suggest 2017. I tried with 2019 and I'm immediately  getting a few errors  when building. Should the SDK work out of the box using VS 2019 or 22 with  cmake and SDK 2.0?
Thanks.

MartyG-RealSense, 2022-02-01:
Hi @arothenberg  As far as I am aware, VS 2015 is still the minimum supported Visual Studio version for the RealSense SDK.  Current official documentations such as those for the C# wrapper and installing the SDK on Windows from source code state support for it.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp#with-cmake-and-visual-studio

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

If you plan to use VS2019 then the starter project in the link below that was contributed by a RealSense user may be a helpful reference for avoiding errors such as **LNK2019**

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

I have not yet seen any cases of the SDK being used with Visual Studio 2022.

arothenberg, 2022-02-01:
Thanks. I'll use Eduardo Wangs project and directions as a template. If I have more trouble I'll check in.
TYVM.

closed 2022-02-01 by arothenberg