realsenseai/librealsense #10214: L500 Depth Sensor No Frames Received

TonyAnnerino, 2022-02-02:
| Camera Model                       |  L515  | 
| Firmware Version                   | 01.05.08.01| 
| Operating System & Version |  Windows 10  | 
| Platform                                 | PC |

### Issue Description
I recently bought an L515 depth tracking unit and immediately was alarmed when I first tried turning it on because the cable that came with the camera does not agree with the personal laptop that runs Windows 10 I plan to run the camera on. As shown in the first picture here, the USB port on my laptop is USB 3.0. I am running RealSense Viewer 2.50.0 with firmware 01.05.08.01.
[image]

To check if the issue was with the camera or my laptop or the cable, I then tried running the camera on my laptop with the USB cable I use to charge my phone. This allowed my laptop to recognize the camera, but it registers the connection as USB 2.1 because of the cable as shown in the next picture. This did not appear to cause any problems, so I went on to spend a while operating the camera to figure out what exact settings work best for my application. Everything was working perfectly fine until the L500 Depth Sensor abruptly threw the error shown in the next picture ("No Frames Received!). Nothing that I could observe in that moment can explain why this happened. The RGB Camera is still fully functional.
[image]

To check if this new problem was related to my laptop or the camera, I had a counterpart install RealSense Viewer 2.50.0 with firmware 01.05.08.01 on her personal laptop that also runs Windows 10. The cable that came with my camera works as intended on her laptop as shown in the next picture, but the same error occurs when the L500 Depth Sensor is turned on with the camera connected to her laptop (as also shown in the same picture). 
[image]

I then reconnected the camera to my laptop and tried everything described in this other thread that seems to be about a similar or maybe identical issue, but nothing helped.
https://github.com/IntelRealSense/librealsense/issues/10062

I would greatly appreciate any further suggestions on what I might try to resolve this issue with my brand new device.

MartyG-RealSense, 2022-02-03:
Hi @TonyAnnerino   As you said that you have tried everything in https://github.com/IntelRealSense/librealsense/issues/10062 may I first confirm that you have tried reversing the orientation of the USB connector at the micro-sized end of the cable that plugs into the base of the camera, as suggested in https://github.com/IntelRealSense/librealsense/issues/10062#issuecomment-994473277

TonyAnnerino, 2022-02-03:
Hi @MartyG-RealSense, I confirm that I did try flipping the end of the cable that plugs into the camera. It changed nothing.

MartyG-RealSense, 2022-02-03:
I checked carefully through your logs and noted the message **Fatal error occurred (16)**.  If you have this error then the camera should be returned to the retailer that you purchased it from using the **Return Material Authorization (RMA)** procedure of that retailer, as it is a hardware fault.  

The links below provide past cases where this L515 error has occurred with the same behaviour on the depth and infrared streams and RMA return has been recommended.

https://github.com/IntelRealSense/librealsense/issues/9329
https://github.com/IntelRealSense/librealsense/issues/9333
https://support.intelrealsense.com/hc/en-us/community/posts/4405115604243-No-frames-received-on-L515

There was a rare case at https://github.com/IntelRealSense/librealsense/issues/8889 though where a RealSense user on Windows resolved the error by uninstalling RealSense and then reinstalling it with **Run As Administrator**.

RealSenseSupport, 2022-02-11:
@TonyAnnerino Please do RMA procedure for the failed L515. And any other questions about this ticket? Looking forward to your reply. Thanks!

RealSenseSupport, 2022-02-23:
@TonyAnnerino We will close this ticket since no further questions from your side. Thanks!

closed 2022-02-23 by RealSenseSupport