realsenseai/librealsense #10215: cannot use L515 in Unity

heojun990322, 2022-02-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.05.08.01 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  v2.50.0                        | 
| Language                            | unity 2020.1.8f1       | 
| Segment			|  VR/AR                        | 

### Issue Description
I downloaded unity package latest version and the others im using are latest version. I played on start scene but i couldnt use L515.. erroe message like below 

[image]

what can i do currently??

MartyG-RealSense, 2022-02-03:
Hi @heojun990322  The L515 camera model has been compatible with the RealSense Unity wrapper since SDK version 2.37.0 and the Unity wrapper is also compatible with Unity 2020.  

May I first confirm that you are able to use the camera normally outside of Unity in the RealSense Viewer tool, please?

If the camera does work correctly in the Viewer, have you followed the Unity wrapper setup method below.

1.  Download the Unity package file **Intel.RealSense.unitypackage** from the 'Assets' file list for version 2.50.0 of the SDK.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

[image]

2.  Launch Unity and open a project.  A clean new project will be best for testing purposes.  

3.  Go to **Assets > Import Package > Custom Package** and select the Intel.RealSense.unitypackage file from the location on your computer that it was downloaded to (likely to be the Downloads directory of your Windows user account folder).  The RealSense Unity wrapper should then import into your project.

4.  Go to the **Assets > RealSenseSDK2.0 > Scenes** folder in your Unity 'Assets' panel and double-leftclick on the scene file called **StartHere**.  When the scene has loaded, click on the small triangular 'Play' button at the top of your Unity screen  (not the **File > Build and Run** method) to run the scene in the Unity editor.

[image]

heojun990322, 2022-02-03:
Thnks for reply @MartyG-RealSense 
I confirmed my L515 works correctly in the RealSense Viewer tool.
Then I followed steps you suggested from 1 to 4 accurately.
But I got same result. 

When I clicked play button is 
[image]

And when I clicked start is
[image]

Below is console..
[image]

MartyG-RealSense, 2022-02-03:
The warning **Couldn't resolve requests** indicates that a resolution or FPS speed was being requested that the L515 camera does not support.

Could you provide an image of profile details listed in the **RsDevice** component for the 'StartHere' scene please?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unity#realsensedevice

heojun990322, 2022-02-04:
Profile listed in my RsDevice component
[image]

MartyG-RealSense, 2022-02-04:
The camera profile looks fine for an L515 that has firmware 1.5.8.1 installed (to provide 640x480 RGB support).  

I ran my own tests in Unity **2021.1.16f1** with an L515 using the 4-step procedure I described earlier.  I didn't experience the **ExternalException** red error that you did.  Furthermore, my **StartHere** scene had a full list of samples provided when run.

[image]

The sample programs were also able to run correctly on my L515.

[image]

heojun990322, 2022-02-07:
I had downloaded Unity 2021.1.16f1 and tesed L515 following steps again. But I experienced same problems. 
And I had tried it on my labtop too, not my PC. Strangely, the result was same..

MartyG-RealSense, 2022-02-07:
You have said that your L515 camera works correctly in the Viewer, which suggests that it is a problem that is not related to the camera hardware or the firmware driver.

Is there any difference if you unplug the micro-sized end of the USB cable from the base of the L515 camera, turn the connector around the other way and re-insert it into the camera base (USB-C cables are two-way insertion at the micro-sized end).

closed 2022-02-19 by MartyG-RealSense