realsenseai/librealsense #10216: Enabling Post-processing Destroy the Point Cloud Video

koalaaaaaaaaa, 2022-02-03:
Hi,  happy Chinese Lunar new year~ I‘m new to Intel Real-Sense camera and I encounter a problem. I have searched for similar issues but found no solution, so I appreciate your kindness help.

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | d455 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |  Win 10 | 
| Platform                                 | PC |
| SDK Version                             |2.50.0.3785   | 
| Language                            | (I use view.exe only) |
| Segment			      |(I'm not sure what it means)|

### Issue Description
<Describe your issue / question / feature request / etc..>
When I try to view point-cloud  video in 3-D mode, I found I can't use "post-processing" function in "Stereo Module" as the tutorial below did.
[image]
What happened is that when I turn on "post-processing", in most cases, the point cloud video disappeared with no points left on screen, while in occasional cases, the video will last for about 5 secs and then disappeared.
(before)
[image]
(after)
[image]
and my settings of "post-processing" is as follows.
[image]
Thank you for your kindness help : )

MartyG-RealSense, 2022-02-03:
Hi @koalaaaaaaaaa  The disappearing point cloud in the Viewer is a known issue.  Please try disabling the **GLSL** options in the Viewer's settings interface using the instructions at https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

koalaaaaaaaaa, 2022-02-03:
Thanks a lot~
I'd expected you replied me after 12 hours or so Ha ha ha, (since I'm suppose you're in the opposite site of the earth.)

MartyG-RealSense, 2022-02-03:
You are very welcome.  Yes, timezones can be a factor.  :)

closed 2022-02-03 by koalaaaaaaaaa