realsenseai/librealsense #10218: pyrealsense2_net module not found when using pypi package on pc

elishafer, 2022-02-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | N/A | 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    |          5.4.0-91-generic                            | 
| Platform                                 | PC  |
| SDK Version                            |  2.50.0                         | 
| Language                            |  python                          | 
| Segment			|  Robot                         | 

### Issue Description
I'm trying to run the [net_viewer](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/net_viewer.py) example but get an error when it tries to import `pyrealsense2_net` module.
```ModuleNotFoundError: No module named 'pyrealsense2_net'```
These are the versions That I have installed:
* pyrealsense2 version(from pypi): 2.50.0.3812
* librealsense version (as part of ros melodic): 2.50.0-1bionic.20211115.141014

Any ideas what I could be doing wrong?

MartyG-RealSense, 2022-02-03:
Hi @elishafer  There is a previous report at https://github.com/IntelRealSense/librealsense/issues/9946 of the **ModuleNotFoundError: No module named 'pyrealsense2_net'** occurring when using the **rs_viewer.py** example program.

The main list of examples in the Python wrapper documentation provides the description "Shows how to connect to rs-server over network" for the rs_viewer.py example.

You may have already come across **rs_server** if you have read Intel's white-paper document on the subject of open-source ethernet networking.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

If you built librealsense and the RealSense ROS wrapper together from packages with the wrapper's **Method 1** instructions then librealsense would not have been built with networking support included.  This support can be included in the SDK by building it from source with CMake with the build flag **-DBUILD_NETWORK_DEVICE=ON** included in the CMake build instruction, as described in https://github.com/IntelRealSense/librealsense/issues/7606#issuecomment-714169261

Including the -DBUILD_NETWORK_DEVICE=ON build flag in the CMake build instruction when building librealsense from source code causes the 'realsense2-net' module and the 'rs-server' tool to be built.

Niko-La, 2022-02-10:
is the wrapper supported for python 3.10?  cant find the module on pip? is there clear instructions on building from source if that is the only option?

on windows 10
[image]

MartyG-RealSense, 2022-02-10:
Hi @Niko-La  At the time of writing this, the most recent Python version that pip install of pyrealsense2 supports is Python 3.9.  Further information about this can be found at https://github.com/IntelRealSense/librealsense/issues/10057

Users of newer Python versions than 3.9 should build the wrapper from source code.  The simplest approach to buiding from source would be to build librealsense and the Python wrapper together at the same time using the **BUILD_PYTHON_BINDINGS** CMake build flag with CMake's visual GUI. There is a clearly written tutorial for doing so on Windows 10 at the link below.

https://rahulvishwakarma.wordpress.com/2019/08/17/realsense-435i-depth-rgb-multi-camera-setup-and-opencv-python-wrapper-intel-realsense-sdk-2-0-compiled-from-source-on-win10/

This guide is based on Intel's official documentation for building librealsense on Windows from source code:

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

The alternative approach is to build the wrapper from source code on its own after librealsense has already been installed.  The guide at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-665650908 may be useful for doing this on a PC.

If you decide to revert to Python 3.9 so that you can use pip install, a guide for a pip install on Windows that I wrote in the past is at https://github.com/IntelRealSense/librealsense/issues/6402#issuecomment-629763906

closed 2022-02-22 by MartyG-RealSense