realsenseai/librealsense #10220: ARM64 compilation issues in Visual Studio 2022

juamar, 2022-02-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | I don't know | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | Android  |
| SDK Version                            |  2.50 }                          | 
| Language                            |  C#                        | 
| Segment			|  others                           | 

### Issue Description

I'm trying to compile the library to ARM64, because we need this library in order to connect to a Intel Realsense D415 connected to an Android Device. As such, we are building a mobile application using Xamarin, by project requirements. At this time, it will be a big step to just execute a simple example demo code in the mobile app using C#. 

However, when i run the application, in logcat, I see the following error:

[DOTNET] System.TypeInitializationException: The type initializer for 'Intel.RealSense.Context' threw an exception.
[DOTNET]  ---> System.DllNotFoundException: realsense2
[DOTNET]    at Intel.RealSense.Context..cctor()
[DOTNET]    --- End of inner exception stack trace ---
[DOTNET]    at mauiExample.Fragments.RealsenseFragment.OnViewCreated(View view, Bundle savedInstanceState) in C:\Users\juan.avendano\Codes\Thales\Singapur\mauiExample\AndroidProject\Fragments\RealsenseFragment.cs:line 28
[DOTNET] System.DllNotFoundException: realsense2
[DOTNET]    at Intel.RealSense.Context..cctor()

Having this code in a simple fragment:

            try
            {
                Context ctx = new Context(); // ***LINE 28 ***//
                var list = ctx.QueryDevices(); // Get a snapshot of currently connected devices
                if (list.Count == 0)
                {
                    textViewtest.Text = "No device detected. Is it plugged in?";
                    return;
                }
                Device dev = list[0];
            } 
            catch (Exception ex)
            {
                textViewtest.Text = "Error Initializing Camera. See Logcat for error.";
                Console.WriteLine(ex);
                Console.WriteLine(ex.InnerException);
                return;
            }

I was checking my compiled SDK in visual studio, and i see that realsense C# library is telling there is an error with realsense2 C++ project:

Warning: The project 'realsense2' cannot be referenced. Intel.RealSense

[image]

[image]

And I think this is happening because the SDK compilation does not end correctly because there are some headers which needs to use only X86 or X64 architecture, no ARM64.

|Severity   |Code   |Description    |Project    |File   |Line   |Suppression State  |
|-----------|-------|---------------|-----------|-------|-------|-------------------|
|Error (active)|E0035|#error directive: This header is specific to X86 and X64 targets|realsense2|C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\emmintrin.h|20||
|Error (active)|E0035|#error directive: This header is specific to X86 and X64 targets|realsense2|C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\emmintrin.h|28||

It also happends in:
	
realsense2	C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\mmintrin.h	17	
realsense2	C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\pmmintrin.h	16	
realsense2	C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\tmmintrin.h	10	
realsense2	C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\xmmintrin.h	27	

Example:

[image]

I have seen similar issues in other libraries googling, and the problem usually is that the compilation in ARM is not well tested. But maybe there is something i can do to override this issue.

And of course im selecting the correct build:

[image]

Thanks in advance.

MartyG-RealSense, 2022-02-03:
Hi @juamar  I should first highlight that you are the first RealSense user I know of who has actually used VS 2022 with the librealsense SDK and so there is no previous information about how well librealsense works with that VS version.  The most commonly used VS before 22 is VS2019.

Furthermore, librealsense is typically used with Arm devices that have Linux installed (e.g Ubuntu 18.04 on Nvidia Jetson, or the Linux Debian-based Raspberry Pi OS on the Arm-based Raspberry Pi).  Windows is not usually used with librealsense on Arm devices.  so that represents another unknown factor.  

**********

The closest project that I know of to your description is one where the C#-based **Unity** engine was used with a RealSense D435i, Windows, the RealSense Unity compatibility wrapper and the RealSense Android wrapper to connect to an Arm / Android based **Oculus Quest** VR headset.  

https://github.com/GeorgeAdamon/quest-realsense

In this way, C# code in Unity was able to be used with an Android-based device (the Quest headset) and a RealSense camera.

If use of Xamarin is currently a requirement of your project though then it is difficult to see how it could be used with the above project, since the C# code would be being input into Unity.

juamar, 2022-02-04:
I'm still having the same issues in VS2019. I did a new Make-Generate with CMake for this version of VS and ARM64, but this time left the VS Solution generated with .NET Framework 3.5. Previously i changed the configuration to .NET 4.8 as proposed by VS.

[image]

Still, i made a quick read to the quest-realsense, i think this can be usefull, i will also try do it that way, but i think it will be better and more efficient by architecture if i can discard using Java in my xamarin app. This can be an unnecesary step in the app architecture.

I'm not quite sure about compiling the library in linux, because i need Xamarin solution to access it, and for that i need a DLL (Maybe it can be done, but is it really necessary?).

juamar, 2022-02-04:
Just to try something different (and at least check error given when trying to load a X64 library into ARM64) i was trying to generate de C# wrapper for X64 arch. The thing as, althought VS tells it successfully compiles and no Errores givven, i still have the  realsense2 warning icon next to the reference.

[image]

i 've tried to delete it, but when i do, I get the following error:

[image]

MartyG-RealSense, 2022-02-04:
The Windows aspect of the setup complicates things a bit.  If Linux was being used to build librealsense on an Arm device such as Nvidia Jetson or Raspberry Pi then I would typically suggest building the SDK from source code using CMake and including a flag called **RSUSB_BACKEND** set to True.  On Windows though, a CMake source code build is typically done though a CMake **GUI** interface, as described in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

My research did locate a reference about compiling on Windows for Arm64 at https://github.com/IntelRealSense/librealsense/issues/9173

juamar, 2022-02-04:
Hmm... https://github.com/IntelRealSense/librealsense/issues/9173 it's not the same issue. I was hoping to at least reach that error XD

MartyG-RealSense, 2022-02-04:
RSUSB backend is most often used with Linux builds of the SDK, as it bypasses Video4Linux so that a build is not dependent on Linux versions or kernel versions and does not require patching.  On Windows the RSUSB backend flag is usable too, but instead replaces Windows Media Foundation as the backend of the build, as described in https://github.com/IntelRealSense/librealsense/issues/8526

I do not know how to check if the DLL is loaded, unfortunately.

In the RealSense C# wrapper, the files **realsense2.dll** and **Intel.RealSense.dll** are placed next to your application, as described in the C# wrapper documentation.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp#building

Have you got these files in the same location as your application?

If you have installed the full RealSense SDK on Windows then pre-built versions of these files that you can simply copy and paste instead of compiling them yourself are provided at:

**C: > Program Files (x86) > Intel RealSense SDK 2.0 > bin > x64**

[image]

juamar, 2022-02-06:
Okey, your appointment is interesting. I was not taking in consideration that you should have two dlls next two each other. But still i'm not being able to build realsense2.dll for ARM64 architecture, and as Intel.RealSense.dll depends on realsense2.dll, i'm neither being able to build Intel.RealSense.dll for the exposed Arch.

I think this ARM64 build should be able to be done using VS and windows on a X64 host.
Maybe someone else can review this issue or explain why this does not make sense (maybe i'm missing something).

Thanks in advance

MartyG-RealSense, 2022-02-06:
My current understanding of Arm64 support is that it can be installed on Linux for Ubuntu 16.04 and 18.04 but not for Ubuntu 20.04 at the time of writing this, as described in https://github.com/IntelRealSense/librealsense/issues/9040#issuecomment-847442449

There are plans to provide Arm64 package support for Ubuntu 20.04 as part of a future update to the networking API, though I do not have a time estimate for when it will be released.

I do not have information about Arm64 compilation of librealsense on Windows though.

juamar, 2022-02-08:
Because of @MartyG-RealSense recomendation, i started to research about binding Android Libraries into my Xamarin App inpired by this:

https://github.com/GeorgeAdamon/quest-realsense

What i finally did was to use a Xamarin Android library binding with the AAR Intel Realsense Library for Android/Java and call it from  the C# code. It works pretty well, but i don't know if this will be the best options for our needs, since we need to integrate some previous project which was made using the C# Intel Realsense Library, which keeps giving me the same result no matter what architecture, emultor, or android version i use.

[DOTNET] System.TypeInitializationException: The type initializer for 'Intel.RealSense.Context' threw an exception.
[DOTNET] ---> System.DllNotFoundException: realsense2
[DOTNET] at Intel.RealSense.Context..cctor()
[DOTNET] --- End of inner exception stack trace ---
[DOTNET] at mauiExample.Fragments.RealsenseFragment.OnViewCreated(View view, Bundle savedInstanceState) in C:\Users\juan.avendano\Codes\Thales\Singapur\mauiExample\AndroidProject\Fragments\RealsenseFragment.cs:line 28
[DOTNET] System.DllNotFoundException: realsense2
[DOTNET] at Intel.RealSense.Context..cctor()

Even i tried to call the API Version first from Contect, some KISS data, but same result!

Maybe will be nice to open a new issue more focused on this matter rather than the ARM64 compilation (which maybe can be sorted out with linux). 
If what i'm pretending to do it's not meant to be (only the Android/Java library is expected to work there) maybe we will have to think in some adapter.

Thanks in advance!

MartyG-RealSense, 2022-02-08:
It's excellent news that you made significant progress in a Xamarin-based approach, @juamar  :)

I found a C# case at https://github.com/IntelRealSense/librealsense/issues/7037 that seems to be relevant to the Context-related error that you are receiving.

juamar, 2022-02-08:
Okay, so i've made myself with a Debian Virtual Machine, and i'm trying to compile realsense2.dll for ARM64, but i'm not being able to compile, i'm getting this error when executing the following command:

```
$ cmake -S ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_SAHRED_LIBS=true -DBUILD_TOOLS=true -DCMAKE_GENERATOR_PLATFORM=ARM64
CMake Error at CMakeLists.txt:3 (project):
  Generator

   Unix Makefiles

  does not support platform specification, but platform

   ARM64

  was specified.

CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!

```

What am i  missing? maybe i need some other generator?
I have already seen https://github.com/IntelRealSense/librealsense/issues/9173 and https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md.

MartyG-RealSense, 2022-02-08:
I note that your CMake instruction has a typing error: **-DBUILD_SAHRED_LIBS=true**.  It should be SHARED rather than sahred.

juamar, 2022-02-08:
Same error with typo fixed. thanks for the appointment XD

juamar, 2022-02-09:
Now no -S and same error:

```
$ cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_SHARED_LIBS=true -DBUILD_TOOLS=true -DCMAKE_GENERATOR_PLATFORM=arm64
CMake Error at CMakeLists.txt:3 (project):
  Generator

    Unix Makefiles

  does not support platform specification, but platform

    arm64

  was specified.

CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
See also "/home/javendano/codes/librealsense/librealsense/buildArm64/CMakeFiles/CMakeOutput.log".
```

MartyG-RealSense, 2022-02-09:
When performing a CMake build for Arm64 on Jetson, a RealSense team member has previously suggested in https://github.com/IntelRealSense/librealsense/issues/5618#issuecomment-575113313 to put an **override** at the start of the CMake instruction.

**override_dh_auto_configure: dh_auto_configure**

Whilst you are not using Jetson, it may be worth trying the above method.

juamar, 2022-02-14:
Finally, On friday we have decided to proceed using the Xamarin Android library binding with the AAR Intel Realsense Library for Android/Java and desist of compiling the realsense2.dll for ARM64. It was getting pretty hard to reach that point. I was starting to deploy some QEMU Emulators in order to run the compilations, but still no luck.

I had gotten stuck when launching `./scripts/patch-realsense-ubuntu-lts.sh` with some Ubuntu Server image because uvcvideo driver was not being found at some step. So i was trying to install a full Ubuntu Server or a Desktop in QEMU, till time run out.

All compilation procedures are expected to be done in the same device you have the Intel RealSense Camera connected. Trying to get out of that flow can be really tought (unless you use the Android Library).

Thanks @MartyG-RealSense for your help!

closed 2022-02-14 by juamar