realsenseai/librealsense #10224: pyrealsense get_distance() returns 0

AndreaMacri-sys, 2022-02-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Ubuntu 20.04} | 
| Kernel Version (Linux Only)    |  (5.13.0-28-generic)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.49. 0}                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Hi,
when I try to get the depth value from my realsense (both 415 and 435) sometimes the method get_distance() of pyrealsense2 returns 0 in some points.  I deduced that this problem is due to lighting, but I was confident I could remedy using the hole filling filter, like:
```
hole_filling = rs.hole_filling_filter(0)  # I tried also the other hole filtering methods
filtered_frame = hole_filling.process(depth_frame)
```
In realsense-viewer the filter seems to works, but in my code the problem is still presents, even if in a reduced way. 
There is a way to eliminate it entirely?

MartyG-RealSense, 2022-02-08:
Hi @AndreaMacri-sys  The **hole_filling_filter** has three settings (0-2) described in the link below.  The RealSense SDK's default for this filter is '1', whilst you are using '0'.  Does the hole filling improve if you use the default of '1'

https://dev.intelrealsense.com/docs/post-processing-filters#section-holes-filling-filter

AndreaMacri-sys, 2022-02-08:
Thanks @MartyG-RealSense for the quick response,
has i wrote in the comment of the code snippet I also tried settings 1 and 2 (I referred to them as methods instead of settings, sorry), with no improvements.

MartyG-RealSense, 2022-02-08:
I thought you meant that you had tried different types of post-processing filter (as the Decimation and Spatial filters can provide a hole-filling function too).  Thanks very much for the clarification.

You could also try increasing the **Laser Power** value with the instruction **rs.option.laser_power**.  The default value is '150' and the maximum is '360'.

https://stackoverflow.com/questions/65597853/transform-camera-to-world-coordinates

[image]

AndreaMacri-sys, 2022-02-10:
That's also true, this is my full block of filters:
`    def apply_depth_filter(self):
        spatial = rs.spatial_filter()
        temporal = rs.temporal_filter()
        hole_filling = rs.hole_filling_filter(0)
        depth_to_disparity = rs.disparity_transform(True)
        disparity_to_depth = rs.disparity_transform(False)

        filtered_frame = depth_to_disparity.process(self._depth_frame)
        filtered_frame = spatial.process(filtered_frame)
        filtered_frame = temporal.process(filtered_frame)
        filtered_frame = disparity_to_depth.process(filtered_frame)
        filtered_frame = hole_filling.process(filtered_frame)

        self._depth_frame = filtered_frame.as_depth_frame()`

Thanks for the advice about laser power, I'm doing some tests with it. For sure it's improving the results but I still have some zero values

MartyG-RealSense, 2022-02-10:
Does the scene that the camera is observing have dark grey or black areas?  If it does then they will absorb light and so make it more difficult for the camera to analyze those surfaces for depth detail.   Projecting a strong light onto those areas can help to bring out depth detail, as illustrated in https://github.com/IntelRealSense/librealsense/issues/6757 with a black office chair.

In regard to the differences between your RealSense Viewer and Python images, it is also worth bearing in mind that the Viewer typically has the **Spatial** post-processing filter enabled by default, which is set to apply hole-filling with a 16-pixel radius.

[image]

[image]

MartyG-RealSense, 2022-02-14:
If you have two cameras (D415 and D435) but only need one for the data gathering then you could mount both of them together and have them streaming but only use one of them for the data.  The second camera would be projecting its own infrared dot pattern onto the scene in addition to the pattern projected by the data-gathering camera.  The more cameras that are present and projecting a dot pattern, the greater the number of overall dots there will be in the scene and cast over the surface of objects.  This is beneficial for analyzing surfaces for depth.

Alternatively, the same goal could be achieved by adding an external LED pattern projector to the location, as described in Intel's projector white-paper document in the link below.  External projectors have a higher power output than the built-in projector of the camera and a greater projection range.

https://dev.intelrealsense.com/docs/projectors

AndreaMacri-sys, 2022-02-17:
I have read the whitepaper and it looks very interesting, but in a practical application with the second camera I have not noticed any significant improvement. Do you have any external devices to recommend?

MartyG-RealSense, 2022-02-17:
There are not recommendations available for specific external projector products, though Intel do provide some outlines for choosing a projector, such as one that projects between 10,000 and 300,000 dots.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-use-external-projector

MartyG-RealSense, 2022-03-01:
Hi @AndreaMacri-sys  Do you require further assistance with this case, please?  Thanks!

AndreaMacri-sys, 2022-03-01:
As i said in previous comments your advice improved my results, anyway the problem is not solved. I still have some zero values from get_distance, while in the realsense-viewer there are no holes.
If is not possible to completely avoid this behavior you can close the issue.
Thanks.

MartyG-RealSense, 2022-03-01:
You mentioned in your opening message that you thought that the problem was due to lighting.  Do you mean the amount of light in the scene, or a type of artificial light that is present in the scene?  For example, fluorescent lights such as ceiling strip lights can cause disruption to the image because they flicker at frequencies that are difficult to see with the human eye.

MartyG-RealSense, 2022-03-02:
Typically in Python cases involving **get_distance**, RealSense users align depth to color first with **align_to** before retrieving the distance.  An example of this is at https://github.com/IntelRealSense/librealsense/issues/6749#issuecomment-654185205

[image]

MartyG-RealSense, 2022-03-02:
Intel recommend that align is performed **after** application of post-processing filters, otherwise distortion effects such as **aliasing** (jagged lines) may occur.

AndreaMacri-sys, 2022-03-02:
I have already read about that and tried it, but this causes another problem to me. This happens at random frame if I apply align after the filters:
`    self.frames = self.aligned_stream.process(self.frames)
RuntimeError: Error occured during execution of the processing block! See the log for more info
`

MartyG-RealSense, 2022-03-02:
A RealSense team member advised at https://github.com/IntelRealSense/realsense-ros/issues/652#issuecomment-479326583 that **RuntimeError: Error occured during execution of the processing block!** may occur if the filters are not keeping up with the delivery of frames.

Post-processing filters are processed on the computer's CPU instead of the camera hardware and so the more filters that are applied, the greater the processing burden that may be placed on the CPU.  Depth-color alignment is also processed on the CPU and can be processing-intensive too.

Another RealSense team member has suggested in the past at https://github.com/IntelRealSense/librealsense/issues/4468#issuecomment-513485662 that the **Spatial** filter could be removed by RealSense users due to having the largest processing time for a modest improvement.

AndreaMacri-sys, 2022-03-03:
Also removing the spatial filter the runtime error still occurs

MartyG-RealSense, 2022-03-03:
Although applying align after post-processing is the ideal **recommendation**, if applying before filtering was working better for you (albeit having distance = 0 sometimes) then let's go back to that and seek further improvement for the filters.

Are you able to achieve better results if you configure the hole-filling function of the Spatial filter, as described in https://github.com/IntelRealSense/librealsense/issues/2851

[image]

AndreaMacri-sys, 2022-03-04:
In my code it's written as:
`        spatial = rs.spatial_filter(0.5, 20, 2, 4) 
`
Does it means to set smooth alpha = 0.2, smooth delta = 20, magnitude = 2, hole filling = 4, right?

MartyG-RealSense, 2022-03-04:
The official SDK instruction documentation for the Spatial Filter does describe it in that order, yes.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1spatial__filter.html#ada718a2b85aaa3cc642766df50fed77d

[image]

AndreaMacri-sys, 2022-03-07:
Ok thanks, so this part was already correct. Any further idea on the main problem?

MartyG-RealSense, 2022-03-07:
You may be able to achieve some further closure of holes by setting the Advanced Mode **Second Peak Threshold** setting to '0' instead of its default of '325'.  The value of this setting can be defined in a custom json configuration file with the **param-secondpeakdelta** parameter though this may not be possible for you if you are still using one of the SDK's pre-built presets as shown above in https://github.com/IntelRealSense/librealsense/issues/10224#issuecomment-1056947923

[image]

The Second Peak Threshold may also be accessible by accessing the **deepSeaSecondPeakThreshold** instruction with Advanced Mode scripting via **STDepthControlGroup**.

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.STDepthControlGroup.html#pyrealsense2.STDepthControlGroup.deepSeaSecondPeakThreshold

An example of code for setting the value of an STDepthControlGroup option with Python can be found in the SDK's **python-rs400-advanced-mode-example.py** Advanced Mode example program.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-rs400-advanced-mode-example.py#L67

In the quoted line, I would speculate that the **scoreThreshA** instruction would be replaced with **deepSeaSecondPeakThreshold** to set the Second Peak Threshold value.

Perhaps something like this:

`current_std_depth_control_group.deepSeaSecondPeakThreshold = int(0)`

AndreaMacri-sys, 2022-03-07:
I tried, applied by setting the value on STDepthControlGroup's option, but the problem still persists.

MartyG-RealSense, 2022-03-07:
Could you provide an image from the Viewer and from your program in this discussion in order to compare them, please?

MartyG-RealSense, 2022-03-07:
The depth image from your Python program is very good quality, so it is not obvious from the picture why a depth of '0' would be being returned by the center pixel when there is clearly correct-looking depth data in that area.

You have said that 'sometimes' it returns zero, suggesting that the code is okay if it can return a correct depth value the rest of the time.

[image]

Both the background and the object are not ideal materials.  Both are glossy, potentially making them hard to read when reflections are being generated, and black objects will absorb light unless strongly illuminated and be even more difficult to analyze for depth detail.  But if you cast a strong light on the black surfaces to deal with the color-associated problem then you make the reflections worse, and if you do not illuminate the black surfaces then they become more difficult to read anyway.  It is a situation that is difficult to win in.

Given how faded into the background the black object is on the Viewer image, I wonder whether it is actually reading the white background rather than the black object because it cannot tell the two apart.  So considering it from that respect, the Python image may actually be better because it is more able to identify the depth points of the black object.

Does the live depth image fluctuate a lot in the area of the black object?  If so then setting a lower 'alpha' value on the **temporal** filter may reduce the rate of fluctuation.

AndreaMacri-sys, 2022-03-08:
> You have said that 'sometimes' it returns zero, suggesting that the code is okay if it can return a correct depth value the rest of the time.

Yes, it acts as if there are holes. In fact my workaround now is, when I recive a zero, to move on neighboring pixels until I found a non-zero depth value.

> Does the live depth image fluctuate a lot in the area of the black object? If so then setting a lower 'alpha' value on the **temporal** filter may reduce the rate of fluctuation.

Fluctuations vary with the ambient light, in the pictures they are very slight but sometimes i have a lot of spikes on the black object. Your values of smooth alpha and delta on temporal filter seems to help.

MartyG-RealSense, 2022-03-08:
The alpha can be reduced further towards zero, which should further reduce fluctuations.  As long as the camera position is fixed and the black object is not moving then a very low alpha should still be able to produce a good depth image.

AndreaMacri-sys, 2022-03-10:
After a couple of days of testing I found that the best solution for fluctuations is to add a second IR emitter, the temporal filter did not reduce them enough.

Coming back to the zero depth problem, I'm starting to think that is a bug of the get_distance function, I can't find a way to avoid them also if the depth image is quite good

MartyG-RealSense, 2022-03-10:
When get_distance is used in Python on an aligned image, there is a tendency for values to be correct at the center of the image and drift to increasingly incorrect values when reading coordinates further away from the center towards the sides of the image.

An alternative to using get_distance to obtain depth is take the **uint16_t** raw pixel depth and multiply it by the **depth unit scale** of the camera (which is a fixed value of **0.001** on 400 Series cameras), like in the example of such code at https://github.com/IntelRealSense/librealsense/issues/2348#issuecomment-418152047

For example, if the pixel depth was 3600 and it was multiplied by 0.001 then the real world distance in meters would be 3.6 (meters).

AndreaMacri-sys, 2022-03-14:
It works! Also when the get_distance method returns 0 the value in the depth pixel is correct. This makes me think even more that it is a bug in the function.

Thank you for the help and for all the advice!

closed 2022-03-14 by AndreaMacri-sys

MartyG-RealSense, 2022-03-14:
You are very welcome, @AndreaMacri-sys - thanks very much for the update about your success   :)

RyanPaulMcKenna, 2024-03-23:
Just in case this happens to anyone else, zero distances can be the results of poor lighting or poor cable quality and this is just experimental but it appears that the one the hardware is running continuously it performs better. A cold start seems to produce invalid results.