realsenseai/librealsense #10225: RealSense Extrinsics to Unity Matrix4x4

sam598, 2022-02-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { RD400 } | 
| Language                            |  {C#/unity }                          | 

### Issue Description

I am trying to convert extrinsics (from the depth to color camera) into a Unity Matrix4x4 transform. This is what I currently have.

```
Extrinsics extrinsics = new Extrinsics();

using (var profileDepth = Source.ActiveProfile.GetStream<VideoStreamProfile>(Stream.Depth))
{
    using (var profileColor = Source.ActiveProfile.GetStream<VideoStreamProfile>(Stream.Color))
    {
        extrinsics = profileColor.GetExtrinsicsTo(profileDepth);
    }
}

var extrinsicMatrix = new Matrix4x4
{
    m00 = extrinsics.rotation[0],
    m10 = extrinsics.rotation[1],
    m20 = extrinsics.rotation[2],
    m01 = extrinsics.rotation[3],
    m11 = extrinsics.rotation[4],
    m21 = extrinsics.rotation[5],
    m02 = extrinsics.rotation[6],
    m12 = extrinsics.rotation[7],
    m22 = extrinsics.rotation[8],
    m03 = extrinsics.translation[0],
    m13 = extrinsics.translation[1],
    m23 = extrinsics.translation[2]
};
```

The matrix appears to very close to accurate, but the alignment is slightly off, likely in the rotation. 

It appears that the online documentation no longer states what coordinate system the camera extrinsic data is in. So it is very difficult to know how to convert from the RealSense coordinate system to Unity's. This is the information I can find, but the axis are unlabeled.
https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20
https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras

Is there any available documentation about the coordinate system used for extrinsics? Or is there an official example for how to convert from the D400 extrinsic coordinate system to Unity?

MartyG-RealSense, 2022-02-09:
Hi @sam598  https://github.com/IntelRealSense/librealsense/issues/6147 has a detailed discussion about the Unity 4x4 matrix in relation to RealSense cameras.

In regard to the coordinate system and extrinsics, there is information about the sensor axes of the **D435i** model at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md#sensor-origin-and-coordinate-system

My understanding is that the D435i coordinate system also applies to the D455 and L515 camera models.

The D435i image below may also be helpful.

[image]

sam598, 2022-02-09:
Thanks @MartyG-RealSense that information is very helpful.

This script has an example for how to flip the Y axis of Matrix4x4 in Unity which was based on a different stackoverflow answer.
https://github.com/EnoxSoftware/MarkerLessARExample/blob/master/Assets/MarkerLessARExample/Texture2DMarkerLessARExample/Texture2DMarkerLessARExample.cs
https://stackoverflow.com/questions/30234945/change-handedness-of-a-row-major-4x4-transformation-matrix

So the code in Unity now becomes:

```
Extrinsics extrinsics = new Extrinsics();

using (var profileDepth = Source.ActiveProfile.GetStream<VideoStreamProfile>(Stream.Depth))
{
    using (var profileColor = Source.ActiveProfile.GetStream<VideoStreamProfile>(Stream.Color))
    {
        extrinsics = profileColor.GetExtrinsicsTo(profileDepth);
    }
}

var extrinsicMatrix = new Matrix4x4
{
    m00 = extrinsics.rotation[0],
    m10 = extrinsics.rotation[1],
    m20 = extrinsics.rotation[2],
    m01 = extrinsics.rotation[3],
    m11 = extrinsics.rotation[4],
    m21 = extrinsics.rotation[5],
    m02 = extrinsics.rotation[6],
    m12 = extrinsics.rotation[7],
    m22 = extrinsics.rotation[8],
    m03 = extrinsics.translation[0],
    m13 = extrinsics.translation[1],
    m23 = extrinsics.translation[2]
};

Matrix4x4 invertYM = Matrix4x4.TRS(Vector3.zero, Quaternion.identity, new Vector3(1, -1, 1));

extrinsicMatrix = invertYM * extrinsicMatrix * invertYM;
```

This new transform seems to align properly, so unless there is official guidance on how to do this conversion I am closing this issue.

MartyG-RealSense, 2022-02-09:
I do not have further guidance to add to your procedure above.  Thanks very much for the update and for sharing your successful solution with the RealSense community!