realsenseai/librealsense #10226: Distortion on camera's depth data

MichaelPan0905, 2022-02-09:
https://user-images.githubusercontent.com/25095900/153148530-87db1d44-727a-42a3-8452-ce6f097b9a52.mp4

| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435|
|**Serial Number**|030522071656|
|**Firmware Version**|05.13.00.50|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|030323023196|
|**Firmware Update Id**|030323023196|

Please provide a description of the problem
Hello there, I've got a terrible distortion on my camera's depth data, as it's shown in the video, which the camera is right torwards a flat white wall.  Calibration didn't help any. How can I deal with this situation？I've got two cameras both have this problem. Thanks.
Besides, I've got the same problem in [#10182](https://github.com/IntelRealSense/librealsense/issues/10182), but the method you've provided does not help, so I used the Dynamic Calibration to calibrate those cameras(But can not fix the distortion).

MartyG-RealSense, 2022-02-09:
Hi @MichaelPan0905   It looks as though the wall may be curving inwards (a concave, or 'pincushion' distortion).  An example of this effect was discussed in detail at https://github.com/IntelRealSense/realsense-ros/issues/1534 and the solution in that particular case was to reset the camera's calibration to its factory-new defaults.  Instructions for doing so in the RealSense Viewer can be found at https://github.com/IntelRealSense/librealsense/issues/10044#issuecomment-997946357

To write the reset to the camera hardware, enable the depth stream first before using the reset option.  After the reset has been performed, the 'Write Data' button button will be unlocked, enabling the reset calibration to be saved to the camera.

MichaelPan0905, 2022-02-09:
@MartyG-RealSense Thanks for responding！I'll try this later.
 By the way, I'm sure that it's not the wall's problem, because I've tried the question camera on some wall else and the curve  still there, and I've also tried another normal camera on that wall in the video and it was flat.

MartyG-RealSense, 2022-02-09:
Yes, the camera calibration rather than the wall itself is typicallu the cause of a curved wall on the depth image.

MichaelPan0905, 2022-02-10:
Hi @MartyG-RealSense , I've got three more isuses for this question:
First, OCC couldn't work, I find that when calibration started, the "Emitter Enabled" which should be "laser" will always be truned to "off", which I think is the reason why occ failed.
[image]
Second, Restore factory in Calibration Data option does not seem to work. I enable the depth stream, open the Calibration Data option, click the restore factory button, and then click the Write Table button, and the option window closed, nothing happened.
Third, using the Dynamic Calibration is succeeded, but it didn't worked. There is still a  concave on the wall, nothing changed.

MartyG-RealSense, 2022-02-10:
There is a known bug in the OCC calibration tool in **2.50.0** that turns the laser off.  Intel are working on fixing it, probably in time for the next SDK version, and recommend using the 2.49.0 version in the meantime as this does not have the bug.

MichaelPan0905, 2022-02-14:
Roger that~ I'll try 2.49.0

MichaelPan0905, 2022-02-14:
2.49.0 is ok. And the concave came from wrong focal length, so when calibrating it needed to choose "OCC Extended" mode and check the "Adjust both sides focal length".
[image]

MartyG-RealSense, 2022-02-14:
It's great to hear that you achieved a good outcome, @MichaelPan0905 - thanks very much for the update and for sharing your solution with the RealSense community!

MichaelPan0905, 2022-02-14:
@MartyG-RealSense Thanks for your help~ So I'm going to close this issue.

closed 2022-02-14 by MichaelPan0905