realsenseai/librealsense #10227: Cuda Cmake with vs2019

arothenberg, 2022-02-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D435  | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC |
| SDK Version                            | 2.32                      | 
| Language                            |  C++                      | 
| Segment			|desktop app  | 

### Issue Description
I'm building the sdk with the Cuda flag and I ran into a problem where I had to manually add the 
Cuda_toolkit_root_dir - 
I downloaded the latest Cuda and the path is
C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v11.6

I was also missing(according to cmake) the Cuda sdk location. After some research I found that "samples" was the sdk location so for
Cuda_sdk_root_dir  I added 
C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v11.6/extras/CUPTI

The files generated but I got this warning 👍 CMake Warning (dev) in CMakeLists.txt:
_```
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.
```_ 
I researched further and was suggested
cmake --help-policy CMP0104 - which showed a large text block about deprecation after v3.18... I didn't understand the implications. 

Does this warning mean Cuda has not been enabled with realsense? 

Thanks.

MartyG-RealSense, 2022-02-09:
Hi @arothenberg  CUDA support requires the computer to have an **Nvidia** graphics chip or video card.  Nvidia Jetson boards are an example of a computing device that has a built-in Nvidia GPU.  If your computer does not have Nvidia graphics then CUDA support cannot work.

For computers without Nvidia graphics hardware, the RealSense SDK has an alternative system called **GLSL Processing Blocks** to offload processing from the CPU to the GPU.  GLSL is 'vendor neutral' and so can work with any GPU brand, though improvements may not be noticable when used with a low-end device.  https://github.com/IntelRealSense/librealsense/pull/3654 has a very good pros and cons analysis of GLSL processing blocks and when it can be used.

There is also a C++ GLSL example program at https://github.com/IntelRealSense/librealsense/tree/master/examples/gl

arothenberg, 2022-02-09:
i double posted this sorry ill got the op

MartyG-RealSense, 2022-02-09:
It is very rare on this forum to have cases where CUDA is being used with librealsense on a PC instead of an Nvidia Jetson.  https://github.com/IntelRealSense/librealsense/issues/4816 is such a case though, where a RealSense team member advises to first follow the CUDA installation instructions in the official Nvidia documentation before building librealsense with CUDA support using the CMake build flag **-DBUILD_WITH_CUDA=true**

If CUDA and the librealsense SDK's CUDA support are both installed sucessfully then CUDA graphics acceleration should be applied automatically.

The only way I know of to check whether CUDA is being applied during the running of a RealSense application is to check CPU percentage usage to see whether the % usage is very low when pointcloud generation, RGB color conversion or depth-color alignment is being performed, as described athttps://github.com/IntelRealSense/librealsense/issues/6302