realsenseai/librealsense #10230: RTOS realsense D4XX

pltrd, 2022-02-10:
Hello ! I am working on a D455 realsense on Linux OS. I want to optimize the real time fps using a cloud/customer network with python ZMQ. For the moment I do not get the best result I want so I wanted to know if some of RTOS are available and usable with realsense camera. I'm a newbie so I dont really know which one use for resolve my issue. Could you provide me some suggestions ? 
Thanks a lot for your help !

MartyG-RealSense, 2022-02-10:
Hi @pltrd  The only case that I know of where a RealSense user was apparently using a Real Time Operating System (RTOS) is at https://github.com/IntelRealSense/librealsense/issues/2197 where an RTOS called **Litmus** was apparently being used.  This is likely referring to LITMUS-RT at https://www.litmus-rt.org/

In regard to ZMQ, someone created a GitHub fork of Intel's **EtherSense** Python ethernet networking project that they adapted to make use of ZMQ.

**EtherSense (original)**
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/ethernet_client_server

**EtherSense (ZMQ fork)**
https://github.com/martingasser/EtherSense

Intel developed a more modern RealSense networking system after EtherSense that is detailed in the guide linked to below.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

pltrd, 2022-02-10:
Thanks for your answer Marty, 
According to EtherSense, Do you know if I will get delay between client and server ? If in client I send frames at 15 fps for example, will I get approximately the same number of fps in server part ? By frame, I mean the pointcloud, the depth or RGB each frame. Another question is when I use 2 cameras which has their client, The number of fps in the server is equal to the minimal number of fps of a client ? Sorry if it is not clear ^^ If I get 14 fps and 5 fps in the client, will I get 5 fps in the server ?

closed 2022-02-25 by MartyG-RealSense