realsenseai/librealsense #10231: IDE's Type Correction - Intellisense - Auto Complete now working well

Maynafiesta, 2022-02-10:
I'm rookie on Python and Realsense things. My SDK is up to date now. When i try to code, IDE's helper programs are not working well. I can compile and run my codes, but it is hard to code in that way.
Neither Visual Code Python nor PyCharm would help me. They are all same. I think that installation of SDK is completed well. I tried several time and nothing were wrong on installation. 

[image]

This is basic CTRL + Space in pycharm
[image]

This is CTRL + Shift + Space in Pycharm and there is no recommendation. 
This code blocks are working and i can run these python scripts.

Maynafiesta, 2022-02-14:
hi @MartyG-RealSense , i checked that page and these steps are looking like okay. Pyrealsense2 is listed in library root file. And i guess this situation is not related to IDE, because Visual Studio Code is also same. These IDE's are okay with running code but they are not auto complete. Thanks for your reply.

MartyG-RealSense, 2022-02-14:
I cannot find a similar past case relating to librealsense.

In Pycharm, does it still show no suggestions if instead of using the Ctrl-Shift-Space shortcut for smart completion, you instead select **Code > Code Completion > Type-Matching** from the main menu?

Maynafiesta, 2022-02-14:
Sir i just tried what you said and, no suggestion again. But i see that i missed a step while installation. it is written that, 
"5.Alternatively, copy the build output (librealsense2.so and pyrealsense2.so) next to your script."
then i found related files and copied these 2 files to same folder with my script. The result is same, no suggestion. 
Do you mind if you can tell me that, which files have to be in same folder with my scripts?

Maynafiesta, 2022-02-14:
thanks a lot for your well explained replies

Maynafiesta, 2022-02-14:
thanks a lot for your well explained replies, i will explain here when i find a way.

Maynafiesta, 2022-02-15:
@MartyG-RealSense hi sir, i got something, auto complete works on some words. as shown below.

[image]

here 'rs.' is workin

[image]

here 'frames.' is not working as previous screenshots. My camera order is still on ship and i am exercising with .bag sample files during ship time. So do you thing this can be the reason that working on file not real camera?

MartyG-RealSense, 2022-02-21:
Hi @Maynafiesta  Do you require further assistance with this case, please?  Thanks!

Maynafiesta, 2022-02-21:
@MartyG-RealSense hi sir, and yes i still need assistance. i looked your last image of piece of code. i am facing this situation even in simple example codes in official sdk examples.

MartyG-RealSense, 2022-02-21:
How much memory in GB does the computer that you are using have, please?  Because auto complete apparently stores its database in memory, some have found that it worked on computers with a good amount of memory but not a low-memory computer.

Maynafiesta, 2022-02-21:
My pc has 16 gb ram sir. Thanks for your helps. Problem is not solved but it is not fatal for now. Should i close the comment?

MartyG-RealSense, 2022-02-21:
I have researched the subject deeply but cannot find a cause that is specifically related to RealSense.  So I would recommend closing, unfortunately.

closed 2022-02-22 by Maynafiesta

JeffR1992, 2022-05-22:
I too am having issues with no intellisense for `pyrealsense2` in VS Code (I haven't tried other editors though). A colleague of mine has also said that it doesn't work for them either. However, intellisense works well for other libraries such as Numpy and OpenCV. It sounds like a solution wasn't found for this issue yet.

MartyG-RealSense, 2023-05-10:
Hi everyone,

Regarding the issue of Intellisense not providing auto-complete for all pyrealsense2 instructions, a RealSense user submitted a solution at https://github.com/IntelRealSense/librealsense/pull/11624