realsenseai/librealsense #10232: Basic questions on depth quality tool and fails

arothenberg, 2022-02-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D435  | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC |
| SDK Version                            | 2.32                      | 
| Language                            |  C++                      | 
| Segment			|desktop app  | 

### Issue Description
The realsense depth quality  calibration tools.

I am restoring a project from last year that uses the  realsense sdk and camera.
Since it's been awhile I decided to recalibrate the camera. I set it to factory defaults through the depth quality tool. 
https://github.com/IntelRealSense/librealsense/issues/10182

It took me a while to get the on board chip calibration to work , trying different lighting and distances, but eventually it did work inconsistently but when it worked it consistently gave a good  rating. 
I assume this is good enough? 
The same with the tare calibration. It was hard to get a reading but when it worked it 
looked close to the original.

I could not get the focal length calibration to work after many trys and many set ups.
"failed to extract target information from frames"

Also, I couldn't figure out how to get into advanced mode.

I am going to try dynamic calibration next and see if that goes better.

Is there a good reason to get focal length calibration working?  I am working with relative distances in my project.
Is there a reason i would need advanced mode? I noticed a hardware reset option in advanced mode in your screen shots. 

If I successfully dynamically calibrate is there then still a reason to use the depth quality calibration tools?

thanks.

Thanks.

arothenberg, 2022-02-11:
I did test and the problem is there are no conditions yet to get it to work(finish), it always ends in error. 
I am using a target from here
https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

arothenberg, 2022-02-11:
focal length calibration ends with "failed to extract target information"

arothenberg, 2022-02-11:
Also OT 
I found an example using realsense in a glfw repository. It's example#2 so if you ever get questions about implementing glfw 
in a realsense app loop you can refer to this. I was surprised. 

https://cpp.hotexamples.com/examples/-/-/glfwCreateWindow/cpp-glfwcreatewindow-function-examples.html

arothenberg, 2022-02-12:
Excellent. I'll try that latter.
So I cant just print the target for the OCC on 8.5x11? Is there a good target for the OCC that I can just print and use?
Thanks.

MartyG-RealSense, 2022-02-12:
The only version of the OCC target image available is the 12.89 x 17.12 cm one in the paper.  It should not matter if the image is stretched to fill a US Letter sheet's dimensions, as the paper states that "the exact nature of the texture is not critical, as long as it is semi-random and fairly “noisy” with high spatial frequencies".

closed 2022-02-25 by MartyG-RealSense