realsenseai/librealsense #10233: D435i Pose data?

MyOtherNamesWereTaken, 2022-02-11:
Camera Model | { D435i }
-- | --
Firmware Version | (05.13.00.50)
Operating System & Version | { Linux (Ubuntu 14/16/17)
Kernel Version (Linux Only) | (5.11.0 -38-generic )
Platform | PC
Language | {python}
Segment | {Robot/Smartphone/VR/AR/others }

### Issue Description
Is it possible to get the pose data via Open CV for the D435i?

MartyG-RealSense, 2022-02-11:
Hi @MyOtherNamesWereTaken  The D435i model does not have built-in support for pose data.  The pose of objects observed by the camera can though be obtained in OpenCV with a **SolvePNP** algorithm.  

OpenCV's official documentation has a Python SolvePNP pose estimation tutorial at the link below.

https://docs.opencv.org/4.x/d7/d53/tutorial_py_pose.html

Another example of an OpenCV SolvePNP object pose tutorial is here:

https://www.pythonpool.com/opencv-solvepnp/

MartyG-RealSense, 2022-02-11:
In the linked-to project, have you set it to the Windows user account path of your own computer instead of using the script's default user account path that relates to the project creator's computer?

https://github.com/alishobeiri/Monocular-Video-Odometery/blob/master/test.py#L9

MyOtherNamesWereTaken, 2022-02-11:
The issue is that i dont have any pose data to work with. Im trying to generate those.

MartyG-RealSense, 2022-02-11:
I do not know a way to generate a text file of pose data in an OpenCV application using a RealSense D435i camera, unfortunately.

If you are able to use ROS and rtabmap_ros SLAM navigation then there is a project in the link below that can calculate and visually display the relative position of a D435i camera.

https://shinkansan.github.io/2019-UGRP-DPoom/SLAM

w-1016-Papman, 2022-02-11:
> 

Hi，Excuse me, I asked a question about point cloud, I sincerely hope you can help me answer it, thank you very  much.

[https://github.com/IntelRealSense/librealsense/issues/10234#issue-1132482954](url)

MartyG-RealSense, 2022-02-11:
Hi @w-1016-Papman  I have posted a reply on your case.  Thanks!

closed 2022-02-26 by MartyG-RealSense