realsenseai/librealsense #10234: Point cloud and display problems during 3D reconstruction

w-1016-Papman, 2022-02-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 }                                                   | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10                                                 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC                                                         |
| SDK Version                            |  { legacy / 2.50.0 }                          | 
| Language                            |  python                          | 
| Segment			|  others                          | 

### Issue Description
I am using python trying to get point cloud data by depth frame and display or export a ply file，and this is the code i wrote, now i have got the verts and textcoords, Below is my code.
`       points = pc.calculate(depth_frame)
        pc.map_to(color_frame)
        v = points.get_vertices()
        t = points.get_texture_coordinates()
        verts = np.asanyarray(v).view(np.float32).reshape(-1, 3)  # xyz
        textcoords = np.asanyarray(t).view(np.float32).reshape(-1, 2)  # uv
`
But in fact, I'm not very clear about the meaning and role of these two parameters (verts and textcoords) (I want to figure them out), now I don't know what to do next, I just simply want my point The cloud map can be displayed first, and then do other plans.I think the next step should be a matching operation between coordinates (from my guess).
So,I hope someone can help me answer my doubts, thank you very much

w-1016-Papman, 2022-02-11:
My device is D455. And can I visualize point clouds with numpy arrays?

MartyG-RealSense, 2022-02-11:
Hi @w-1016-Papman  My understanding is that **verts** and **texcoords** represent two separate numpy arrays (named 'v' and 't' in your code), and that the 'verts' array stores the xyz pointcloud coordinates in array 'v' whilst the 'texcoord' array stores the texture coordinates (uv map) for the point cloud in array 't'.

The RealSense SDK's Python example program **opencv_pointcloud_viewer.py** demonstrates the generation and viewing of a live pointcloud via numpy array storage using the technique that you highlighted in your question.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py#L310

w-1016-Papman, 2022-02-11:
> 嗨@w-1016-Papman 我的理解是**verts**和**texcoords**代表两个单独的 numpy 数组（在您的代码中命名为“v”和“t”），并且“verts”数组将 xyz 点云坐标存储在数组“v”中而'texcoord'数组将点云的纹理坐标（uv贴图）存储在数组't'中。
> 
> RealSense SDK 的 Python 示例程序**opencv_pointcloud_viewer.py**演示了使用您在问题中强调的技术通过 numpy 数组存储生成和查看实时点云。
> 
> [https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py ](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py) [https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py #L310](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py#L310)

I'm sorry to come to you again. Thank you very much for the link, that helped me a lot, but I found a phenomenon,
Why is the value of the textcoords array all zeros after I align the depth frame with the color frame and then execute the following part of the code (at least from my run window it has all zeros in its internal data)
`points = pc.calculate(depth_frame)`
`pc.map_to(color_frame)`
`v = points.get_vertices()`
`t = points.get_texture_coordinates()`
 `verts = np.asanyarray(v).view(np.float32).reshape(-1, 3)  # xyz`
 `textcoords = np.asanyarray(t).view(np.float32).reshape(-1, 2)  # uv`
 `print(textcoords)`
I just added two lines of code:
`algin_to = rs.align(rs.stream.color)`
`algin_frames = algin_to.process(frames)`
And I also found that even if I don't have an alignment operation, if I don't add the decimation filter, the output of the running window also shows zero, But after I add the decimation filter code, it shows some non-zero data, I'd love to know why this is
`decimate = rs.decimation_filter(2)`
`depth_frame = decimate.process(depth_frame)`

MartyG-RealSense, 2022-02-11:
I believe that you do not need to use depth-color alignment when generating a pointcloud with the **pc.map_to()** method, as it is already mapping beween depth and color.  Alignment would be used if a pointcloud was being generated with **rs2_deproject_pixel_to_point**, the other main method of pointcloud generation other than points.Get_vertices()

w-1016-Papman, 2022-02-11:
But why does the decimation filter affect the value change? I see his reduced resolution, I think it should be good for noise reduction, but why does it affect the internal value of textcoords?

MartyG-RealSense, 2022-02-11:
Intel's post-processing guide states about the decimation filter that after the decimated frame is produced, the frame intrinsic parameters are recalculated to compensate for the resolution changes.

https://dev.intelrealsense.com/docs/post-processing-filters#section-decimation-filter

closed 2022-02-16 by w-1016-Papman