realsenseai/librealsense #10235: .bag file convert to csv file

LeeChangwon-remote, 2022-02-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455  | 
| Firmware Version                   | 4.9.253-tegra| 
| Operating System & Version |   Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  18.04                                      | 
| Platform                                 | NVIDIA Jetson (Nano)  |
| SDK Version                            |  SDK-2.50.0                          | 
| Language                            |  Python                          | 
| Segment			|  Robot                        | 

### Issue Description
Hello, everyone.
I entered rs-convert command but problem called segmentaion fault (core doped) occurs when converting the bag file into a csv file. 
Can you tell me the reason and solution?
[image]

LeeChangwon-remote, 2022-02-11:
Thank you for your quick response. @MartyG-RealSense .
[image]
As you can see, there's a bag file where you run rs-convert.

MartyG-RealSense, 2022-02-11:
My reason for suggesting changing to the usr/local/bin file and putting the bag there, instead of being in python/examples like the image above, was that inputting the **rs-convert** command could be causing the pre-built version of the program in usr/local/bin to run, but it looks for your bag file in the usr/local/bin folder and can't find it because it is in python/examples.

LeeChangwon-remote, 2022-02-12:
I misunderstood what you said.
I moved the bag file to /usr/local/bing and performed it, but there is still a segmentation fault error without a separate error message. I looked for other issues with GitHub, but most of them are rsusb backend issues, but I still can't find a solution because they put options on the first installation.
[image]

MartyG-RealSense, 2022-02-12:
A small number of RealSense users have reported problems with bag to csv conversion in the past when using rs-convert, whilst bag to PNG image extraction worked for them.   Examples of this problem can be found at https://github.com/IntelRealSense/librealsense/issues/6221 and also the link below. 

https://support.intelrealsense.com/hc/en-us/community/posts/360042055394/comments/360011063953

Similarly, problems with bag to ply conversion have been reported, like in https://github.com/IntelRealSense/librealsense/issues/4171#issuecomment-631293971

If csv or ply conversion issues occur in rs-convert then there is not a known fix for it, unfortunately.  If your main goal is to achieve a csv file and the bag recording is not important to you then the **rs-capture** tool can dump depth and color frames to csv, as described at https://github.com/IntelRealSense/librealsense/issues/1485#issuecomment-380157114

For RealSense users who want to extract image files instead of csv or ply, there is an alternative extractor to rs-convert that was created by a RealSense user that is called **rs_bag2image**

https://github.com/UnaNancyOwen/rs_bag2image

MartyG-RealSense, 2022-02-13:
The depth data would be in the **/device_0/sensor_0/Depth_0/image/data** topic.

I cannot find descriptions for the other highlighted topics, unfortunately.

LeeChangwon-remote, 2022-02-14:
Okay, Thank you

closed 2022-02-14 by LeeChangwon-remote