realsenseai/librealsense #10236: No color frame with rs-converter

BH4DKC, 2022-02-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | d 435                                                      | 
| Firmware Version                   | 05.13.00.50                                            | 
| Operating System & Version |   win 10                                                  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC                                                         |
| SDK Version                            |  2.50.0                                                  | 
| Language                            |  c++                          | 
| Segment			|  Robot                          | 

### Issue Description
I am trying to using the rs-convert to convert the recorded .bag file to csv files that include color map and depth map. However, I am unable to have the color map converted, can some one help me with the code?

[image]

MartyG-RealSense, 2022-02-11:
Hi @BH4DKC  Your conversion instruction looks fine to me.

`rs-convert.exe -c -i C:\Users\zxm171530\Downloads\PointCloudTest\CameraTest\book_test.bag -v C:\Users\zxm171530\Downloads\PointCloudTest\CameraTest\book_test`

May I begin by asking a couple of questions, please.

1.  Was the bag file recorded in the RealSense SDK wth a tool such as the RealSense Viewer?  If it was originally recorded in ROS using **rosbag record** then it may have incompatabilities with being accessed in a RealSense SDK tool such as rs-convert.  

2.  Can you confirm that this bag file has color stream data recorded in it.  If the color stream was enabled before recording was activated then it should be included in the bag file after the recording process was activated.  If there was not a color stream already enabled when recording began though then color frames would not be included in the bag file.

3.  Are you able to extract color frames if you use the 'Outdoor scene with D435i' bag file from the RealSense **Sample Data** page instead of your own bag?

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

4.  Are you able to extract color frames to a csv from your own bag file if you copy it into the SDK **tools** folder so that it is in the same folder as rs-convert.exe and then run this simplified conversion instruction:

`rs-convert.exe -c -i book_test.bag -v book_test`

5.  Are you able to extract color frames if you change -v to -p to extract PNG image files instead.

BH4DKC, 2022-02-12:
Thanks for responding, regards to your questions
1. The bag file is recorded with the RealSense Viewer

2. How do I confirm whether the bag file contain color stream data? All I know is that when I replay the bag file in RealSense viewer, it has RGB images as well as the stereo images.

3. I have not tried move the bag file to tool folder nor converting to PNG files yet. However, I did try move the rs-convert tool from tool folder to the bag file location, and it is still only extracting depth data. 

Regards to rest of the quesionts, I am not in my office now, and once I am in my office on Monday, I can try it and let you know.

MartyG-RealSense, 2022-02-12:
Thanks very much for the information and for testing the bag in the tools folder.  If the bag file has an RGB color stream when played back then it has extractable color frames.

I look forward to your further test feedback.  Good luck!

BH4DKC, 2022-02-14:
Thanks for waiting for the response. 

So, I tested for the rest of the problem, and here are the results:

1. I tried to use the d435i_walking.bag to extract the csv file, and attached are the program log. I tried it once with color frame only, and once with both color frame and depth frame. The 08_51_06 log is the color frame only, and the 08_53_42 log is with both.

2. I have also tried to extract to PNG files, and that seems to be working  correctly, I got both color image and depth image with the camera.

Please let me know what other tests you might want me to run in order to fix the problems. Thanks again for your help.

[2022-02-14-08_51_06.log](https://github.com/IntelRealSense/librealsense/files/8061688/2022-02-14-08_51_06.log)
[2022-02-14-08_53_42.log](https://github.com/IntelRealSense/librealsense/files/8061768/2022-02-14-08_53_42.log)

MartyG-RealSense, 2022-02-14:
I have not seen that particular error **(dispatcher.cpp:35) Dispatcher [000002871BFF67E0] exception caught** before.  A change to dispatcher code was made in SDK 2.48.0, so it may be worth installing **2.47.0** to see whether the problem still occurs with rs-convert in that earlier version.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.47.0

When the SDK is installed on Windows using the automated installer file **Intel.RealSense.SDK-WIN10** in the 'Assets' file list of the SDK Releases page then it completely overwrites the previous SDK installed using this method.  This makes it easy to return to 2.50.0 by downloading and running the installer file for 2.50.0 if a test of rs-convert with the 2.47.0 SDK does not make a difference.

[image]

BH4DKC, 2022-02-14:
Thanks for the response. I tried to use 2.47.0, the log did not generate any errors. However, I still do not have the color map in the csv files.

I checked the coding for converter-csv under converters in tools folder. If I am reading the converter file correctly, it seems that the csv converter does not convert color frames to RGB data? Or did I read the wrong file? Below is the link to the file I read.

https://github.com/IntelRealSense/librealsense/blob/master/tools/convert/converters/converter-csv.cpp

MartyG-RealSense, 2022-02-14:
The documentation that accompanies rs-convert states that the **-v** csv export setting supports depth, color, IMU and pose formats.  I cannot see any color export lines in the csv export script either though.

https://github.com/IntelRealSense/librealsense/blob/master/tools/convert/converters/converter-csv.hpp#L77-L79

BH4DKC, 2022-02-14:
Yes, so I think I might just need to find a 3rd company tool to convert the .png file to rgb values and then process them with the depth data that way.

MartyG-RealSense, 2022-02-20:
Hi @BH4DKC  Do you require further assistance with this case, please?  Thanks!

BH4DKC, 2022-02-20:
Thanks for confirming with me. I do not think I need further assistance. I just wrote my own code to convert png file from the rs-converter to the RGB csv file and linked them together using the metadata. Thanks again for help me with the issue.

Best Regards,
Zhiqi Mao
Engineer In Training
TX EIT 59174
AMImechE

MartyG-RealSense, 2022-02-20:
It's excellent news that you developed a solution, @BH4DKC - thanks very much for the update.  :)  As you do not need further assistance, I will close this case.  Thanks again!

closed 2022-02-20 by MartyG-RealSense