realsenseai/librealsense #10237: Failed to run examples

daniele-gusmini, 2022-02-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | last one | 
| Operating System & Version |  MacOS  | 
| Kernel Version (Linux Only)    |  - | 
| Platform                                 | Mac  |
| SDK Version                            |  - | 
| Language                            |  C/nodejs }                          | 
| Segment			|  others                     | 

### Issue Description
Hello everyone!
I tried to compile Librealsense both on node.js directly (npm procedure) and compiling manually the code. The build succeeded in both case but I cannot run the examples.
If I do not connect the camera the code (ie rs-hello-realsense) states "No device connected" as expected
If I connect my d455 it provides this error:

`Assertion failed: (list_empty(&darwin_cached_devices)), function darwin_init, file darwin_usb.c, line 603.
zsh: abort      ./rs-hello-realsense`

I cannot find anything related, could you help me?

Thank you

MartyG-RealSense, 2022-02-14:
Hi @daniele-gusmini  May I first ask whether you are using MacOS **Monterey** as this is currently known to have issues with USB devices such as RealSense cameras.  The subject is discussed in detail at https://github.com/IntelRealSense/librealsense/issues/9916

daniele-gusmini, 2022-02-14:
Hi @MartyG-RealSense 
Yes, I’m using Monterey 
Thanks for the help, I try to follow the discussion
If I’m not wrong there isn’t a solution for now right?

Thanks

Daniele

closed 2022-02-26 by MartyG-RealSense