realsenseai/librealsense #10239: 11111

hsyzhxy, 2022-02-15:
Required Info |  
-- | --
Camera Model | D415
Firmware Version | 05.12.07.150
Operating System & Version | 20.04.1-Ubuntu
Kernel Version (Linux Only) | 5.11.0-38-generic
Platform | NUC
SDK Version | { legacy / 2.. }
Language | C++
Segment | others }

hsyzhxy, 2022-02-15:
[image]

MartyG-RealSense, 2022-02-15:
Hi @hsyzhxy  There are two separate exposure settings for the depth and RGB sensors.  These exposure settings also use different scales for their exposure values.  Depth uses large exposure values such as '8500', whilst RGB uses small exposure values such as '156' (or the '100' that your script uses).

Your script is accessing the **depth** sensor exposure but setting it to a small value typically used for RGB exposure, resulting in the image that you experienced above.  

If you wanted to control RGB exposure then you can set up a device query with the index number '1' to access the RGB sensor (the depth sensor has an index of '0') and then set a small exposure value.

https://support.intelrealsense.com/hc/en-us/community/posts/360052346893/comments/360014794613

hsyzhxy, 2022-02-16:
Thanks a lot!
I want to  control depth exposure, Is there any example of C++scripting for depth?

MartyG-RealSense, 2022-02-16:
You should be able to use the C++ script for RGB exposure in the link above and just change the sensor index number from '1' to '0' to access depth sensor exposure.  Something like this:

```
auto depthSensor = profile.get_device().query_sensors()[0];
depthSensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);
depthSensors.set_option(RS2_OPTION_EXPOSURE, 7000);
```

hsyzhxy, 2022-02-17:
Thanks， and what is the range for depth exposure when I set?

MartyG-RealSense, 2022-02-17:
The RealSense Viewer program's manual depth exposure option allows an exposure range between 1 and 165000 .

When setting manual exposure, certain ranges of values may though cause the chosen FPS speed to decrease, as described at 
https://github.com/IntelRealSense/librealsense/issues/1957#issuecomment-400715579

roelofvandijk, 2022-02-19:
You can also request the option range for each option.
https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1options.html#a1fd95ff915c351e2cf2382b63ca7a28b

Example for the laser power to max:
```
auto depth_sensor = selected_device.first<rs2::depth_sensor>();
auto range = depth_sensor.get_option_range(RS2_OPTION_LASER_POWER);
depth_sensor.set_option(RS2_OPTION_LASER_POWER, range.max);
```

closed 2022-03-07 by MartyG-RealSense