realsenseai/librealsense #10241: RS2_OPTION_AUTO_EXPOSURE_LIMIT operation

zwjgit, 2022-02-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (0.5.12.07.150) | 
| Operating System & Version |   Ubuntu 16 | 
| Platform                                 | PC |
| SDK Version                            |  2.5 }                          | 
| Language                            |  C }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
when i use “sensor.set_option(RS2_OPTION_AUTO_EXPOSURE_LIMIT,5000);”  to set Stereo Module exposure limit, terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  object doesn't support option #85

MartyG-RealSense, 2022-02-16:
Hi @zwjgit  There have only been three previous cases in which the **RS2_OPTION_AUTO_EXPOSURE_LIMIT** instruction has been used by RealSense users and all three of them experienced the **object doesn't support option #85** error.  There is no known solution and so I would recommend to not use this instruction, unfortunately.

closed 2022-02-16 by zwjgit

HViktorTsoi, 2022-08-22:
You need to update the latest Firmware, auto exposure limit is not supported in 0.5.12.07.150

AndreV84, 2023-01-24:
@HViktorTsoi @MartyG-RealSense 
folks tried with the latest firmware
[Signed_Image_UVC_5_14_0_0.bin](https://www.intelrealsense.com/wp-content/uploads/2022/12/Signed_Image_UVC_5_14_0_0.zip) but it won't work.
Any ideas?
Thanks

MartyG-RealSense, 2023-01-24:
@AndreV84  As mentioned above, I have not encountered any cases where the **RS2_OPTION_AUTO_EXPOSURE_LIMIT** has worked and not generated the `doesn't support option #85` error.  However, you could try toggling the limit on first with the **RS2_OPTION_AUTO_EXPOSURE_LIMIT_TOGGLE** instruction before setting the limit value, using a line like the Python one below.

`sensor.set_option(rs.option.auto_exposure_limit_toggle, True)`

MartyG-RealSense, 2023-01-24:
I recalled that there were recently problems reported with the related **RS2_OPTION_AUTO_EXPOSURE_LIMIT_TOGGLE** option at https://github.com/IntelRealSense/librealsense/issues/11230 when using firmware 5.14.0.0.  A RealSense team member advised that the option was deprecated in 5.14.0.0 and the previous firmware **5.13.0.50** should be used instead.

lfescobarc, 2023-02-17:
I am having the same issue, tried to downgrade the firmware, still doesn't work, searched for version 2.54, but it has not been released yet. Any solution?

MartyG-RealSense, 2023-02-18:
Hi @lfescobarc  The **development** source-code branch of librealsense at the link below becomes the latest 'master' version of the SDK when released, which will likely be numbered 2.54.0.

https://github.com/IntelRealSense/librealsense/tree/development

AndreV84, 2023-03-23:
I managed to deploy to jetson dockerized aitoware that runs autoware rviz
https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/
 but I do not see any clue how to integrate d457 gmsl  to it
Moreover I discovered xavier gmsl tier4 project that I haven't managed to deploy but again  there is no clue how to integrate it with d457 gmsl either
https://github.com/tier4/perception_ecu_container
the thing in the shared video I haven't managed to try as they won't give it to users like me who has no nvidia drive [50k usd unit] but only jetson + gmsl