realsenseai/librealsense #10242: control_transfer returned error, index: 768, error: Broken pipe, number: 32 and then repetitive Out of frame resources! Exception

juamar, 2022-02-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | I don't know | 
| Operating System & Version |   Win 10  | 
| Platform                                 | Android (ARM64!) |
| SDK Version                            |  2.50.0                          | 
| Language                            |  C#                       | 
| Segment			|  Smartphone                         | 

### Issue Description
I keep on making Librealsense work on a Xamarin Android application. This time i want to report some issue i'm having, in order to check with community if there is a way to improve my situation.

I'm trying to develop a nice and simple HelloWorld application as coded in the examples from the Android Wrapper but in Xamarin which calculates distance information from an object in front of the camera. I'm using the Android AAR library by using Xamarin Android Library Binder. I was facing an issue the previous days, I was just being able to retrieve the first 15 frames from the camera (StreamType.Depth). After that, i was catching the following exception:

Frame didn't arrive within 5000

Seems to be that the Pipeline object is not able to receive more Frames. Executing the test with the Plain Simple Android Example App was working perfectly, so it may be some software problem, rather than a Hardware problem, as i have seen in previous Issues. Also, i'm not doing anything weird in my code, the logic it's almost copy paste from the example, so i don't know what can be wrong.

Nowadays, as a workarround, i've made some PipeManager, so when we reach 15 frames, I replace the current Pipeline with a new one, and i keep switching them using Threads in a way that no Pipeline's management is perceibed for the user.

So before i was having this:

```
private void stream0()
    {
        int i = 0;
        Pipeline pipe = new Pipeline();
        pipe.Start();

        while (!mStreamingThread.StopRequest)
        {
            using (FrameSet frames = pipe.WaitForFrames())
            {

                using (Frame f = frames.First(StreamType.Depth))
                {
                    DepthFrame depth = (DepthFrame)f.As(Extension.DepthFrame);
                    float deptValue = depth.GetDistance(depth.Width / 2, depth.Height / 2);

                    i++;
                    this.RunOnUiThread(() => realsenseFragment.PrintToTest("Distance: " + deptValue + "\n" + i));
                }
            }
            Thread.Sleep(1000 / 30);
        }
        pipe.Stop();
    }
```

And now i have this, with my Custom pipe's manager:

```
private void stream2()
    {
        int i = 0;
        PipeManager pipeManager = new PipeManager();

        while (!mStreamingThread.StopRequest)
        {
            using (FrameSet frames = pipeManager.ActualPipeline.WaitForFrames())
            {
                using (Frame f = frames.First(StreamType.Depth))
                {
                    DepthFrame depth = (DepthFrame)f.As(Extension.DepthFrame);
                    float deptValue = depth.GetDistance(depth.Width / 2, depth.Height / 2);

                    i++;
[... 134 lines cut ...]
[librs] control_transfer returned error, index: 768, error: Broken pipe, number: 32
[librs] control_transfer returned error, index: 768, error: Broken pipe, number: 32
[librs] Failed to allocate composite frame
[librs] Dropped frame. alloc_frame(...) returned nullptr
[librs] Exception was thrown during callback: Out of frame resources!
[librs] Dropped frame. alloc_frame(...) returned nullptr
[librs] Exception was thrown during callback: Out of frame resources!
[librs] Dropped frame. alloc_frame(...) returned nullptr
[librs] Exception was thrown during callback: Out of frame resources!
[librs] Dropped frame. alloc_frame(...) returned nullptr
[librs] Exception was thrown during callback: Out of frame resources!
[librs] Dropped frame. alloc_frame(...) returned nullptr
[librs] Exception was thrown during callback: Out of frame resources!
[librs] Dropped frame. alloc_frame(...) returned nullptr
[librs] Exception was thrown during callback: Out of frame resources!
[librs] Dropped frame. alloc_frame(...) returned nullptr
[librs] Exception was thrown during callback: Out of frame resources!
[librs] Dropped frame. alloc_frame(...) returned nullptr
[librs] Exception was thrown during callback: Out of frame resources!
... //And then it just repeats itself
```

I know what i'm doing it's not supposed to be supported by intel, but i was expecting this to work.

BR

MartyG-RealSense, 2022-03-07:
Case closed due to no further comments received.

closed 2022-03-07 by MartyG-RealSense

juamar, 2022-03-27:
Hi after almost a month i returned to this. Sorry i did not replied before. I just wanted to update this because i resolved this in a different way.

As we are working with MAUI and we cannot work wirh .NET's Intel Realsense Wrapper, we are working with a Xamarin Java Binding to make the Intel Realsense SDK work.

The thing is, even thought the library is we are using is the java one, we were facing the same problem metioned on issue https://github.com/IntelRealSense/librealsense/issues/7201 but java library does not have Keep() method. So we are managing dispose and close manually (instead of using `using` statement). So managing frames and FrameSets like this solved this issue:

 ```
    private void stream0()
    {
        int i = 0;
        Pipeline pipe = new Pipeline();
        pipe.Start();

        while (!mStreamingThread.StopRequest)
        {
            FrameSet frames = pipe.WaitForFrames();
                Frame f = frames.First(StreamType.Depth);

                    DepthFrame depth = (DepthFrame)f.As(Extension.DepthFrame);
                    float deptValue = depth.GetDistance(depth.Width / 2, depth.Height / 2);                    

                    i++;
                    this.RunOnUiThread(() => realsenseFragment.PrintToTest("Distance: " + deptValue + "\n" + i));
                    f.Close();
                    f.Dispose();

                frames.Close();
                frames.Dispose();

            Thread.Sleep(1000 / 30);
        }
        pipe.Stop();
    }
```

Thank you very much @MartyG-RealSense!

MartyG-RealSense, 2022-03-27:
It's great news that you achieved a solution, @juamar - thanks so much for sharing it with the RealSense ROS community  :)

juamar, 2022-03-27:
Also, when trying to align Depth and Color frames, we can do this:

```
    private void stream0()
    {
        int i = 0;
        Config _cfg = new Config();
        _cfg.EnableStream(StreamType.Depth, -1, 1280, 720, StreamFormat.Z16, 30);
        _cfg.EnableStream(StreamType.Color, -1, 1280, 720, StreamFormat.Bgr8, 30);

        Pipeline pipe = new Pipeline();
        pipe.Start(_cfg);

        while (!mStreamingThread.StopRequest)
        {
            Align AlignFilter = new Align(StreamType.Color);
            FrameSet frames = pipe.WaitForFrames();
                //frames = (FrameSet)frames.ApplyFilter(AlignFilter);
                FrameSet framesAligned = AlignFilter.Process(frames);
                Frame f = frames.First(StreamType.Depth);

                    DepthFrame depth = (DepthFrame)f.As(Extension.DepthFrame);
                    float deptValue = depth.GetDistance(depth.Width / 2, depth.Height / 2);                    

                    i++;
                    this.RunOnUiThread(() => realsenseFragment.PrintToTest("Distance: " + deptValue + "\n" + i));
                    f.Close();
                    f.Dispose();

                Frame f2 = frames.First(StreamType.Color);
                    VideoFrame frame = (VideoFrame)f2.As(Extension.VideoFrame);
                    Console.WriteLine(frame.Height);
                    f2.Close();
                    f2.Dispose();
                
                framesAligned.Close();
                framesAligned.Dispose();

            frames.Close();
            frames.Dispose();
            AlignFilter.Close();
            AlignFilter.Dispose();

            Thread.Sleep(1000);
        }
        pipe.Stop();
    }
```

You must close and dispose both FrameSets, the original (`frames` in my example) and `framesAligned`. `framesAligned = frames.ApplyFilter()` was giving me problems when disposing both, as runtime was telling me that `frames` was already disposed when first dispossing framesAligned.