realsenseai/librealsense #10243: Local peak detection from point cloud

GrubbyHalo, 2022-02-17:
I am looking for a good way to find local peaks from the point cloud data from the d450 camera's.

MartyG-RealSense, 2022-02-17:
Hi @GrubbyHalo  The RealSense SDK has a feature called **Second Peak Threshold**, also known as Second Peak Delta.  Intel's **Depth Map Improvements for Stereo-based Depth Cameras on Drones** white paper provides a definition for the threshold's Second Peak Delta.

***************

When analyzing the disparities of an area in the stereo images for a match, there might be one clear candidate indicating a large peak in terms of correlation.  In some cases, multiple candidates could be viable at different peak levels.  The second peak threshold determines how big the difference from another peak needs to be, in order to have conﬁdence in the current peak being the correct one. 

****************

https://dev.intelrealsense.com/docs/depth-map-improvements-for-stereo-based-depth-cameras-on-drones#section-b-depth-camera-settings

GrubbyHalo, 2022-02-17:
Hi @MartyG-RealSense 
Thanks for the response. 
I am not sure how the second peak threshold setting would help me extract local peaks from the raw depth data. For example, in the attached image the local peaks are identified with yellow circles
[image]

How would I go about isolating those peaks.
Regards

MartyG-RealSense, 2022-02-17:
A number of researchers have used RealSense with a **convolutional network** for peak detection.  Some example papers are in the link below.

https://arxiv.org/pdf/1910.11515.pdf
https://www.ncbi.nlm.nih.gov/pmc/articles/PMC7506964/
https://www.biorxiv.org/content/10.1101/2020.05.21.109629v1.full

In the last link in that list, they based their method on a tool called **DeepPoseKit**

https://elifesciences.org/articles/47994

The **PCL** library also has a **convolution filter**.

https://pointclouds.org/documentation/classpcl_1_1filters_1_1_convolution.html

A paper in the link below describes how it used "a peak-finding algorithm is applied to search for the crest of a convolution".

https://www.researchgate.net/publication/339381286_Detection_System_for_U-Shaped_Bellows_Convolution_Pitches_Based_on_a_Laser_Line_Scanner

roelofvandijk, 2022-02-19:
Assuming your point cloud is oriented along the z-axis (or any other axis), you could try to find the points that have a large positive z-difference to the center of mass of all points in their neighborhood (in a radius of your choice).

MartyG-RealSense, 2022-02-19:
Thanks so much @roelofvandijk for your advice to @GrubbyHalo    :)

GrubbyHalo, 2022-02-20:
@roelofvandijk . That is a very interesting take on the problem and one that I will definitely have a look at. I am already masking off the area of interest . According to your suggestion, I will take all points in the area of interest, compute their centroid and look for points with a large z difference within that centroid.

GrubbyHalo, 2022-02-20:
@MartyG-RealSense Thank you for the references.

MartyG-RealSense, 2022-02-26:
Hi @GrubbyHalo  Do you require further assistance with this case, please?   Thanks!

GrubbyHalo, 2022-03-01:
@MartyG-RealSense Unfortunately I haven't come up with a satisfactory solution to this problem. Even when implementing the suggestions given by you and @roelofvandijk . I suspect that the solution is very case specific. Anyway I have a solution that kind of works but, that I feel could be improved on greatly. On that note I don't think there is a valid reason to keep this issue open. Thanks

closed 2022-03-01 by GrubbyHalo

roelofvandijk, 2022-03-01:
@GrubbyHalo, I found a kdtree-based implementation in python that should do what you want reasonably quickly:
https://stackoverflow.com/questions/27032562/local-maxima-in-a-point-cloud

If you share an example point cloud (*.ply) here can give it a shot.