realsenseai/librealsense #10245: Lifetime about start pipeline without stop for D455

seliber, 2022-02-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   |5.13.xxx | 
| Operating System & Version |  arm64 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Imx8  |
| SDK Version                            |  2.50                          | 
| Language                            | C++                         | 
| Segment			| Other                         | 

### Issue Description
From the latest data sheet of D455 in this link:
https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

The only part related with lifetime is under Depth Quality Specification, for D455, it is 5 years.

My question is:
If we enable the pipeline of RGB and Depth during the whole usage, will it impact the lifetime of the camera? And what's the lifetime for RGB camera?

Another issue is:
We found deadlock when stop pipeline calling pipeline.stop() in a thread and fetch frames in another thread called method: pipeline.try_wait_for_frames(&frame, 200) after long time running and frequently start and stop pipeline. Any suggestions on this?

MartyG-RealSense, 2022-02-18:
Hi @seliber   According to the data sheet document for the 400 Series cameras, table 4.8 containing the lifetime estimates for 400 Series models was a new addition in the August 2021 edition of the data sheet.  The accompanying notes for the table state that 'lifetime' is defined as "operating in default settings within the supported temperature for up to 16 hours a day (66%)".

[image]

Having RGB and depth both enabled should not affect the camera lifetime.  400 Series cameras are capable of running indefinitely so long as they remain within recommended operating temperatures.  

The camera's laser can degrade over time but the camera is able to continue operating, though depth quality may be reduced, as advised by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/3452#issuecomment-471492320 

There is not information available about the lifetime of the RGB sensor.  In the four years (at the time of writing this) since the launch of the 400 Series though, reported cases involving possible hardware failure of the RGB sensor component of 400 Series cameras have been extremely rare.

There have been a small number of past reports regarding long-run sessions where a program has frozen when stopping the pipeline after hours / days of running.  A couple of RealSense users found that a solution was to start the **pipeline** but stop the **sensor** instead of both starting and stopping the pipeline.  Links to these cases can be found at https://github.com/IntelRealSense/librealsense/issues/7252#issuecomment-695194103

Another approach is to create multiple pipelines and use sensor instead of pipeline for both the start and stop, as discussed at https://github.com/IntelRealSense/librealsense/issues/891 and https://github.com/IntelRealSense/librealsense/issues/1984

seliber, 2022-02-21:
Thanks for the advice, will try based on these.

closed 2022-03-09 by seliber

MartyG-RealSense, 2022-03-09:
Thanks very much @seliber for the update!

seliber, 2022-03-09:
@MartyG-RealSense  thanks for the support!