realsenseai/librealsense #10246: Cannot enumerate devices while in use (RS2_USB_STATUS_BUSY) (worked on v2.41.0)

stwirth, 2022-02-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.13.0-28-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.50.0                          | 
| Language                            | C++                           | 
| Segment			|  Robot                          | 

### Issue Description

While the camera is in use (running via realsense2_camera ROS driver), the enumeration of the camera fails with:
```
$ rs-enumerate-devices -s
Could not create device - failed to set power state . Check SDK logs for details
 18/02 14:32:07,080 ERROR [140097041467136] (handle-libusb.h:100) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 18/02 14:32:07,080 ERROR [140097075065024] (sensor.cpp:572) acquire_power failed: failed to set power state
 18/02 14:32:07,087 ERROR [140097049859840] (handle-libusb.h:100) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
Could not create device - failed to set power state . Check SDK logs for details
Device Name                   Serial Number       Firmware Version    
 18/02 14:32:07,087 ERROR [140097075065024] (sensor.cpp:572) acquire_power failed: failed to set power state
```

This used to work before.

The following is output from an older system, commands are run while the cameras are streaming images:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.7 LTS
Release:	16.04
Codename:	xenial
$ rs-enumerate-devices --version

rs-enumerate-devices  version: 2.41.0

$ uname -srvpo
Linux 4.4.0-210-generic #242-Ubuntu SMP Fri Apr 16 09:57:56 UTC 2021 x86_64 GNU/Linux

$ rs-enumerate-devices -s
Device Name                   Serial Number       Firmware Version    
Intel RealSense D410          747512021143        05.08.15.00         
Intel RealSense D410          747512020737        05.08.15.00  
```

MartyG-RealSense, 2022-02-18:
Hi @stwirth  There was a recent case at https://github.com/IntelRealSense/librealsense/issues/10207 where a RealSense user was having problems such as Ubuntu OS freezes when using firmware **5.13.0-27-generic** (very close to the 5.13.0-28-generic that you are using).  

The librealsense SDK does not have support for kernel 5.13 at the time of writing this.  The most recent formally supported kernel at the time of writing this is 5.11, and that support was added in librealsense 2.50.0.

Using more recent kernels than those that librealsense supports can work but it can have unpredictable effects in regard to stability.  In the case that I linked to, the RealSense user in that case achieved a fully stable librealsense build by using kernel **5.11.0-34-generic** as described at https://github.com/IntelRealSense/librealsense/issues/10207#issuecomment-1028836935

stwirth, 2022-02-18:
Thanks @MartyG-RealSense, we'll try with kernel 5.11 and report back here.

stwirth, 2022-02-18:
I added the tested 5.11 kernel version to the initial issue description above.

MartyG-RealSense, 2022-02-18:
It is curious that it worked for you in 2.41.0.  I would normally expect the camera to be inaccessible if ROS was already accessing it when another program tried to access it.  This is because under the RealSense SDK's **Multi-Streaming Model** rules, once a particular stream on a particular camera is accessed by a program then that program puts a claim on that specific stream, preventing it from being accessed by another program until the first program releases its claim (by stopping streaming or closing down).

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

For example, if the RealSense Viewer starts using the depth stream and then the 'rs-capture' example program is launched whilst the Viewer is still streaming, rs-capture cannot work because the Viewer already claimed the depth stream.  The reverse is also true - if rs-capture is launched first and the Viewer launched secondly then the Viewer cannot access the stream.

I recall that SDK 2.41.0 was a significant release in that it was causing problems for some RealSense users, so Intel made changes to how the V4L buffers were handled in 2.42.0 (and presumably kept that change in the SDK from that point onwards).  This is described in https://github.com/IntelRealSense/librealsense/issues/8154#issuecomment-767175826 and https://github.com/IntelRealSense/librealsense/pull/8234

stwirth, 2022-02-18:
I understand that streaming is limited to one consumer. But I don't see a reason for putting a read lock on meta information like serial number and firmware version.

FWIW 2.18.1 allowed this as well.

stwirth, 2022-02-18:
Yes, this could as well be caused by the firmware. Do you want me to test a specific version?

I see this as a regression because inspecting a running system without interfering in its function is a very useful feature.

MartyG-RealSense, 2022-02-18:
A good starting point would be **5.12.7.100**, the version before the internal firmware changes were introduced.  This would help confirm whether or not those internal changes are a factor in your problem.

MarcoAndrade26, 2022-02-18:
@stwirth This is caused because you need to use a USB Powered Hub to connect the camera.

MartyG-RealSense, 2022-02-19:
@stwirth  I performed a test with SDK 2.50.0 and the latest firmware where I launched a depth stream in the RealSense Viewer and then ran **rs-enumerate-devices -s** and confirmed your statement that you should be able to access the camera information even when streaming is active.  I also performed the test with two cameras attached like with your own setup and confirmed that the information of both cameras could be accessed while they were depth streaming.

I can understand the concern about not now being able to access the rs-enumerate-devices camera information whilst the ROS launch is running.  Most of the information provided by **rs-enumerate-devices - s** is also printed in the launch log during roslaunch for each launched camera though.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_camera.launch#L2-L4

[image]

MartyG-RealSense, 2022-03-03:
Case closed due to no further comments received.

closed 2022-03-03 by MartyG-RealSense

stwirth, 2022-03-03:
I haven't had time to work on this, will re-open when I get more insight.
As I said before, I think this is a regression and therefore should be addressed by the developers. If you can run `rs-enumerate-devices` while streaming with the RealSense Viewer then maybe the problem is on the ROS driver side.

MartyG-RealSense, 2022-03-03:
@stwirth  This issue is not likely to be addressed by RealSense developers.  If a RealSense user is able to create a fix then they are welcome to submit it via a pull request (PR) to the librealsense SDK or the ROS wrapper, depending on whether the fix is made in librealsense or the wrapper.

**librealsense SDK pull page**
https://github.com/IntelRealSense/librealsense/pulls

**RealSense ROS wrapper pull page**
https://github.com/IntelRealSense/realsense-ros/pulls

stwirth, 2022-03-03:
OK, thanks for the information.