realsenseai/librealsense #10247: Getting this error when running the depth quality application. Having trouble with getting point cloud from camera as well.

Skraldefar, 2022-02-18:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

MartyG-RealSense, 2022-02-18:
Hi @Skraldefar   Please try downloading and running an older version of the RealSense Viewer program and then launching the 2.50.0 version of the Viewer afterwards to see whether the Access Violation error disappears.  Instructions for this process can be found at https://github.com/IntelRealSense/librealsense/issues/10053#issuecomment-993256941

MartyG-RealSense, 2022-02-25:
Hi @Skraldefar  Do you require further assistance with this case, please?  Thanks!

Skraldefar, 2022-02-25:
No, thank you

MartyG-RealSense, 2022-02-25:
Okay, thanks very much for the update.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2022-02-25 by MartyG-RealSense