realsenseai/librealsense #10248: Get D4xx multi-camera example

woodal, 2022-02-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (5.12.9) | 
| Operating System & Version |   Win (10) | 
| Kernel Version (Linux Only)    |  -                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.41}                          | 
| Language                            |  {labview }                          | 
| Segment			|  {others }                          | 

### Issue Description
I refer to document "RealSense_Multiple_Camera_WhitePaper.pdf" and try to do D414 camera multi-connection.
I saw multi-connect LabVIEW example in the documentation.
[image]
I wish I could get this example.
Is it possible?
(Not in the basic wrapper example.)

Thanks!

MartyG-RealSense, 2022-02-19:
Hi @woodal  The demo in this paper that you highlighted was an internal Intel application from 2018.  It was not available to the public.  My understanding is that Intel decided to integrate multiple camera hardware sync support into the **RealSense Viewer** instead.

MartyG-RealSense, 2022-02-26:
I researched again but could not find any examples or guidance for use of multiple RealSense cameras with the LabVIEW wrapper unfortunately.

woodal, 2022-03-04:
It's okay. Thank you for your kind support as always.

MartyG-RealSense, 2022-03-04:
Thanks very much.  As further assistance is not required, I will close the case.  Thanks again!

closed 2022-03-04 by MartyG-RealSense