realsenseai/librealsense #10249: RS Viewer: unhandled exception, 1920x1080 NIR in RAW10 (auto-selected)

roelofvandijk, 2022-02-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400, D415 OEM module, D415 consumer model| 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |  Windows 11  | 
| Kernel Version (Linux Only)    |  -                                       | 
| Platform                                 | PC |
| SDK Version                            | 2.50.0  RELEASE                          | 
| Language                            |  Realsense Viewer                        | 
| Segment			|   -                  | 

### Issue Description

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

When:
* changing the resolution from 1280x720 to 1920x1080 
* turning on the depth sensor

An unhandled exception occurs and the Realsense Viewer closes.

[image]

The problem is maybe that RAW10 is automatically selected as data type?

Manually selecting a different data type (Y8/Y16) works, but choosing RAW10 or Y10BPACK reliably crashes the viewer (at the automatically selected resolution of 1920x1080).

I reproduced this on both a consumer D415 and on an OEM module.

roelofvandijk, 2022-02-19:
I tried the same steps with the 2.48 viewer as suggested in https://github.com/IntelRealSense/librealsense/issues/10053#issuecomment-993256941, this showed the expected error message on screen and did not crash.

[image]

However, it did not resolve the 2.50 viewer access violation.

MartyG-RealSense, 2022-02-19:
Hi @roelofvandijk  1920x1080 resolution is not supported for depth.  The highest supported depth resolution is 1280x720.

When 1920x1080 was selected in the Stereo Module with depth enabled in my own Viewer with a D415 camera, the Depth selection box automatically disabled and the Infrared box enabled at 1920x1080, Y8 format and 15 FPS (though 25 FPS was selectable for this infrared resolution too).  It was possible to stream both Infrared and Infrared 2 with this resolution and FPS.

[image]

[image]

As your image above shows, the RAW10 format cannot be streamed in the RealSense Viewer.  

Also, an attempt to stream the left infrared channel at 1920x1080 and RAW10 format resulted in the **Unhandled exception escaping from a worker thread!  Error type: Access Violation!** error that you experienced.

[image]

My tests found that if Depth is enabled and 1920x1080 then Depth is automatically de-selected and Infrared is automatically selected, **and** the infrared format is set as RAW10.  The format only sets RAW10 as the default when selecting 1920x1080 for depth.  So it should not occur if you avoid setting Depth to 1920x1080 and instead use 1280x720 as the maximum resolution for Depth.

In conclusion, it appears that the problems that you described come only from setting Depth to 1920x1080, and do not occur if you avoid doing so.

roelofvandijk, 2022-02-19:
Hello @MartyG-RealSense, thank you for your reply.

I follow your reasoning and know how to avoid this error, but this seems suboptimal in two ways:
* When choosing a different resolution, in most cases the other parameters switch to a supported combination of data types/enabled stream types/FPS. That is not the case in this situation.
* This case was handled gracefully in the previous SDK and now causes an unhandled exception.

I need no further support, I just wanted to file this as a bug report. You can close this issue if so desired.

MartyG-RealSense, 2022-02-20:
This particular issue only occurs with the D415 model, as the other 400 Series models such as D435 / D435i and D455 do not offer a 1920x1080 option in their depth resolution selector drop-down.  Until it becomes a more widely experienced problem, I will close for now but the issue can certainly be revisited if it becomes a problem for other RealSense users.  Thanks very much for highlighting the issue!

closed 2022-02-20 by MartyG-RealSense